MichaIng/DietPi #6936: Error while updating to 9.02 to current version 9.1.1

sfbuch, 2024-02-22:
#### Details:
- Date           | Thu Feb 22 23:38:36 GMT 2024
- DietPi version | v9.0.2 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 2 Model B (armv7l) (ID=2)
- Kernel version | `Linux DietPi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -y --with-new-pkgs upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. dietpi-update 
2. ...
#### Expected behaviour:

- 
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 perl : Depends: libperl5.32 (= 5.32.1-4+deb11u3) but 5.32.1-4+deb11u2 is installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).

ISquishWorms, 2024-02-22:
Just to say I am also getting this error and was about to make a post on the same topic.

#### Details:
- Date           | Thu Feb 22 23:54:16 GMT 2024
- Bug report     | 59276acf-c7a9-409c-b97f-e72ce1f121ea
- DietPi version | v9.0.2 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux RaspberryPi-4 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `curl -sSfLO https://github.com/MichaIng/DietPi/archive/master.tar.gz`
- Exit code      | 6
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
curl: (6) Could not resolve host: github.com
```

Joulinar, 2024-02-23:
@ISquishWorms your issue is completely different as you have some DNS challenges.. check your DNS server used as you are not able to resolve github.com

@sfbuch did you already tried `apt --fix-broken install`?

ISquishWorms, 2024-02-23:
> @ISquishWorms your issue is completely different as you have some DNS challenges. Check your DNS server used as you are not able to resolve github.com

Will do, I have my DNS set to PiHole using Unbound at the moment I will change that just for the update and try again.  Although I did have the same error sfbuch had initially before running the "apt --fix-broken install".

holocronology, 2024-03-26:
Also getting this on first run:
Reference code: 51bd4ec2-d122-43e9-8502-a129b74dbbf0

MichaIng, 2024-04-04:
Are you having multiple GitHub accounts or are you a different person? 😄

If Pi-hole or Unbound have issues resolving a hostname, both have query logs to check what went wrong. Also check whether both services are running and whether they report errors:
```sh
systemctl status unbound pihole-FTL
journalctl -u unbound -u pihole-FTL
```

Resolving hostnames to IPv6 addresses and actually connecting via IPv6 are two different things. The first should always work, even if the system does not actually support/have an IPv6 address. But connecting to/downloading from that resolved IPv6 would then fail. When doing a `dietpi-update` or `dietpi-software` installation, we actually check whether connecting to an IPv6 IP works, in case it is even enabled, and do a prompt in case if fails, before doing any APT upgrade, download or whatever. If this check fails, the error dialog allows you to disable IPv6 quickly. But also quite possible that IPv6 in general works, but has issues from a particular location/network to a particular remote host. Still not 100% reliable in 2024 😞.

Since OP is quite a while ago and the other found solutions, I'll mark this as closed.

closed 2024-04-04 by MichaIng