MichaIng/DietPi #6938: Orange Pi Zero 3 | 1.5 GB RAM variant looses Ethernet adapter on soft reboot

MichaIng, 2024-02-23:
Mainline U-Boot does not support the 1.5 GB RAM variant, and the Orange Pi bootloader has this issue. For more details: #6594

- [ ] Test own build of Orange Pi bootloader with `CONFIG_DRAM_SUN50I_H616_TRIM_SIZE=y`: https://github.com/orangepi-xunlong/u-boot-orangepi/tree/v2021.07-sunxi

Also we should keep an eye on the work of iuncuim from OpenWRT:
- https://forum.openwrt.org/t/can-someone-make-a-build-for-orange-pi-zero-3/168930/34
- https://gist.github.com/iuncuim

lalakii, 2024-03-23:
I have the 1.5GiB board

I found out the difference by comparing the source code.

source code: [u-boot/arch/arm/mach-sunxi/dram_sun50i_h616.c](https://github.com/u-boot/u-boot/blob/master/arch/arm/mach-sunxi/dram_sun50i_h616.c)
```
static unsigned long mctl_calc_size(const struct dram_config *config)
{
        u8 width = config->bus_full_width ? 4 : 2;

        /* 8 banks */
        unsigned long size;
        size = (1ULL << (config->cols + config->rows + 3)) * width * config->ranks;
        //  -----   CONFIG_DRAM_SUN50I_H616_TRIM_SIZE=y   -----
        size = (size * 3) / 4;
        //  ------------------------ END -----------------------
        return size;
} 
```

[image]

lalakii, 2024-03-23:
I think I may need to add a configuration menu to the mainline u-boot, but I don't know much about that, would anyone do that.

MichaIng, 2024-03-23:
Ah nice, you just nees to add `size = (size * 3) / 4;` and this works for U-Boot and also the final Linux system booted?

This looks so easy and obvious, I wonder whether iuncuim did not find that.

Also not sure how to implement into mainline U-Boot. Since this is not done in the device tree, I also have no other idea than basically adding this very same flag. At least it can potentially serve other H616/H618 boards as well.

lalakii, 2024-03-23:
Hello!

I've tried different kernels and found that they all have the wrong temperature information, which makes them unbootable

One problem has been identified so far:

```
[    2.143164] thermal thermal_zone0: gpu-thermal: critical temperature reached, shutting down
[    2.151581] reboot: HARDWARE PROTECTION shutdown (Temperature too high)
[    2.185438] reboot: Power down
```

My previous kernel could boot fine because I streamlined out the temperature module.

lalakii, 2024-03-23:
However there doesn't seem to be anything else I can do about this problem.      :(

MichaIng, 2024-03-23:
Next week I can generate a U-Boot build which matches the one we use for the 1/2/4 GB variants, just with this patch added. That one (using the Armbian build system), does have the right temperatures.

lalakii, 2024-03-24:
This u-boot seems to have a lot of problems with "dram_init". 

In the same way.

both values have the potential to occur.
```
        u8 width = config->bus_full_width ? 4 : 2;

        // Sometimes it's 2, sometimes it's 4
```

When memory size is incorrectly recognized, memory usage may crash if it exceeds what is actually available.

MichaIng, 2024-03-26:
Hmm, one would expect that `config->bus_full_width` is the correct condition for `width` to result in the correct memory size?

MichaIng, 2024-05-08:
Fixed with: #7064
Armbian just added a patch to fix this in mainline U-Boot. I compiled a new package and added it to our APT repository. If you want to test it now:
```sh
apt purge linux-u-boot-orangepizero3-next
apt install linux-u-boot-orangepizero3-current
. /usr/lib/u-boot/platform_install.sh
# To show progress and errors, in case
eval "$(declare -f write_uboot_platform | sed -e 's| > /dev/null 2>&1||g' -e 's|status=none|status=progress|g')"
write_uboot_platform "$DIR" "$(lsblk -npo PKNAME "$G_ROOTFS_DEV")"
```

closed 2024-05-12 by MichaIng