MichaIng/DietPi #6939: LXQt fails to install due to HTTP 404 (b4211d42-1740-4b1c-9fc1-d3fd26ae004d)

jtmoon79, 2024-02-24:
### Creating a bug report/issue
- [X] I have searched the existing open and closed issues

#### Required Information
- DietPi version
  ```
  G_DIETPI_VERSION_CORE=9
  G_DIETPI_VERSION_SUB=1
  G_DIETPI_VERSION_RC=1
  G_GITBRANCH='master'
  G_GITOWNER='MichaIng'
  G_LIVE_PATCH_STATUS[0]='not applicable'
  ```
- Distro version | `trixie`
- Kernel version | `Linux host 6.1.78 #1 SMP Mon Feb 19 18:07:08 UTC 2024 riscv64 GNU/Linux`
- SBC model | `StarFive VisionFive 2 (riscv64)`

#### Additional Information (if applicable)
- Software title | LXQt
- Was the software title installed freshly or updated/migrated? Freshly installed.
- Can this issue be replicated on a fresh installation of DietPi? Not sure.
- Bug report ID | `b4211d42-1740-4b1c-9fc1-d3fd26ae004d`

#### Steps to reproduce

1. start `dietpi-software`
2. select `LxQT`
3. select `install`
4. during install, a dietpi custom curl commands fails with HTTP Status 404
   `curl -sSfL https://github.com/MichaIng/DietPi/raw/master/.conf/desktop/lxqt/lxqt-trixie.7z -o lxqt-trixie.7z`

#### Expected behaviour

No failures for installs.

#### Actual behaviour

[image]

MichaIng, 2024-02-24:
Ah right, I did not yet create a config archive for Trixie/Sid, hence affecting all RISC-V boards.

MichaIng, 2024-02-24:
Another issue for Trixie systems (but not Sid, hence not RISC-V) is that the SpeedCrunch package has been removed from the repo due to some other block: https://tracker.debian.org/pkg/speedcrunch
I tries to find another Qt calculator, but could not find any. Hence for now LXQt installs on Trixie won't ship with a GUI calculator. But again Sid contains everything, hence RISC-V is not affected.

MichaIng, 2024-02-24:
Fixed for next DietPi release with: https://github.com/MichaIng/DietPi/commit/e643873
You can test it via `dev` branch:
```sh
G_DEV_BRANCH dev
```
It is not yet perfect. Somehow some of the applications lack icons, and the Adwaita icon theme does not provide fallback/override icons for them either. Even the LXQt panel itself lacks icons for some of its widgets, like the volume control and others in panel settings. One could install `lxde-icon-theme` and enable it ("nuoveXT2"), which adds some more icons. But they have worse quality and override some native application icons with older/worse ones, like Firefox.

Probably normal for Debian testing and/or unstable, not sure. I tested it on Trixie, so on Sid/RISC-V it could be theoretically different.

jtmoon79, 2024-02-25:
Wow, fast turnaround! Thanks @MichaIng !

> You can test it via dev branch:
> `G_DEV_BRANCH dev`

Which file do I add that line? Or is `G_DEV_BRANCH` an environment variable?

MichaIng, 2024-02-25:
It's a shell function: https://github.com/MichaIng/DietPi/blob/f891ed8/dietpi/func/dietpi-globals#L1966-L1973

MichaIng, 2024-02-25:
That sounds like a different issue indeed. There was no kernel upgrade in the meantime, was it? Checking back patches, nope, the `dev` branch does not apply a new one yet, and the latest on our server is `v6.1.78-dietpi1`, matching the one you had before. No other hardware changes/patches either 🤔.

closed 2024-03-17 by MichaIng