MichaIng/DietPi #6941: Nanopi R6C : nvme M2 disk not detected

SadarSSI, 2024-02-27:
Hi,

In first, thank you for your incredible work !

For information with last version downloaded from https://dietpi.com/downloads/images/DietPi_NanoPiR6C-ARMv8-Bookworm.img.xz, nvme disk is not detected : lsblk doesn't work

The disk is detect with friendlyElec, ArmBian and this distrib : https://github.com/inindev/nanopi-r6/releases/tag/v12-6.7-rc7

MichaIng, 2024-02-27:
Many thanks for your report.

That is strange, since we use the Armbian kernel and bootloader builds. Which kernel version is used on the Armbian image?
```sh
uname -a
```

SadarSSI, 2024-02-28:
Linux nanopi 6.7.0-rc1-edge-rockchip-rk3588 #1 SMP PREEMPT Mon Nov 13 00:19:07 UTC 2023 aarch64 GNU/Linux

`$ cat /etc/armbian-image-release
# PLEASE DO NOT EDIT THIS FILE
BOARD=nanopi-r6s
BOARD_NAME="NanoPi R6S"
BOARDFAMILY=rockchip-rk3588
BUILD_REPOSITORY_URL=https://github.com/armbian/build
BUILD_REPOSITORY_COMMIT=014eb55b5
LINUXFAMILY=rockchip-rk3588
ARCH=arm64
IMAGE_TYPE=stable
BOARD_TYPE=conf
INITRD_ARCH=arm64
KERNEL_IMAGE_TYPE=Image
FORCE_BOOTSCRIPT_UPDATE=
VENDOR="Armbian"
VENDORDOCS="https://docs.armbian.com"
VENDORURL="https://www.armbian.com"
VENDORSUPPORT="https://forum.armbian.com"
VENDORBUGS="https://www.armbian.com/bugs"
BOOTSCRIPT_FORCE_UPDATE="no"
BOOTSCRIPT_DST="boot.cmd"
VERSION=23.11.1
REVISION=23.11.1
IMAGE_UUID=0a1699d4-b850-4a61-9e5d-1ca28fa9f5d7
`

MichaIng, 2025-05-31:
Is this still an issue with recent vendor branch kernel (Linux 6.1)? We have people using an NVMe SSD successfully on the R6C.

closed 2025-06-22 by MichaIng