MichaIng/DietPi #6943: Onboard wifi not detected, no hotspot possible ROCK 4SE

MidG971, 2024-02-28:
### Creating a bug report/issue
- My onboard wifi card is not detected on my ROCK 4SE, maybe the wrong dri

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
**G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=1
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'**

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- bookworm

- Kernel version | `uname -a`
Linux DietPi 6.6.15-current-rockchip64 #2 SMP PREEMPT Thu Feb  1 00:19:14 UTC 2024 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
ROCK 4 (aarch64) [Actually ROCK 4SE]
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)
USB PD adapter
#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Freshly installed
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`
- fac68c52-ff4b-4cec-8920-f92e69b83da0

#### Steps to reproduce

1. No wifi card detected in dietpi-config

#### Expected behaviour

- card  detected, hotspot configurable

#### Actual behaviour

- No wifi
#### Extra details

dmseg to follow :

`root@DietPi:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.6.15-current-rockchip64 (armbian@next) (aarch64-linux-gnu-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #2 SMP PREEMPT Thu Feb  1 00:19:14 UTC 2024
[    0.000000] Machine model: Radxa ROCK 4SE
[    0.000000] efi: UEFI not found.
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000] NUMA: NODE_DATA [mem 0xf77cb980-0xf77cdfff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 128 MiB at 0x00000000e8200000 on node -1
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[... 628 lines cut ...]
v4l2_h264              16384  2 rockchip_vdec,hantro_vpu
rockchip_rga           24576  0
videobuf2_dma_sg       16384  1 rockchip_rga
v4l2_mem2mem           40960  3 rockchip_vdec,hantro_vpu,rockchip_rga
cfg80211              991232  1 brcmfmac
snd                   106496  9 snd_soc_hdmi_codec,snd_timer,snd_compress,snd_soc_core,snd_pcm
videobuf2_memops       16384  2 videobuf2_dma_contig,videobuf2_dma_sg
soundcore              12288  1 snd
videobuf2_v4l2         32768  4 rockchip_vdec,hantro_vpu,rockchip_rga,v4l2_mem2mem
videodev              294912  5 rockchip_vdec,videobuf2_v4l2,hantro_vpu,rockchip_rga,v4l2_mem2mem
rfkill                 28672  3 bluetooth,cfg80211
videobuf2_common       65536  8 rockchip_vdec,videobuf2_dma_contig,videobuf2_v4l2,hantro_vpu,rockchip_rga,videobuf2_dma_sg,v4l2_mem2mem,videobuf2_memops
mc                     77824  6 rockchip_vdec,videodev,videobuf2_v4l2,hantro_vpu,videobuf2_common,v4l2_mem2mem
cpufreq_dt             16384  0
binfmt_misc            24576  1
dm_mod                151552  0
ip_tables              32768  0
x_tables               57344  1 ip_tables
autofs4                45056  2
realtek                32768  1
dwmac_rk               28672  0
stmmac_platform        24576  1 dwmac_rk
stmmac                245760  3 stmmac_platform,dwmac_rk
pcs_xpcs               24576  1 stmmac
`

MichaIng, 2024-02-28:
Hmm, it is defined in the device tree: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm64/boot/dts/rockchip/rk3399-rock-4se.dts

Do you know which WiFi chip is used? Obviously not the same as on ROCK 4B. I don't own the device, so cannot test. Let's check back with some other users: https://dietpi.com/forum/t/radxa-rock-4-se-supported-not-booting/19263

MidG971, 2024-02-28:
According to Radxa[https://dl.radxa.com/rockpi4/docs/hw/rockpi4/ROCK-4-SE-V1.53-SCH.pdf](url) **AP6255**
[image]
I found this datasheet :
[https://datasheet.lcsc.com/lcsc/2202221430_AMPAK-Tech-AP6255_C2687149.pdf](url)

**BUT**
Prior to the release of the new image for the board, I tried to manually, setup a hotspot with Radxa's latest github image, here are the relevant parts of what I did, or  what info I had a the time:

`eth0: connected to Wired connection 1
        "eth0"
        ethernet (rk_gmac-dwmac), 5E:00:CF:5C:B8:65, hw, mtu 1500
        ip4 default, ip6 default
        inet4 192.168.1.116/24
        route4 0.0.0.0/0
        route4 192.168.1.0/24
        inet6 2a01:e0a:58f:dd60:481b:4685:fad8:bdfd/64
        inet6 fe80::3fe2:d92:5890:e917/64
        route6 2a01:e0a:58f:dd60::/64
        route6 ::/0
        route6 fe80::/64

wlan0: connected to Passione
        "Broadcom Wi-Fi"
        wifi (brcmfmac), E8:FB:1C:FE:D7:25, hw, mtu 1500
        inet4 192.168.1.85/24
        route4 0.0.0.0/0
        route4 192.168.1.0/24
        inet6 2a01:e0a:58f:dd60:3c3a:8d49:5512:c370/64
        inet6 fe80::1f64:9113:ba2a:ccef/64
        route6 2a01:e0a:58f:dd60::/64
        route6 ::/0
        route6 fe80::/64` 
        
        `rock@rock-4se:~$ iw list sudo 

Wiphy phy0
wiphy index: 0
max # scan SSIDs: 10
max scan IEs length: 2048 bytes
max # sched scan SSIDs: 0
max # match sets: 0
Retry short limit: 7
Retry long limit: 4
Coverage class: 0 (up to 0m)
Device supports roaming.
Supported Ciphers:
* WEP40 (00-0f-ac:1)
* WEP104 (00-0f-ac:5)
* TKIP (00-0f-ac:2)
* CCMP-128 (00-0f-ac:4)
* CCMP-256 (00-0f-ac:10)
* GCMP-128 (00-0f-ac:8)
* GCMP-256 (00-0f-ac:9)
* CMAC (00-0f-ac:6)
* CMAC-256 (00-0f-ac:13)
* GMAC-128 (00-0f-ac:11)
* GMAC-256 (00-0f-ac:12)
Available Antennas: TX 0 RX 0
Supported interface modes:
 * IBSS
 * managed
 * AP
 * P2P-client
 * P2P-GO
 * P2P-device
Band 1:
Capabilities: 0x1062
HT20/HT40
Static SM Power Save
RX HT20 SGI
[... 130 lines cut ...]
 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
 * P2P-device: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
Supported RX frame types:
 * managed: 0x40 0xb0 0xd0
 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
 * P2P-client: 0x40 0xd0
 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
 * P2P-device: 0x40 0xd0
Supported extended features:
* [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
* [ DFS_OFFLOAD ]: DFS offload
* [ AP_PMKSA_CACHING ]: PMKSA caching supported in AP mode
rock@rock-4se:~$ journalctl | grep -i wireless

journalctl | grep -i wireless

Hint: You are currently not seeing messages from other users and the system.
      Users in groups 'adm', 'systemd-journal' can see all messages.
      Pass -q to turn off this notice.
rock@rock-4se:~$ journalctl | grep -i wirelesssudo `

If I can tomorrow I'll do a fresh install one another emmc module I've got to get more info.

MidG971, 2024-02-29:
So here we are: 
**dmesg:**
`root@rock-4se:~# dmesg
[   14.801657] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[   14.801721] Linux version 5.10.110-18-rockchip (runner@be0cbbc3fde5) (aarch64-linux-gnu-gcc (Debian 10.2.1-6) 10.2.1 20210110, GNU ld (GNU Binutils for Debian) 2.35.2) #f22751370 SMP Wed Sep 13 12:33:32 UTC 2023
[   14.818094] Machine model: Radxa ROCK 4SE
[   14.818351] Malformed early option 'earlycon'
[   14.819022] efi: UEFI not found.
[   14.824444] OF: fdt: Reserved memory: failed to reserve memory for node 'drm-logo@00000000': base 0x0000000000000000, size 0 MiB
[   15.071746] Zone ranges:
[   15.071773]   DMA      [mem 0x0000000000200000-0x00000000f7ffffff]
[   15.071805]   DMA32    empty
[   15.071825]   Normal   empty
[   15.071844] Movable zone start for each node
[   15.071855] Early memory node ranges
[   15.071872]   node   0: [mem 0x0000000000200000-0x00000000f7ffffff]
[   15.071897] Initmem setup node 0 [mem 0x0000000000200000-0x00000000f7ffffff]
[   15.071918] On node 0 totalpages: 1015296
[   15.071935]   DMA zone: 15864 pages used for memmap
[   15.071949]   DMA zone: 0 pages reserved
[   15.071964]   DMA zone: 1015296 pages, LIFO batch:63
[   15.136515] cma: Reserved 16 MiB at 0x00000000f6400000
[   15.136657] psci: probing for conduit method from DT.
[   15.136682] psci: PSCIv1.0 detected in firmware.
[   15.136694] psci: Using standard PSCI v0.2 function IDs
[   15.136710] psci: MIGRATE_INFO_TYPE not supported.
[   15.136725] psci: SMC Calling Convention v1.0
[   15.137734] percpu: Embedded 32 pages/cpu s93416 r8192 d29464 u131072
[   15.137826] pcpu-alloc: s93416 r8192 d29464 u131072 alloc=32*4096
[   15.137848] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5
[   15.138174] Detected VIPT I-cache on CPU0
[   15.138311] CPU features: detected: ARM erratum 845719
[   15.138328] CPU features: detected: GIC system register CPU interface
[   15.138609] Built 1 zonelists, mobility grouping on.  Total pages: 999432
[   15.138640] Kernel command line: root=UUID=c06f09d4-e3e5-4e47-bb43-8f190b1726aa quiet splash loglevel=4 rw earlycon consoleblank=0 console=tty0 console=ttyFIQ0,1500000n8 console=ttyAML0,115200n8 console=ttyS2,1500000n8 console=ttyS0,1500000n8 coherent_pool=2M irqchip.gicv3_pseudo_nmi=0 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1
[   15.142193] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[   15.143013] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[   15.143034] mem auto-init: stack:off, heap alloc:off, heap free:off
[   15.306653] Memory: 3923488K/4061184K available (13696K kernel code, 3134K rwdata, 4848K rodata, 5760K init, 611K bss, 121312K reserved, 16384K cma-reserved)
[   15.307074] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=6, Nodes=1
[   15.307177] ftrace: allocating 53042 entries in 208 pages
[   15.571976] ftrace: allocated 208 pages with 3 groups
[   15.572762] rcu: Hierarchical RCU implementation.
[   15.572777] rcu:     RCU event tracing is enabled.
[   15.572794] rcu:     RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=6.
[   15.572808]  Rude variant of Tasks RCU enabled.
[   15.572818]  Tracing variant of Tasks RCU enabled.
[   15.572835] rcu: RCU calculated value of scheduler-enlistment delay is 30 jiffies.
[   15.572849] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=6
[   15.587775] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[   15.592263] GICv3: GIC: Using split EOI/Deactivate mode
[   15.592289] GICv3: 256 SPIs implemented
[   15.592302] GICv3: 0 Extended SPIs implemented
[   15.592403] GICv3: Distributor has no Range Selector support
[   15.592458] GICv3: 16 PPIs implemented
[   15.593656] GICv3: CPU0: found redistributor 0 region 0:0x00000000fef00000
[   15.594127] ITS [mem 0xfee20000-0xfee3ffff]
[   15.594682] ITS@0x00000000fee20000: allocated 65536 Devices @600000 (flat, esz 8, psz 64K, shr 0)
[   15.594734] ITS: using cache flushing for cmd queue
[   15.595432] GICv3: using LPI property table @0x00000000005c0000
[   15.595841] GIC: using cache flushing for LPI property table
[   15.595882] GICv3: CPU0: using allocated LPI pending table @0x00000000005d0000
[   15.595968] GICv3: GIC: PPI partition interrupt-partition-0[0] { /cpus/cpu@0[0] /cpus/cpu@1[1] /cpus/cpu@2[2] /cpus/cpu@3[3] }
[   15.596284] GICv3: GIC: PPI partition interrupt-partition-1[1] { /cpus/cpu@100[4] /cpus/cpu@101[5] }
[   15.597893] random: get_random_bytes called from start_kernel+0x3e0/0x5d8 with crng_init=0
[   15.878112] rockchip_clk_register_frac_branch: could not find dclk_vop0_frac as parent of dclk_vop0, rate changes may not work
[   15.894037] rockchip_clk_register_frac_branch: could not find dclk_vop1_frac as parent of dclk_vop1, rate changes may not work
[   16.056907] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[   16.056927] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[   16.056943] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[... 833 lines cut ...]
systemd-ask-password            systemd-notify
systemd-cat                     systemd-path
systemd-cgls                    systemd-resolve
systemd-cgtop                   systemd-run
systemd-delta                   systemd-socket-activate
systemd-detect-virt             systemd-stdio-bridge
systemd-escape                  systemd-sysusers
[... cut ...]
systemd-ask-password            systemd-notify
systemd-cat                     systemd-path
systemd-cgls                    systemd-resolve
systemd-cgtop                   systemd-run
systemd-delta                   systemd-socket-activate
systemd-detect-virt             systemd-stdio-bridge
systemd-escape                  systemd-sysusers
systemd-hwdb                    systemd-tmpfiles
systemd-id128                   systemd-tty-ask-password-agent
systemd-inhibit                 systemd-umount
root@rock-4se:~#
`

MidG971, 2024-02-29:
I dont know why when I copy MobaXterm's output it get this ugly on github, sorry for the inconvenience

Joulinar, 2024-02-29:
> I dont know why when I copy MobaXterm's output it get this ugly on github, sorry for the inconvenience

This has less to do with MobaXterm. Rather, it is the way you perform the formatting in GitHub. The following should be used for multiple code or log lines

````
```
code block line 1
code block line 2
code block line 3
```
````

Will result in

```
code block line 1
code block line 2
code block line 3
```

jannoke, 2024-03-04:
I have proof that i have AzuerWave wifi on board. BUT looking at the datasheet it seems pin compatible with AP6255. So I guess Radxa just changed/shuffles wifi module and did not bother to update docs.

[image]

MichaIng, 2024-03-05:
@jannoke 
See also my answer on the forum, to test with the correct device tree.

I tried to check whether AW-CM256SM and AP6255 are rebrands of the same chip, as I remember there was something like that. But Actually AW-CM256SM seems to be based on CYW43455 instead.

Checking further, AP6255 instead is based on BCM43455, which is suspiciously close. ... and indeed, CYW43455 is a rebrand of BCM43455. So this means AW-CM256SM and AP6255 use the same chip and hence should work with the same driver. They might still expect different firmware file names, but I remember that those two were just symlink to each other in the `firmware-brcm80211` package.

Ah, actually on several other boards with SDIO onboard WiFi, it was required to load this kernel module manually. Does this help?
```sh
modprobe sprdbt_tty
```

MidG971, 2024-03-05:
Here we go : 

```
root@DietPi:~# modprobe sprdwl_ng
Segmentation fault
root@DietPi:~# modprobe brcmfmac-cyw
root@DietPi:~# dietpi-config
root@DietPi:~# modprobe brcmfmac
root@DietPi:~# lsmod
Module                  Size  Used by
brcmfmac_cyw           12288  0
sprdwl_ng             356352  1
btsdio                 16384  0
bluetooth             811008  1 btsdio
snd_soc_simple_card    20480  0
snd_soc_audio_graph_card    16384  0
snd_soc_es8316         40960  1
snd_soc_rockchip_i2s    24576  2
snd_soc_simple_card_utils    32768  2 snd_soc_audio_graph_card,snd_soc_simple_card
snd_soc_spdif_tx       12288  0
snd_soc_core          294912  6 snd_soc_spdif_tx,snd_soc_audio_graph_card,snd_soc_simple_card_utils,snd_soc_rockchip_i2s,snd_soc_simple_card,snd_soc_es8316
snd_compress           28672  1 snd_soc_core
panfrost               77824  0
snd_pcm_dmaengine      16384  1 snd_soc_core
gpu_sched              49152  1 panfrost
brcmfmac              405504  1 brcmfmac_cyw
snd_pcm               143360  5 snd_compress,snd_soc_simple_card_utils,snd_soc_core,snd_soc_es8316,snd_pcm_dmaengine
drm_shmem_helper       24576  1 panfrost
rk_crypto              28672  0
brcmutil               24576  1 brcmfmac
hantro_vpu            262144  0
rng_core               20480  1 rk_crypto
rockchip_vdec          77824  0
v4l2_vp9               24576  2 rockchip_vdec,hantro_vpu
cfg80211              991232  2 brcmfmac,sprdwl_ng
snd_timer              40960  1 snd_pcm
snd                   106496  4 snd_timer,snd_compress,snd_soc_core,snd_pcm
videobuf2_dma_contig    24576  2 rockchip_vdec,hantro_vpu
rockchip_rga           24576  0
soundcore              12288  1 snd
v4l2_h264              16384  2 rockchip_vdec,hantro_vpu
rfkill                 28672  3 bluetooth,cfg80211
v4l2_mem2mem           40960  3 rockchip_vdec,hantro_vpu,rockchip_rga
videobuf2_dma_sg       16384  1 rockchip_rga
videobuf2_memops       16384  2 videobuf2_dma_contig,videobuf2_dma_sg
videobuf2_v4l2         32768  4 rockchip_vdec,hantro_vpu,rockchip_rga,v4l2_mem2mem
videodev              294912  5 rockchip_vdec,videobuf2_v4l2,hantro_vpu,rockchip_rga,v4l2_mem2mem
videobuf2_common       65536  8 rockchip_vdec,videobuf2_dma_contig,videobuf2_v4l2,hantro_vpu,rockchip_rga,videobuf2_dma_sg,v4l2_mem2mem,videobuf2_memops
mc                     77824  6 rockchip_vdec,videodev,videobuf2_v4l2,hantro_vpu,videobuf2_common,v4l2_mem2mem
cpufreq_dt             16384  0
dm_mod                151552  0
ip_tables              32768  0
x_tables               57344  1 ip_tables
autofs4                45056  2
realtek                32768  1
dwmac_rk               28672  0
stmmac_platform        24576  1 dwmac_rk
stmmac                245760  3 stmmac_platform,dwmac_rk
pcs_xpcs               24576  1 stmmac
root@DietPi:~#
```

```
root@DietPi:~# cd /tmp
root@DietPi:/tmp# ls
root@DietPi:/tmp# wget https://dietpi.com/downloads/binaries/linux-{image,dtb}-current-rockchip64
--2024-03-05 20:31:16--  https://dietpi.com/downloads/binaries/linux-image-current-rockchip64
Resolving dietpi.com (dietpi.com)... 2606:4700:20::681a:4f3, 2606:4700:20::ac43:4565, 2606:4700:20::681a:5f3, ...
Connecting to dietpi.com (dietpi.com)|2606:4700:20::681a:4f3|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2024-03-05 20:31:16 ERROR 404: Not Found.

--2024-03-05 20:31:16--  https://dietpi.com/downloads/binaries/linux-dtb-current-rockchip64
Reusing existing connection to [dietpi.com]:443.
HTTP request sent, awaiting response... 404 Not Found
2024-03-05 20:31:16 ERROR 404: Not Found.

root@DietPi:/tmp# ping google.fr
PING google.fr(par21s19-in-x03.1e100.net (2a00:1450:4007:813::2003)) 56 data bytes
64 bytes from par21s06-in-x03.1e100.net (2a00:1450:4007:813::2003): icmp_seq=1 ttl=112 time=3.09 ms
64 bytes from par21s06-in-x03.1e100.net (2a00:1450:4007:813::2003): icmp_seq=2 ttl=112 time=3.62 ms
64 bytes from par21s06-in-x03.1e100.net (2a00:1450:4007:813::2003): icmp_seq=3 ttl=112 time=4.00 ms
^C
--- google.fr ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2004ms
rtt min/avg/max/mdev = 3.087/3.568/3.997/0.373 ms

```

MichaIng, 2024-03-05:
```
root@DietPi:/tmp# wget https://dietpi.com/downloads/binaries/linux-{image,dtb}-current-rockchip64
```
There was the `.deb` file ending missing, my fault. Here again the commands:
```sh
cd /tmp
wget https://dietpi.com/downloads/binaries/linux-{image,dtb}-current-rockchip64.deb
dpkg -i linux-{image,dtb}-current-rockchip64.deb
rm linux-{image,dtb}-current-rockchip64.deb
G_CONFIG_INJECT 'fdtfile=' 'fdtfile=rockchip/rk3399-rock-4se.dtb' /boot/dietpiEnv.txt
reboot
```

MidG971, 2024-03-05:
Yes, figured this ;)
Still the same nonetheless :
```
root@DietPi:~# G_CONFIG_INJECT 'fdtfile=' 'fdtfile=rockchip/rk3399-rock-4se.dtb' /boot/dietpiEnv.txt
[  OK  ] G_CONFIG_INJECT | Desired setting in /boot/dietpiEnv.txt was already set: fdtfile=rockchip/rk3399-rock-4se.dtb

root@DietPi:~# lsmod
Module                  Size  Used by
btsdio                 16384  0
bluetooth             811008  1 btsdio
snd_soc_hdmi_codec     24576  1
panfrost               77824  0
snd_soc_audio_graph_card    16384  0
gpu_sched              49152  1 panfrost
drm_shmem_helper       24576  1 panfrost
snd_soc_simple_card    20480  0
snd_soc_spdif_tx       12288  0
dw_hdmi_i2s_audio      12288  0
snd_soc_simple_card_utils    32768  2 snd_soc_audio_graph_card,snd_soc_simple_card
brcmfmac              405504  0
snd_soc_es8316         40960  1
snd_soc_rockchip_i2s    24576  3
dw_hdmi_cec            12288  0
brcmutil               24576  1 brcmfmac
rockchip_vdec          77824  0
hantro_vpu            262144  0
rk_crypto              28672  0
snd_soc_core          294912  7 snd_soc_spdif_tx,snd_soc_hdmi_codec,snd_soc_audio_graph_card,snd_soc_simple_card_utils,snd_soc_rockchip_i2s,snd_soc_simple_card,snd_soc_es8316
cfg80211              991232  1 brcmfmac
rockchip_rga           24576  0
rng_core               20480  1 rk_crypto
snd_compress           28672  1 snd_soc_core
v4l2_vp9               24576  2 rockchip_vdec,hantro_vpu
snd_pcm_dmaengine      16384  1 snd_soc_core
snd_pcm               143360  6 snd_soc_hdmi_codec,snd_compress,snd_soc_simple_card_utils,snd_soc_core,snd_soc_es8316,snd_pcm_dmaengine
videobuf2_dma_contig    24576  2 rockchip_vdec,hantro_vpu
videobuf2_dma_sg       16384  1 rockchip_rga
v4l2_h264              16384  2 rockchip_vdec,hantro_vpu
v4l2_mem2mem           40960  3 rockchip_vdec,hantro_vpu,rockchip_rga
snd_timer              40960  1 snd_pcm
rfkill                 28672  3 bluetooth,cfg80211
videobuf2_memops       16384  2 videobuf2_dma_contig,videobuf2_dma_sg
videobuf2_v4l2         32768  4 rockchip_vdec,hantro_vpu,rockchip_rga,v4l2_mem2mem
snd                   106496  5 snd_soc_hdmi_codec,snd_timer,snd_compress,snd_soc_core,snd_pcm
soundcore              12288  1 snd
videodev              294912  5 rockchip_vdec,videobuf2_v4l2,hantro_vpu,rockchip_rga,v4l2_mem2mem
videobuf2_common       65536  8 rockchip_vdec,videobuf2_dma_contig,videobuf2_v4l2,hantro_vpu,rockchip_rga,videobuf2_dma_sg,v4l2_mem2mem,videobuf2_memops
mc                     77824  6 rockchip_vdec,videodev,videobuf2_v4l2,hantro_vpu,videobuf2_common,v4l2_mem2mem
cpufreq_dt             16384  0
dm_mod                151552  0
ip_tables              32768  0
x_tables               57344  1 ip_tables
autofs4                45056  2
realtek                32768  1
dwmac_rk               28672  0
stmmac_platform        24576  1 dwmac_rk
stmmac                245760  3 stmmac_platform,dwmac_rk
pcs_xpcs               24576  1 stmmac
root@DietPi:~# cat /boot/dietpiEnv.txt                                                  rootdev=UUID=88a725de-ac8d-428f-ae0e-2964c016dc23
rootfstype=ext4
# The init system logs to the console defined last.
consoleargs=console=ttyS2,1500000 console=tty1
usbstoragequirks=
extraargs=net.ifnames=0
docker_optimizations=off
overlay_path=rockchip
overlay_prefix=rockchip
overlays=
user_overlays=
fdtfile=rockchip/rk3399-rock-4se.dtb
[... 576 lines cut ...]
[    7.654757] panfrost ff9a0000.gpu: Features: L2:0x07120206 Shader:0x00000000 Tiler:0x00000809 Mem:0x1 MMU:0x00002830 AS:0xff JS:0x7
[    7.654764] panfrost ff9a0000.gpu: shader_present=0xf l2_present=0x1
[    7.659388] brcmfmac: F1 signature read @0x18000000=0x15264345
[    7.670063] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    7.684294] brcmfmac mmc2:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.radxa,rock-4se.bin failed with error -2
[    7.703993] usbcore: registered new interface driver brcmfmac
[    7.720121] [drm] Initialized panfrost 1.2.0 20180908 for ff9a0000.gpu on minor 1
[    7.778399] Bluetooth: Core ver 2.22
[    7.778542] NET: Registered PF_BLUETOOTH protocol family
[    7.778548] Bluetooth: HCI device and connection manager initialized
[    7.778578] Bluetooth: HCI socket layer initialized
[    7.778589] Bluetooth: L2CAP socket layer initialized
[    7.778616] Bluetooth: SCO socket layer initialized
[    8.087650] rk_gmac-dwmac fe300000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[    8.181038] rk_gmac-dwmac fe300000.ethernet eth0: PHY [stmmac-0:00] driver [RTL8211F-VD Gigabit Ethernet] (irq=POLL)
[    8.191455] rk_gmac-dwmac fe300000.ethernet eth0: No Safety Features support found
[    8.191517] rk_gmac-dwmac fe300000.ethernet eth0: PTP not supported by HW
[    8.193057] rk_gmac-dwmac fe300000.ethernet eth0: configuring for phy/rgmii link mode
[    8.743217] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[    9.774283] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[   12.322625] rk_gmac-dwmac fe300000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   18.914824] platform sound-dit: deferred probe pending
root@DietPi:~#

```

MichaIng, 2024-03-05:
```
[    7.659388] brcmfmac: F1 signature read @0x18000000=0x15264345
[    7.670063] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    7.684294] brcmfmac mmc2:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.radxa,rock-4se.bin failed with error -2
```
Strange that it does not try to load a fallback firmware without the device-specific appendix. Please try this:
```sh
ln -s brcmfmac43455-sdio.bin /lib/firmware/brcm/brcmfmac43455-sdio.radxa,rock-4se.bin
ln -s brcmfmac43455-sdio.txt /lib/firmware/brcm/brcmfmac43455-sdio.radxa,rock-4se.txt
modprobe -r brcmfmac
modprobe brcmfmac
ls -l
```

MidG971, 2024-03-05:
```
root@DietPi:~# dmesg | grep brcmfmac
[    7.659388] brcmfmac: F1 signature read @0x18000000=0x15264345
[    7.670063] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    7.684294] brcmfmac mmc2:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.radxa,rock-4se.bin failed with error -2
[    7.703993] usbcore: registered new interface driver brcmfmac
[    8.743217] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[    9.774283] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[ 2071.166243] usbcore: deregistering interface driver brcmfmac
[ 2071.356399] brcmfmac: F1 signature read @0x18000000=0x15264345
[ 2071.361975] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[ 2071.363413] usbcore: registered new interface driver brcmfmac
[ 2072.404884] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[ 2073.416585] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
root@DietPi:~# dietpi-config
```

MichaIng, 2024-03-05:
The firmware load error has gone, but otherwise it does not look like something has changed. The WiFi adapter is still not visible? If not, can you try it with the ROCK 4B device tree:
```sh
G_CONFIG_INJECT 'fdtfile=' 'fdtfile=rockchip/rk3399-rock-pi-4b.dtb' /boot/dietpiEnv.txt
reboot
```
The same is done in the Armbian config: https://github.com/armbian/build/blob/main/config/boards/rock-4se.csc
And on their website they state to use the 4B images: https://www.armbian.com/rockpi4/

This did work in the past for us as well, boot just broke very recently. Using the different U-Boot build worked, but maybe the kernel device tree for ROCK 4 SE has still issues.

MidG971, 2024-03-05:
> The firmware load error has gone, but otherwise it does not look like something has changed. The WiFi adapter is still not visible? If not, can you try it with the ROCK 4B device tree:
> 
> ```shell
> G_CONFIG_INJECT 'fdtfile=' 'fdtfile=rockchip/rk3399-rock-pi-4b.dtb' /boot/dietpiEnv.txt
> reboot
> ```
> 
> The same is done in the Armbian config: https://github.com/armbian/build/blob/main/config/boards/rock-4se.csc And on their website they state to use the 4B images: https://www.armbian.com/rockpi4/
> 
> This did work in the past for us as well, boot just broke very recently. Using the different U-Boot build worked, but maybe the kernel device tree for ROCK 4 SE has still issues.

I did that, still the same : led blinking but wlan card not found
```
root@DietPi:~# dmesg | grep brcmfmac
[    7.411031] brcmfmac: F1 signature read @0x18000000=0x15264345
[    7.430174] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    7.442834] brcmfmac mmc2:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.radxa,rockpi4b.bin failed with error -2
[    7.468786] usbcore: registered new interface driver brcmfmac
[    8.509931] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[    9.539272] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50

```

MichaIng, 2024-07-10:
Is this still an issue with latest kernel version? In case with ROCK 4 SE device tree?
```sh
G_CONFIG_INJECT 'fdtfile=' 'fdtfile=rockchip/rk3399-rock-4se.dtb' /boot/dietpiEnv.txt
reboot
```
Also try to reflash latest U-Boot
```sh
/boot/dietpi/func/dietpi-set_hardware flash-u-boot-mmc
```

spanner3003, 2024-07-15:
> ### Creating a bug report/issue
> * My onboard wifi card is not detected on my ROCK 4SE, maybe the wrong dri
> 
> #### Required Information
> * DietPi version | `cat /boot/dietpi/.version`
>   **G_DIETPI_VERSION_CORE=9
>   G_DIETPI_VERSION_SUB=6
>   G_DIETPI_VERSION_RC=1
>   G_GITBRANCH='master'
>   G_GITOWNER='MichaIng'
>   G_LIVE_PATCH_STATUS[0]='not applicable'**
> * Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
> * bookworm
> * Kernel version | `uname -a`
>   `Linux DietPi 6.6.34-current-rockchip64 #2 SMP PREEMPT Sun Jun 16 11:47:49 UTC 2024 aarch64 GNU/Linux`
> * SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
>   ROCK 4 (aarch64) [Actually ROCK 4SE]
> * Power supply used | (EG: 5V 1A RAVpower)
> * SD card used | (EG: SanDisk ultra)
>   USB PD adapter
> 
> #### Steps to reproduce
> 1. No wifi card detected in dietpi-config
> 
> #### Expected behaviour
> * card  detected, hotspot configurable
> 
> #### Actual behaviour
> * No wifi

spanner3003, 2024-07-15:
```bash
root@DietPi:/home/spanner# lsmod
Module                  Size  Used by
brcmfmac              368640  0
brcmutil               16384  1 brcmfmac
cfg80211              802816  1 brcmfmac
xt_conntrack           12288  1
nft_chain_nat          12288  3
xt_MASQUERADE          16384  1
nf_nat                 45056  2 nft_chain_nat,xt_MASQUERADE
nf_conntrack_netlink    45056  0
nf_conntrack          118784  4 xt_conntrack,nf_nat,nf_conntrack_netlink,xt_MASQUERADE
nf_defrag_ipv6         20480  1 nf_conntrack
nf_defrag_ipv4         12288  1 nf_conntrack
xfrm_user              40960  1
xfrm_algo              12288  1 xfrm_user
xt_addrtype            12288  2
nft_compat             16384  4
nf_tables             225280  75 nft_compat,nft_chain_nat
nfnetlink              16384  4 nft_compat,nf_conntrack_netlink,nf_tables
br_netfilter           28672  0
bridge                241664  1 br_netfilter
rc_tt_1500             12288  0
ts2020                 20480  1
m88ds3103              32768  1
i2c_mux                12288  1 m88ds3103
tda18250               20480  1
mn88472                20480  1
dvb_usb_dib0700       147456  5
dib7000m               20480  1 dvb_usb_dib0700
dib0090                28672  1 dvb_usb_dib0700
dib0070                16384  1 dvb_usb_dib0700
dib3000mc              16384  1 dvb_usb_dib0700
dvb_usb_dw2102         77824  7
dibx000_common         16384  2 dib7000m,dib3000mc
snd_soc_hdmi_codec     20480  1
dvb_usb                20480  2 dvb_usb_dw2102,dvb_usb_dib0700
dvb_core              122880  2 dvb_usb,dvb_usb_dw2102
rc_core                45056  5 dvb_usb,dvb_usb_dw2102,rc_tt_1500,dvb_usb_dib0700
videobuf2_vmalloc      16384  1 dvb_core
snd_soc_spdif_tx       12288  0
snd_soc_simple_card    20480  0
btsdio                 16384  0
snd_soc_audio_graph_card    16384  0
panfrost               69632  4
snd_soc_simple_card_utils    24576  2 snd_soc_audio_graph_card,snd_soc_simple_card
snd_soc_es8316         36864  1
bluetooth             663552  1 btsdio
dw_hdmi_cec            12288  0
snd_soc_rockchip_i2s    24576  4
gpu_sched              36864  1 panfrost
snd_soc_core          208896  7 snd_soc_spdif_tx,snd_soc_hdmi_codec,snd_soc_audio_graph_card,snd_soc_simple_card_utils,snd_soc_rockchip_i2s,snd_soc_simple_card,snd_soc_es8316
drm_shmem_helper       16384  1 panfrost
snd_compress           24576  1 snd_soc_core
dw_hdmi_i2s_audio      12288  0
snd_pcm_dmaengine      12288  1 snd_soc_core
snd_pcm               106496  6 snd_soc_hdmi_codec,snd_compress,snd_soc_simple_card_utils,snd_soc_core,snd_soc_es8316,snd_pcm_dmaengine
rk_crypto              28672  0
snd_timer              36864  1 snd_pcm
snd                    77824  5 snd_soc_hdmi_codec,snd_timer,snd_compress,snd_soc_core,snd_pcm
rng_core               16384  1 rk_crypto
soundcore              12288  1 snd
rockchip_vdec          77824  0
hantro_vpu            249856  0
rfkill                 24576  3 bluetooth,cfg80211
v4l2_vp9               20480  2 rockchip_vdec,hantro_vpu
videobuf2_dma_contig    20480  2 rockchip_vdec,hantro_vpu
rockchip_rga           20480  0
v4l2_h264              16384  2 rockchip_vdec,hantro_vpu
videobuf2_dma_sg       16384  1 rockchip_rga
[... 760 lines cut ...]
[   18.345691] m88ds3103 9-0068: downloading firmware from file 'dvb-demod-m88ds3103.fw'
[   19.422485] m88ds3103 9-0068: found a 'Montage Technology M88DS3103' in warm state
[   19.422520] m88ds3103 9-0068: firmware version: 3.B
[   19.440651] usb 1-1.4: DVB: adapter 0 frontend 0 frequency 0 out of range (950000..2150000)
[   19.447826] dw2102: su3000_power_ctrl: 0, initialized 1
[   19.451280] dw2102: su3000_power_ctrl: 1, initialized 1
[   19.455090] mn88472 10-0018: downloading firmware from file 'dvb-demod-mn88472-02.fw'
[   19.621403] usb 1-1.2: DVB: adapter 1 frontend 0 frequency 0 out of range (42000000..870000000)
[   19.767633] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[   19.770841] Bridge firewalling registered
[   20.431726] Initializing XFRM netlink socket
[   25.824248] platform sound-dit: deferred probe pending
[   44.636476] systemd-journald[349]: Time jumped backwards, rotating.
[  340.305811] usbcore: deregistering interface driver brcmfmac
[  350.832007] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[  350.832777] Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[  350.833448] Loaded X.509 cert 'wens: 61c038651aabdcf94bd0ac7ff06c7248db18c600'
[  350.833563] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
[  350.833578] cfg80211: failed to load regulatory.db
[  350.845571] brcmfmac: F1 signature read @0x18000000=0x15264345
[  350.851614] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[  350.852715] usbcore: registered new interface driver brcmfmac
[  351.917237] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[  352.919825] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
```

c00ldchan, 2024-08-12:
Hi everyone, 

I'm experiencing exactly the same behavior, but I'm running a Rock Pi 4C (not SE or PLUS). 
I tried both the stable img from the website as well as building my own image using the latest revision from the dev branch. 
The dmesg output is essentially the same: 
```
[    7.068786] brcmfmac: F1 signature read @0x18000000=0x15294345
[    7.074980] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43456-sdio for chip BCM4345/9
[    7.076386] usbcore: registered new interface driver brcmfmac
[    8.207483] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[    9.220768] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
```

Btw, Armbian works just fine, but since I'm not that experienced in Linux for SBC, I don't know where to dig. 
Not sure if this is the case, but the board is detected as `Model: Radxa ROCK Pi 4B` in the U-Boot output. 

Thanks!

c00ldchan, 2024-08-28:
Tried a few things for tracking down the issue:
* Switching to the Armbian kernel does not resolve the issue -- the error is still there
* Updating U-Boot to the one for RockPI 4C from Armbian does not help either
* No luck with the vanilla armbian-firmware package
* Detailed debug output from brcmfmac is also not very useful - the error is the same with no additional details
* Armbian (where the driver loads just fine) is using exactly the same brcmfmac drivers -- I've compared their checksums

MichaIng, 2024-09-19:
@c00ldchan 
You used the minimal/IoT image from here? https://www.armbian.com/rockpi4/

MichaIng, 2024-09-19:
Weird, so the image from the download page has functional WiFi, but building a fresh one with Armbian build system has not?

I also checked the image and could not find anything that would be related, like explicitly loaded module, module config, udev rule, dedicated package or service, boot config and script, device tree overlay etc.

On the functional Armbian image could you check `dmesg` as well, and compare to the non-functional one?

c00ldchan, 2024-09-19:
> Weird, so the image from the download page has functional WiFi, but building a fresh one with Armbian build system has not?

No, I mean both self-compiled and official builds produce same results - on Armbian the driver initializes fine, but not with Diet, so essentially this: 
Armbian Bookworm Minimal - OK 
Armbian Noble Minimal - OK 
DietPi (self-built & ISO) - keep getting `clkctl 0x50` when enabling brcmf

> I also checked the image and could not find anything that would be related, like explicitly loaded module, module config, udev rule, dedicated package or service, boot config and script, device tree overlay etc.

True, I haven't found anything that can cause this. Do you know if brcmf depends on some other driver/library which is not included in armbian-firmware? I can also try stealing it from Armbian repositories. 
Also is there a way to roll back to an older DietPI build and build an image locally? I recall it was working just fine some time ago, so I can try to bisect this. 

> On the functional Armbian image could you check dmesg as well, and compare to the non-functional one?

Yes, I tried that, even with brcmf verbose debug logging, but unfortunately it was useless - there are no significant differences: DietPI loads just fine, and then boom.

c00ldchan, 2024-09-21:
OK, attached two dmesg dumps, from Armbian and DietPi. I diffed them and found nothing suspicious.

<details>
<summary>Armbian</summary>

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.6.52-current-rockchip64 (build@armbian) (aarch64-linux-gnu-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1 SMP PREEMPT Wed Sep 18 17:24:10 UTC 2024
[    0.000000] KASLR disabled due to lack of seed
[    0.000000] Machine model: Radxa ROCK Pi 4C
[    0.000000] efi: UEFI not found.
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000] NUMA: NODE_DATA [mem 0xf77cb980-0xf77cdfff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 128 MiB at 0x00000000e9000000 on node -1
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 31 pages/cpu s88296 r8192 d30488 u126976
[    0.000000] pcpu-alloc: s88296 r8192 d30488 u126976 alloc=31*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: root=UUID=481ecb8c-ed11-45c6-b2d0-10fa6599afa5 rootwait rootfstype=ext4 splash=verbose console=ttyS2,1500000 console=tty1 consoleblank=0 loglevel=1 ubootpart=e21d3eb5-01 usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u   cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory
[    0.000000] Unknown kernel command line parameters "splash=verbose ubootpart=e21d3eb5-01 cgroup_enable=memory cgroup_memory=1", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Fallback order for Node 0: 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 999432
[    0.000000] Policy zone: DMA
[    0.000000] mem auto-init: stack:off, heap alloc:on, heap free:off
[    0.000000] software IO TLB: SWIOTLB bounce buffer size adjusted to 3MB
[    0.000000] software IO TLB: area num 8.
[    0.000000] software IO TLB: mapped [mem 0x00000000f2e00000-0x00000000f3200000] (4MB)
[    0.000000] Memory: 3803044K/4061184K available (16960K kernel code, 2318K rwdata, 5248K rodata, 4736K init, 604K bss, 127068K reserved, 131072K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=6, Nodes=1
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=6.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=6
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 256 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fef00000
[    0.000000] ITS [mem 0xfee20000-0xfee3ffff]
[    0.000000] ITS@0x00000000fee20000: allocated 65536 Devices @500000 (flat, esz 8, psz 64K, shr 0)
[    0.000000] ITS: using cache flushing for cmd queue
[    0.000000] GICv3: using LPI property table @0x00000000004f0000
[    0.000000] GIC: using cache flushing for LPI property table
[    0.000000] GICv3: CPU0: using allocated LPI pending table @0x0000000000580000
[... 1206 lines cut ...]
[    7.167480] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43456-sdio for chip BCM4345/9
[    7.168880] usbcore: registered new interface driver brcmfmac
[    7.182917] hantro-vpu ff650000.video-codec: Adding to iommu group 0
[    7.183148] rockchip_vdec: module is from the staging directory, the quality is unknown, you have been warned.
[    7.184093] hantro-vpu ff650000.video-codec: registered rockchip,rk3399-vpu-enc as /dev/video1
[    7.184289] hantro-vpu ff650000.video-codec: registered rockchip,rk3399-vpu-dec as /dev/video2
[    7.185747] rkvdec ff660000.video-codec: Adding to iommu group 1
[    7.289437] Bluetooth: Core ver 2.22
[    7.289531] NET: Registered PF_BLUETOOTH protocol family
[    7.289536] Bluetooth: HCI device and connection manager initialized
[... cut ...]
[    8.582249] rk_gmac-dwmac fe300000.ethernet eth0: No Safety Features support found
[    8.582265] rk_gmac-dwmac fe300000.ethernet eth0: PTP not supported by HW
[    8.582613] rk_gmac-dwmac fe300000.ethernet eth0: configuring for phy/rgmii link mode
[    9.325085] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[   13.696877] rk_gmac-dwmac fe300000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   18.401998] platform sound-dit: deferred probe pending
```

</details>

c00ldchan, 2024-09-22:
Oh wow, you're correct - it's the Bluetooth blacklist causing this. Nice catch, thank you! 

I think I managed to track it down to the `hci_uart` module. The driver loads fine with only the first line commented out in `dietpi-disable_bluetooth.conf`:
```
# blacklist hci_uart
blacklist hidp
blacklist rfcomm
blacklist btbcm
blacklist bnep
blacklist bluetooth
```

MichaIng, 2024-10-16:
This should fix it: https://github.com/MichaIng/DietPi/commit/5af8a85

closed 2024-10-16 by MichaIng