MichaIng/DietPi #6945: [Pine64] Update from v8.25.1 to 9.1.1 fail

yin0203, 2024-03-01:
### Creating a bug report/issue
- [* ] I have searched the existing open and closed issues

#### Details:
- Date           | Fri Mar  1 14:23:15 HKT 2024
- DietPi version | v8.25.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- Hardware       | PINE A64 (aarch64) (ID=40)
- Kernel version | `Linux DietPi 6.1.63-current-sunxi64 #1 SMP Mon Nov 20 10:52:                                                                                                                                                             19 UTC 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `curl -sSfLO https://github.com/MichaIng/DietPi/archive/maste                                                                                                                                                             r.tar.gz`
- Exit code      | 139
- Software title | DietPi-Update
#### Steps to reproduce:

1. run the dietpi-update
2. during the command curl -sSfLO https://github.com/MichaIng/DietPi/archive/maste                                                                                                                                                             r.tar.gz ,  it will show error code 139
#### Expected behaviour:

- suppose the upgrade will be successful
#### Actual behaviour:

- upgrade failed
#### Extra details:
- ...
#### Additional logs:
```

```

Joulinar, 2024-03-01:
hmm maybe a temp issue to download the achive. Can you try again? Or following on a differnt system

```
curl -sSfLO https://github.com/MichaIng/DietPi/archive/master.tar.gz
```

Oleks1y, 2024-03-03:
the same for ASUS TinkerBoard 
i have a brick after the update to 9.1.1

Joulinar, 2024-03-03:
@Oleks1y well I doubt it's the same as on the first report, it not even start the update.

yin0203, 2024-03-11:
FYI, I have re-flashed the image...... and using the 9.1 version with no problem.
Checked the sd card with no read/write error.

closed 2024-03-11 by Joulinar