MichaIng/DietPi #6948: issue rpi-eeprom-config -e

Uli007git, 2024-03-02:
### Creating a bug report/issue
- [ x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`

G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=1
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`

bookworm 0

- Kernel version | `uname -a`

Linux ThunderPi5 6.1.0-rpi8-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

RPi 5 Model B (aarch64)

- Power supply used | (EG: 5V 1A RAVpower)

Original RPI 5 power supply 27 W

- SD card used | (EG: SanDisk ultra)

SSD PLUS SanDisk 240 GB

#### Additional Information (if applicable)

I was trying to fix the RPI5 power consumption during standby as per https://www.jeffgeerling.com/blog/2023/reducing-raspberry-pi-5s-power-consumption-140x

#### Steps to reproduce

Edited the eeprom and when I tried to save it it gave me the error and didn't save.

#### Expected behaviour

Eeprom should be modified after editing and saving it.

#### Actual behaviour
I'm getting an error message when I try to modify the eeprom

ERROR: EEPROM image 'WARNING: flashrom not found. Setting RPI_EEPROM_USE_FLASHROM to 0
/lib/firmware/raspberrypi/bootloader-2712/default/pieeprom-2024-02-16.bin' not found

#### Extra details

-

MichaIng, 2024-03-02:
Fixed already:
- https://github.com/raspberrypi/rpi-eeprom/pull/547
- https://github.com/raspberrypi/rpi-eeprom/pull/549

Until the new package is live, either:
```sh
G_CONFIG_INJECT 'RPI_EEPROM_USE_FLASHROM=' 'RPI_EEPROM_USE_FLASHROM=0' /etc/default/rpi-eeprom-update
```
or:
```sh
apt install flashrom
```
Background:
- Recently, support for `flashrom` has been integrated into `rpi-eeprom-update` to allow flashing the EEPROM binary directly, instead of having this done on reboot.
- The variable `RPI_EEPROM_USE_FLASHROM` has been introduced to use or not use it, which defaults to `1` (use it).
- When you did not set this to `0`, but the `flashrom` tool/package is not installed, `rpi-eeprom-update` did output a warning whenever executed, even when it is just used to list the latest EEPROM image and not flash anything.
- This warning message hence mixed in the output which should actually contain the latest EEPROM path only, causing an invalid path for `rpi-eeprom-config` to choke on.

The fix moved the warning form STDOUT to STDERR (so that is is not used by  `rpi-eeprom-config`) and additionally the `flashrom` check is only done, when actually updating the EEPROM, not anymore when listing available EEPROM binaries or doing other things.

Uli007git, 2024-03-02:
Thanks... great job :) !!! 

I did the 

> apt install flashrom

closed 2024-03-17 by MichaIng