MichaIng/DietPi #6954: Open Beta v9.2 | Please help testing and hardening our upcoming release

MichaIng, 2024-03-10:
RC version | v9.2.0
----------- | -------
v9.2.0 PR | #6953
Raw changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release date | **2024-02-19**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av9.2
____
### Beta v9.2.0
_(2024-03-09)_

#### Enhancements
- NanoPi R4S | Resolved an issue where Ethernet adapter of the "LAN" port could disappear after a soft reboot. Many thanks to @idaanx for reporting this issue: https://github.com/MichaIng/DietPi/issues/6342
- DietPi-Software | It is now possible to run `dietpi-software list` concurrent to other `dietpi-software` instances, and as non-root user. This avoids an issue in DietPi-Dashboard, where opening `dietpi-software` in the Terminal and switching to the Software page, caused an infinite hang.

#### Bug fixes
- DietPi-Globals | Resolved an issue where a concurrent DietPi script detection could have prevented the start of a script, if a previous instance was killed via SIGKILL.
- DietPi-Config | Resolved an issue where a WiFi connection, configured for the first time via `dietpi-config` from a local terminal session, was automatically stopped when exiting or logging out from the terminal session.
- DietPi-Software | PaperMC: Resolved an issue where installing the Geyser and Floodgate plugins failed due to changed download URLs, and a false syntax in the Geyser config file. Many thanks to @boterocamilo for reporting this issue: https://github.com/MichaIng/DietPi/issues/6898
- DietPi-Software | Nextcloud: Resolved an issue with the updated Nginx config where assets were served with the wrong MIME type. This fix was backported to v9.1 and a live patch offered. Many thanks to `@howardroark` for reporting this issue: https://dietpi.com/forum/t/nextcloud-theme-broken-after-upgrade-v9-1-1/19372
- DietPi-Software | Home Assistant: Resolved an issue on x86_64 and ARMv8 systems where some core integrations did not work if no C++ compiler was installed. If you are affected by this, `apt install g++ && systemctl restart home-assistant` will fix it. Many thanks to @pdsakurai for fixing this issue in our install code: https://github.com/MichaIng/DietPi/pull/6931
- DietPi-Software | DarkIce: Resolved an issue where the service failed to start due to false config file permissions.
- DietPi-Software | LXQt: Resolved an issue where the installation failed on Trixie and RISC-V systems, since we did not host a config archive for those Debian versions yet. Many thanks to @jtmoon79 for reporting this issue: https://github.com/MichaIng/DietPi/issues/6939

christiancelona, 2024-03-10:
It is possible to update libopus to version 1.5.1

MichaIng, 2024-03-10:
This is not controlled by us but by Debian. See the `libopus0` versions on different Debian versions here: https://packages.debian.org/libopus0
Stable is Debian Bookworm, hence `libopus0` v1.3.1-3. v1.5.1 is not even in Sid/unstable, but it has been recognised: https://tracker.debian.org/pkg/opus

Hence if you need a newer version, you need to compile it from source: https://github.com/xiph/opus/blob/main/README#L64

MichaIng, 2024-03-17:
### DietPi v9.2 has been released: #6955
Many thanks to all testers ❤️ !

closed 2024-03-17 by MichaIng