MichaIng/DietPi #6958: Update DietPi from v8.22.3 to v8.25.2

bogus666, 2024-03-11:
#### Details:
- Date           | Mon Mar 11 08:39:40 CET 2024
- Bug report     | 4432dfdb-f3f7-40d2-9b09-d8a8e5c69a28
- DietPi version | v8.22.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | `Linux DietPi 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=1)                                                                                              
- Command        | `apt-get -y update`                                                                                                   
- Exit code      | 255                                                                                                                   
- Software title | DietPi-Update                                                                                                         
#### Steps to reproduce:                                                                                                                 
                                                                                              
1. ...                                                                                                                                   
2. ...                                                                                                                                   
#### Expected behaviour:                                                                                                                 
                                                                                                             
- updating DietPi from v8.22.3 to v8.25.2                                                                                                                        
#### Actual behaviour:                                                                                                                   
                                                                                                              
- Error message "The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>"                                                                                                            
#### Extra details:                                                                                                                      
                                                                   
- No changes have been made to the Diet-PI configuration, all previous updates worked fine                                                                                                                              
#### Additional logs:                                                                                                                    
```                                                                                                                                      
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease                                                                          
```

Joulinar, 2024-03-11:
You are running quite old Debian Buster system. And the PGP for one package source seems to be invalid. You would need to update the key. A possible workaround could be https://www.faqforge.com/linux/solved-the-following-signatures-were-invalid-expkeysig-b188e2b695bd4743-deb-sury-org-automatic-signing-key/

In addition, you should think on updating from Buster to Bullseye because we already dropped support on this old Debian version https://dietpi.com/blog/?p=811

bogus666, 2024-03-11:
Thanks for the quick reply, worked perfectly! Bullseye update is next on the list.

closed 2024-03-11 by Joulinar

MichaIng, 2024-03-11:
I would btw not use the `packages.sury.org` repository anymore, unless you have a good reason to use newer PHP but old Debian. We used this on Stretch to pull PHP 7.3, but in our Stretch => Buster upgrade howto, we removed the repository. And if you require newer PHP, it makes sense to instead upgrade further to Debian Bullseye and Bookworm.