MichaIng/DietPi #6959: Installing MATE alongside other WMs during initial setup causes a package conflict

anijatsu, 2024-03-12:
### Creating a bug report/issue
- [X] I have searched the existing open and closed issues
- The closest I have found is here, but I don't understand what I'd need to do (nor do I know why this has been apparently marked as solved?)
https://github.com/MichaIng/DietPi/issues/5214#issuecomment-1024288638

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
9.1.1
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bullseye 0
- Kernel version | `uname -a`
linux dietpi 6.1.21-v8+ 1642 SMP PREEMPT Mon Apr 3 17:24:16 BST 2023 aarch64 GNU/LINUX
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 4 Model B (aarch64)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)
n/a, external SSD enclosure

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
mate-desktop-environment-core
- Was the software title installed freshly or updated/migrated?
Fresh install
- Can this issue be replicated on a fresh installation of DietPi?
Pretty sure, since this was on a clean install of the OS.
- Bug report ID | `echo $G_HW_UUID`
c661b6cc-bf35-401c-b2b2-75d3c7d9602e
#### Steps to reproduce

1. Try to install several WMs during the initial setup of DietPI (there was a Dietpi-software step or something)
I remember selecting LXDE, LXQT, MATE, thinking I'd go and test them one by one to find one that fits me best.
2. During one of the APT commands, an error occurs:
```The following packages have unmet dependencies:
libgail-3-0 : Depends: libgtk-3-0 (= 3.24.24-4+deb11u3) but 3.24.24-4+rpt12+deb11u3 is to be installed
E: Unable to correct problems, you have held broken packages.
```

#### Expected behaviour
- If the WMs are incompatible, a workaround should be introduced to containerise them somehow, or
- inform the user and/or prevent the installation

#### Actual behaviour
- The installation is blocked from proceeding:
```
APT install mate-desktop-environment-core mate-media upower xterm
- Command: apt-get -y --allow-change-held-packages install mate-desktop-environment-core mate-media upower xterm
- Exit code: 100
- DietPi version: v9.1.1 (MichaIng/master) | HW_MODEL: 4 | HWARCH: 3 DISTRO: 6
- Error log:
Reading package lists...
Building dependency tree...
Reading state information...
upower is already the newest version (0.99.11-2) •
Some packages could not be installed. This may mean that you have requested an impossible situation or if you are using the unstable distribution that some required packages have not yet been created or been moved out of Incoming.
The following information may help to resolve the situation:
The following packages have unmet dependencies:
libgail-3-0 : Depends: libgtk-3-0 (= 3.24.24-4+deb113) but 3.24.24-4+rpt12+deb11u3 is to be installed
E: Unable to correct problems, you have held broken packages.
```
[image]

#### Extra details
- I stumbled upon this OS when I was looking through different distro options in PINN
- Showing the bug report reference code as a QR code would make it a little easier to get off of a physical display

MichaIng, 2024-03-19:
Thanks for your report. When LXDE is installed, we install an APT config to prevent the installation of packages from the RPi OS desktop. Since LXDE is installed first when you do all in one run, the config should be present already before other desktops can pull some of those packages. So it should be even worse when installing e.g. MATE first and LXDE afterwards. Looks like we missed some package, causing the dependency conflict.

We need to make this in a better way: No pinning individual binary packages, but whole source packages, and do that OOTB on RPi images, not just on LXDE installs.

`libgtk-3-0` should be however pinned, so not 100% sure why it wants to install the version from the RPi repo. Probably another RPi repo package, not installed with LXDE but with MATE, causes this.

Can you show:
```sh
apt policy libgtk-3-0
```

MichaIng, 2025-08-05:
This should prevent such issues: https://github.com/MichaIng/DietPi/commit/6949ab2