MichaIng/DietPi #6960: Impossible to connect to FREEBOX server (first boot)

CR95580, 2024-03-12:
### Creating a bug report/issue
- [x ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | v9.1.1
- Distro version | bookworm (ID=7,RASPBIAN=0)
- Kernel version | Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr 3 ...
- SBC model | RPi3B
- Power supply used | 
- SD card used | SanDisk 

#### Additional Information (if applicable)
This is during the first boot. I only changed before :
AUTO_SETUP_NET_WIFI_ENABLED=1 
and DOMAIN = 192.168.1.254 in the file "dietpi.txt "

#### Actual behaviour
Installation stops on the command "getent hosts DOMAIN" and there is no communication with the server !

#### Extra details
If I ping the server (ping 192.168.1.254), raspberry sends and receives returns from the server !

CR95580, 2024-03-12:
Correction.
The second line changed in "dietpi.txt" was not DOMAIN but 
# Hostname
AUTO_SETUP_NET_HOSTNAME=Freebox-39042E

MichaIng, 2024-03-13:
First of all, apart of `AUTO_SETUP_NET_WIFI_ENABLED=1` in `dietpi.txt`, you did also set the WiFi APs SSID and key in `dietpi-wifi.txt`?

`getent hosts <someDomain>` is used to check whether DNS is setup correctly to resolve host names. If this step was reached, it means that the IPv4 and in case IPv6 connection/ping test did already succeed. Did you see two lines:
```
[  OK  ] DietPi-Update | Checking IPv4 network connectivity
[  OK  ] DietPi-Update | Checking IPv6 network connectivity
```
Before the `getent hosts` error prompt? If so, only the DNS settings seem to be not correct. If you did not apply a static IP in `dietpi.txt`, the DHCP server would have need to serve one functional. From the error prompt (or any other console), you can check:
```sh
cat /etc/resolv.conf
```
for the `nameserver` entry, which should contain the IP address of your router, in rare cases of your ISP. If this seems valid, please also try testing DNS with another hostname:
```sh
getent hosts dietpi.com
getent hosts google.com
```
If those all fail, try with a different DNS server:
```sh
echo 'nameserver 1.1.1.1' > /etc/resolv.conf
getent hosts dietpi.com
```
which is Cloudflare DNS. If this works, you can either apply a static IP (including static DNS), or check your router/DHCP server's settings, why it serves a non-functional DNS server to clients.

CR95580, 2024-03-13:
@MichaIng 
Thanks for the quick response and the explanations.
No, I did not see this dietpi-wifi.txt file before my first installation.
I redo a brand new installation and I added in this file not only the SSID and the WPA key for the WiFi slot 0 BUT I also changed the WiFi Country Code = FR (yes, I am in France). And now everything is going well.
I am ready to begin my Diet !

MichaIng, 2024-03-19:
Okay great. The WiFi country code is usually set/overridden by the AP, i.e. even if you choose another one, after connecting to the AP, it would switch to what the AP enforces. But there are exceptions from this, based on router/AP model.

closed 2024-03-19 by MichaIng