MichaIng/DietPi #6961: nanopi r6c dietpi-update failed.

r3jack, 2024-03-15:
#### Details:
- Date           | Fri Mar 15 11:11:58 CST 2024
- DietPi version | v8.20.0 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | NanoPi 6 series (aarch64) (ID=79)
- Kernel version | `Linux sto 5.10.110 #1 SMP Tue Mar 14 21:59:07 CST 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `curl -sSfO https://dietpi.com/downloads/nanopi6.7z`
- Exit code      | 22
- Software title | DietPi-Patch
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
curl: (22) The requested URL returned error: 404 
```

Joulinar, 2024-03-15:
@MichaIng is this patch still required for NanoPi? Because the file is missing but it's still part of the v8.20 patch

https://github.com/MichaIng/DietPi/blob/master/.update/patches#L1250-L1251

MichaIng, 2024-03-16:
Oh right, we removed the patch from `dev` but removed the files from our server too quickly. However, release is later tonight, hence removing the patch from master is probably not required anymore. Next update instead contains a migration to Armbian-based kernel and bootloader builds.

r3jack, 2024-03-17:
dietpi-update to 9.2 success, thx.

closed 2024-03-17 by r3jack

MichaIng, 2024-03-17:
Great, thanks for the feedback!