MichaIng/DietPi #6963: Update from old 6.x version broken

leonkiriliuk, 2024-03-15:
### Creating a bug report/issue
- [ ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`

**G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=31
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'**

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`

**buster 1**

- Kernel version | `uname -a`

**Linux Netbooter 5.10.17-v7+ #1414 SMP Fri Apr 30 13:18:35 BST 2021 armv7l GNU/Linux**

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

**RPi 3 Model B+ (armv7l)**

- Power supply used | (EG: 5V 1A RAVpower) : 5V 2A
- SD card used | (EG: SanDisk ultra) : Sandisk 32Gb

#### Additional Information (if applicable)

execute: dietpi-update 

Result:

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/server_version-6
curl: (22) The requested URL returned error: 404 
[ INFO ] DietPi-Update | No or error response from: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/server_version-6 ()
[ INFO ] DietPi-Update | Checking mirror: https://dietpi.com/downloads/dietpi-update_mirror/master/server_version-6
curl: (22) The requested URL returned error: 404 
[ INFO ] DietPi-Update | No or error response from: https://dietpi.com/downloads/dietpi-update_mirror/master/server_version-6 ()
[FAILED] DietPi-Update | Unable to access any update server. Please check your network connection, then rerun dietpi-update.
If this error persists, please report at: https://github.com/MichaIng/DietPi/issues

#### Extra details
I'm able to do a Distro update; I've upgraded to Bullseye 11 on another micrSD card, but I've never been able to get dietpi-update to function correctly to get itself up to date.

Joulinar, 2024-03-15:
Holy moly, this is such an old system. I'm not sure if the update path is still available. 

@MichaIng definitely something we need your support on

leonkiriliuk, 2024-03-15:
Developers put out images for others to use on their Pi as an IOT device. As a software developer, I like to keep my devices up to date so decided to upgrade the image others use. I already opened an issue against this project but was hoping to find a fix without needing to go down the route of a fresh install due to the high level of customization done for this image.

Joulinar, 2024-03-15:
Your system is outdated for years. Even support for Buster has been dropped already. Only chance is with @MichaIng to have a look

MichaIng, 2024-03-15:
Yeah, at some point we needed to get rid of legacy backwards-compatible code. However, the branch which allows upgrading from v6.x to v8.x exists. Please try this:
```sh
G_DEV_BRANCH 8
```
To upgrade to latest DietPi v9.x, you'd need to upgrade the underlying Debian version to Bullseye. Buster is EOL, and we dropped support at the beginning of this year: https://dietpi.com/blog/?p=811#manual-upgrade

You could even further upgrade to the current stable Debian Bookworm, which we made easier, using a script: https://dietpi.com/blog/?p=3128#upgrade

Joulinar, 2024-03-15:
> G_DEV_BRANCH 8

Does this global already exist on 6.x?

MichaIng, 2024-03-15:
Yes: https://github.com/MichaIng/DietPi/blob/a32dc3ba941f5b8d92252b5954fe54b07f543504/dietpi/func/dietpi-globals#L2043-L2050
This is v6.29.

leonkiriliuk, 2024-03-15:
It worked! Thank you!

DietPi v8.25.2 is now listed and I can now update to the V9 branch if I want.

closed 2024-03-15 by leonkiriliuk