MichaIng/DietPi #6967: Bullseye to Bookworm upgrade broken

leonkiriliuk, 2024-03-18:
### Creating a bug report/issue
- [X] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`

G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=2
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`

bullseye 1

- Kernel version | `uname -a`

Linux netbooter 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux

#### Steps to reproduce

Following the upgrade instructions here: https://dietpi.com/blog/?p=3128#upgrade 
I executed the following command: bash <(curl -sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/master/.meta/dietpi-bookworm-upgrade')

The upgrade stopped with an error on a specific command asking me what I want to do.  When I run the command manually in another window I get the error repeated:

apt-get -y -eany update
Ign:1 https://chunksin.github.io/wipi_ppa  InRelease
Hit:2 https://chunksin.github.io/wipi_ppa  Release                                                     
Ign:3 https://chunksin.github.io/wipi_ppa  Release.gpg                                                 
Ign:4 https://deb.nodesource.com/node_10.x bookworm InRelease                                          
Err:5 https://deb.nodesource.com/node_10.x bookworm Release                                            
  404  Not Found [IP: 172.67.10.205 443]
Hit:6 http://raspbian.raspberrypi.com/raspbian bookworm InRelease                
Hit:7 https://archive.raspberrypi.com/debian bookworm InRelease                  
Reading package lists... Done                              
E: The repository 'https://deb.nodesource.com/node_10.x bookworm Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

leonkiriliuk, 2024-03-18:
I removed the nodesource from the apt repo. Must be something that stuck around from a long time ago and now works.

closed 2024-03-18 by leonkiriliuk

MichaIng, 2024-03-18:
Right, in the meantime that repo seems to be distro-agnostic, see the "nodistro" here: https://deb.nodesource.com/

Amd of course the Node 10 part might not be updated anymore.

We used that repository as well until some years ago, but switched to a tarball based installation, because the repo does not support all architectures, IIRC. Not sure whether we in turn remover the repo cleanly from all systems in that turn.

leonkiriliuk, 2024-03-18:
Is there a prescribed method when upgrading from one Debian release to another when it comes to Python? With PHP, given the installed packages are maintained by apt, moving from PHP 7.3 to PHP 8, the extra user installed packages are upgraded for the most part.  But with Python, one must maintain a list of everything installed by pip/pip3 as moving between Python 3 releases wipes away all the customizations done in the older release.