MichaIng/DietPi #6968: Segmentation fault for all calls to HTTPs (curl, python, ...). Probably due to openssl

cazzoo, 2024-03-18:
### Creating a bug report/issue
- [X] I have searched the existing open and closed issues

#### Required Information
- DietPi version
```G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=1
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
```
- Distro version | `bookworm 0`
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- SBC model | `RPi 3 Model B (aarch64)`
- Power supply used | (EG: 5V 2A RAVpower)
- SD card used | ?

#### Additional Information (if applicable)
- Software title | dietpi, curl, home-assistant, any python, anything that uses SSL
- Was the software title installed freshly or updated/migrated? Installed, working and suddenly stopped
- Can this issue be replicated on a fresh installation of DietPi? I don't know

#### Steps to reproduce

1. get the latest updates
2. run a curl command: `curl --trace - https://webservice.fanart.tv`

#### Expected behaviour
- Respond from the service without any segmentation fault

#### Actual behaviour
- Result of the command:
```
== Info:   Trying 51.89.210.79:443...
== Info: Connected to webservice.fanart.tv (51.89.210.79) port 443 (#0)
== Info: ALPN: offers h2,http/1.1
Segmentation fault
```

#### Extra details

- curl --version
```
curl 7.88.1 (aarch64-unknown-linux-gnu) libcurl/7.88.1 OpenSSL/3.0.11 zlib/1.2.13 brotli/1.0.9 zstd/1.5.4 libidn2/2.3.3 libpsl/0.21.2 (+libidn2/2.3.3) libssh2/1.10.0 nghttp2/1.52.0 librtmp/2.3 OpenLDAP/2.5.13
Release-Date: 2023-02-20, security patched: 7.88.1-10+deb12u5
Protocols: dict file ftp ftps gopher gophers http https imap imaps ldap ldaps mqtt pop3 pop3s rtmp rtsp scp sftp smb smbs smtp smtps telnet tftp
Features: alt-svc AsynchDNS brotli GSS-API HSTS HTTP2 HTTPS-proxy IDN IPv6 Kerberos Largefile libz NTLM NTLM_WB PSL SPNEGO SSL threadsafe TLS-SRP UnixSockets zstd
```

MichaIng, 2024-03-18:
That is weird. The versions/builds look all as expected, and also thr OpenSSL/libssl3 package from the RPi repo did not receive any update recently.

Does a reinstall fix it?
```sh
apt install --reinstall libssl3
```
Since curl runs until that point and the CLI works it is one of the libraries involved, libssl3 likely, but it could be probably another one as well. Do other kind of requests fail as well, like HEAD:
```
curl -I https://dietpi.com/
```

closed 2024-03-18 by cazzoo

MichaIng, 2024-03-18:
I'm glad it worked. Keep an eye on
```
dmesg -l 0,1,2,3
```
Whether you see I/O or filesystem errors from the kernel which might have caused the corruption.

cazzoo, 2024-04-11:
I just got the issue occuring again. I suspect the dietpi backup failed.
Here's what I get from the previous command you shared :
[image]

Do you think I may change the sdcard ? I feel the FS got corrupted somehow and then it became RO

MichaIng, 2024-04-19:
Indeed, a filesystem corruption lead to an automatic R/O remount. In case it is feasible and not the case yet, it can make sense to move `/mnt/dietpi_userdata` to an external drive, like a USB stick or something, instead of having these often write-intense dirs on the SD card.

A reboot should do an automatic filesystem check and repair (if possible), and results can be checked with:
```sh
journalctl -t systemd-fsck
```
If it happens more often, a new more durable SD card, or enabling and use boot the whole image from USB, can make sense.

cazzoo, 2024-05-02:
I tried to fix the OS but everything went wrong.
I ended up with changing my sdcard to a new one, it appears it was the source of most of my issues. I just restored my backed OS from external USB drive and that was all good.
Now I switched, my dietpi system is much more quicker (maybe better perfs from the SD itself) and no issues happened since (a week from now).
Thanks for the support anyway