MichaIng/DietPi #6973: The systemd-logind service fails with timeout when running in a Proxmox LXC container

seud0nym, 2024-03-19:
### Creating a bug report/issue
- [X] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `9.2.1 (MichaIng/master)`
- Distro version | `bookworm`
- Kernel version | `Linux DietPi 6.5.13-1-pve #1 SMP PREEMPT_DYNAMIC PMX 6.5.13-1 (2024-02-05T13:50Z) x86_64 GNU/Linux`
- SBC model | `Proxmox 8.1.4`

#### Steps to reproduce
1. Create a new minimal VM using standard install instructions
2. Use the https://github.com/thushan/proxmox-vm-to-ct script to convert the  new VM to a container
3. Log in, and after completing DietPi setup, run `journalctl -f`. The following messages will be displayed:
```
Mar 19 06:46:15 DietPi systemd[1]: systemd-logind.service: start operation timed out. Terminating.
Mar 19 06:46:15 DietPi systemd[1]: systemd-logind.service: Failed with result 'timeout'.
Mar 19 06:46:15 DietPi systemd[1]: Failed to start systemd-logind.service - User Login Management.
Mar 19 06:46:15 DietPi systemd[1]: systemd-logind.service: Scheduled restart job, restart counter is at 1.
Mar 19 06:46:15 DietPi systemd[1]: Stopped systemd-logind.service - User Login Management.
Mar 19 06:46:15 DietPi systemd[1]: Starting modprobe@drm.service - Load Kernel Module drm...
Mar 19 06:46:15 DietPi systemd[1]: modprobe@drm.service: Deactivated successfully.
Mar 19 06:46:15 DietPi systemd[1]: Finished modprobe@drm.service - Load Kernel Module drm.
Mar 19 06:46:15 DietPi systemd[1]: Starting systemd-logind.service - User Login Management...
Mar 19 06:46:15 DietPi systemd-logind[1987]: New seat seat0.
Mar 19 06:47:45 DietPi systemd[1]: systemd-logind.service: start operation timed out. Terminating.
Mar 19 06:47:45 DietPi systemd[1]: systemd-logind.service: Failed with result 'timeout'.
Mar 19 06:47:45 DietPi systemd[1]: Failed to start systemd-logind.service - User Login Management.
Mar 19 06:47:45 DietPi systemd[1]: systemd-logind.service: Scheduled restart job, restart counter is at 2.
Mar 19 06:47:45 DietPi systemd[1]: Stopped systemd-logind.service - User Login Management.
Mar 19 06:47:45 DietPi systemd[1]: Starting modprobe@drm.service - Load Kernel Module drm...
Mar 19 06:47:45 DietPi systemd[1]: modprobe@drm.service: Deactivated successfully.
Mar 19 06:47:45 DietPi systemd[1]: Finished modprobe@drm.service - Load Kernel Module drm.
Mar 19 06:47:45 DietPi systemd[1]: Starting systemd-logind.service - User Login Management...
Mar 19 06:47:45 DietPi systemd-logind[1992]: New seat seat0.
```

The `systemctl status systemd-logind` command shows:
```
● systemd-logind.service - User Login Management
     Loaded: loaded (/lib/systemd/system/systemd-logind.service; static)
     Active: activating (start) since Tue 2024-03-19 06:56:47 GMT; 1min 2s ago
       Docs: man:sd-login(3)
             man:systemd-logind.service(8)
             man:logind.conf(5)
             man:org.freedesktop.login1(5)
   Main PID: 2038 (systemd-logind)
     Status: "Processing requests..."
      Tasks: 1 (limit: 76770)
     Memory: 1.3M
        CPU: 20ms
     CGroup: /system.slice/systemd-logind.service
             └─2038 /lib/systemd/systemd-logind

Mar 19 06:56:47 DietPi systemd[1]: Starting systemd-logind.service - User Login Management...
Mar 19 06:56:47 DietPi systemd-logind[2038]: New seat seat0.
```

#### Expected behaviour
The systemd-logind service should move to active state, like the VM:
```
● systemd-logind.service - User Login Management
     Loaded: loaded (/lib/systemd/system/systemd-logind.service; static)
     Active: active (running) since Tue 2024-03-19 06:40:46 GMT; 20min ago
       Docs: man:sd-login(3)
             man:systemd-logind.service(8)
             man:logind.conf(5)
             man:org.freedesktop.login1(5)
   Main PID: 2993 (systemd-logind)
     Status: "Processing requests..."
      Tasks: 1 (limit: 2349)
     Memory: 1.6M
        CPU: 49ms
     CGroup: /system.slice/systemd-logind.service
             └─2993 /lib/systemd/systemd-logind

Mar 19 06:40:46 DietPi systemd[1]: Starting systemd-logind.service - User Login Management...
Mar 19 06:40:46 DietPi systemd-logind[2993]: New seat seat0.
Mar 19 06:40:46 DietPi systemd-logind[2993]: Watching system buttons on /dev/input/event1 (Power Button)
Mar 19 06:40:46 DietPi systemd-logind[2993]: Watching system buttons on /dev/input/event0 (AT Translated Set 2 keyboard)
Mar 19 06:40:46 DietPi systemd[1]: Started systemd-logind.service - User Login Management.
```

#### Actual behaviour
The service times out.

#### Extra details
I thought it might be to do with detecting power buttons, so I modified the /etc/systemd/logind.conf and uncommented the Handle... options and changed them all to `ignore`, but it made no difference.

MichaIng, 2024-03-19:
Hmm, that should result in a dependency error instead of timeout, but `dbus` runs?
```sh
systemctl status dbus
journalctl -u dbus
```
```

seud0nym, 2024-03-19:
```
root@DietPi:~# systemctl status dbus
● dbus.service - D-Bus System Message Bus
     Loaded: loaded (/lib/systemd/system/dbus.service; static)
     Active: active (running) since Tue 2024-03-19 06:44:45 GMT; 1h 52min ago
TriggeredBy: ● dbus.socket
       Docs: man:dbus-daemon(1)
   Main PID: 128 (dbus-daemon)
      Tasks: 1 (limit: 76770)
     Memory: 1.4M
        CPU: 43ms
     CGroup: /system.slice/dbus.service
             └─128 /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation --syslog-only

Mar 19 06:44:45 dietpi-test-ct systemd[1]: Starting dbus.service - D-Bus System Message Bus...
Mar 19 06:44:45 dietpi-test-ct systemd[1]: Started dbus.service - D-Bus System Message Bus.
root@DietPi:~# journalctl -u dbus
Mar 19 06:44:45 dietpi-test-ct systemd[1]: Starting dbus.service - D-Bus System Message Bus...
Mar 19 06:44:45 dietpi-test-ct systemd[1]: Started dbus.service - D-Bus System Message Bus.
root@DietPi:~# 
```

MichaIng, 2024-03-19:
Hmm, no idea. Does reinstalling systemd help?
```
apt install --reinstall systemd
```
Which Debian and kernel version is the Proxmox host system?

seud0nym, 2024-03-19:
Proxmox host:
```
root@pve-master:~# cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
root@pve-master:~# uname -a
Linux pve-master 6.5.13-1-pve #1 SMP PREEMPT_DYNAMIC PMX 6.5.13-1 (2024-02-05T13:50Z) x86_64 GNU/Linux
root@pve-master:~# 
```

Your suggestion of reinstalling systemd seems to have fixed systemd-logind:
```
● systemd-logind.service - User Login Management
     Loaded: loaded (/lib/systemd/system/systemd-logind.service; static)
     Active: active (running) since Tue 2024-03-19 09:39:14 GMT; 5min ago
       Docs: man:sd-login(3)
             man:systemd-logind.service(8)
             man:logind.conf(5)
             man:org.freedesktop.login1(5)
   Main PID: 105 (systemd-logind)
     Status: "Processing requests..."
      Tasks: 1 (limit: 76770)
     Memory: 1.3M
        CPU: 20ms
     CGroup: /system.slice/systemd-logind.service
             └─105 /lib/systemd/systemd-logind

Mar 19 09:39:14 dietpi-test-ct systemd[1]: Starting systemd-logind.service - User Login Management...
Mar 19 09:39:14 dietpi-test-ct systemd-logind[105]: New seat seat0.
Mar 19 09:39:14 dietpi-test-ct systemd[1]: Started systemd-logind.service - User Login Management.
```

Thank you for your help.

closed 2024-03-19 by seud0nym

MichaIng, 2024-03-19:
Okay great, whatever the issue was that it was corrupted.