MichaIng/DietPi #6975: APT upgrade Exit code: 100

fagron, 2024-03-20:
#### Details:
- Date           | Wed Mar 20 18:54:48 CET 2024
- Bug report     | 7ea5bcf6-7c72-45a4-9430-a72738a1ad8b
- DietPi version | v8.23.3 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux backup 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y --with-new-pkgs upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
```

Joulinar, 2024-03-20:
Can you share the whole output of following 

```
apt update
apt upgrade
```

Joulinar, 2024-03-20:
Should be similar to this one https://github.com/MichaIng/DietPi/issues/6747#issuecomment-1808348281

fagron, 2024-03-20:
1.
G_AGP rpi-eeprom raspi-firmware
G_AGUP
G_AGI rpi-eeprom

2. 
dietpi-update

Result:
DietPi v9.2.1 : 19:42 - Mi 20.03.2024

Thank you, it worked really well.

closed 2024-03-20 by Joulinar