MichaIng/DietPi #6976: unmount: /boot error while executing the migration script

Denguido, 2024-03-20:
#### Details:
- Date           | Wed Mar 20 22:42:49 CET 2024
- DietPi version | v9.3.-1 (MichaIng/dev)
- Image creator  | DietPi Core Team
- Pre-image      | From scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux Pandora 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `umount /boot`
- Exit code      | 32
- Software title | dietpi-rpi-firmware-migration

#### Steps to reproduce:
execute the migration script
bash <(curl -sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/.meta/dietpi-rpi-firmware-migration')

#### Expected behaviour:
no error

#### Actual behaviour:
I get the following error:
[FAILED] dietpi-rpi-firmware-migration | umount /boot

#### Extra details:

- ...
#### Additional logs:
```
umount: /boot: target is busy.
```

Denguido, 2024-03-20:
Stopping the vmtouch service before running the migration script resolved the problem.
systemctl stop vmtouch.service

It works fine on the RPI 5... Many thanks!

closed 2024-03-20 by Denguido

MichaIng, 2024-04-04:
Thanks for reporting. Seems we should forcefully uninstall `vmtouch` in the script, in case it is the instance we installed years ago as temporary replacement for the RAM disk used before for DietPi scripts. This can be safely detected by the `/etc/default/vmtouch` config file, where `/boot/dietpi` was defined for being locked into filesystem cache. Or we just uninstall it for everyone in this case on next DietPi update. There is really not much reason to keep it.