MichaIng/DietPi #6977: DietPI Sample rate issue

KenMoreira, 2024-03-21:
I have been using a USBridge (allo) for a long time now)
recently I have had my Sample rate locked to 48k While using Audirvana from PC. As well as when I use Mconnect from my phone.

The issue is consistent weather using qobuz or local files.
Neither limiting to 44.1k nor upsampling in Audirvana change the sample rate displayed , stuck at 48k
the output signal in the Audirvana App shows the proper Sample sample 41/96/192 etc..

However, when using Hqplayer. I can successfully Change sample rates.

I am starting to think, it maybe an MPD or GMrender issue.  I do NOT have the ALLO gui. Is their a specific string I can input into my dietpi from putty.

(Currently on Bookworm)

KenMoreira, 2024-03-21:
looking to view / editing those audio render settings. to allow maximum / whatever the file played is natively

MichaIng, 2024-04-04:
You are accessing the audio device via MPD? In the past we did, but do not hardcode the sample rate (or other audio format parameters) in `/etc/mpd.conf` anymore. The stream is just forwarded to the default ALSA PCM as received from the player, and that one, as defined in `/etc/asound.conf`, does not hardcode the format either.

But the MPD audio format can be changed (or reset) via `dietpi-justboom`. Probably you did it that way?

KenMoreira, 2024-04-04:
I've only updated using dietpi - update.

It's really the only ssh line I execute.

I'm just looking to see in my settings. I checked through comfig but since
I'm using usb dac I get
No alsa audio error or something when looking into mpd

MichaIng, 2024-04-04:
Not sure then. At some point, Allo stopped building these ASIX Ethernet driver for the USBridge Sig, hence now the upstream driver is used which can cause flickers in audio playback: https://github.com/raspberrypi/linux/issues/2942

But the audio driver itself AFAIK has no changed. Can you show your MPD and ALSA config?
```sh
cat /etc/asound.conf
cat /etc/mpd.conf
```

GMediaRender is not related to MPD, but accesses audio devices directly in its own way. `/etc/default/gmediarender` defines the used ALSA PCM, which is just `default` by default, as defined in `/etc/asound.conf`.

> the sample rate displayed , stuck at 48k
the output signal in the Audirvana App shows the proper Sample sample 41/96/192 etc..

Where do you see this 48k displayed? So you mean you can define the output signal in players, but at the server/hardware side you see 48k, when doing playback via GMediaRender or MPD, and only higher sample rates when using HQPlayer?

And does it work to enforce a sample rate via e.g.
```sh
speaker-test -r 192000
```

KenMoreira, 2024-04-04:
> can cause flickers in audio playback: raspberrypi/linux#2942

Correct.  I have a spring 3 kte
It has an on dac display.

Using mconnect or audirvana, any sample rate just shows as 48k.  Weather
it's 44.1 or up to 192k

But when using hqplayer the issue goes away as I up sample to dsd 512x
I could see if hqplayer pcm up sampling has issues.

KenMoreira, 2024-04-05:
root@DietPi:~# cat /etc/asound.conf
cat: /etc/asound.conf: No such file or directory
root@DietPi:~# cat /etc/mpd.conf
# An example configuration file for MPD.
# Read the user manual for documentation: http://www.musicpd.org/doc/user/
# or /usr/share/doc/mpd/html/user.html

# Files and directories #######################################################
#
# This setting controls the top directory which MPD will search to discover the
# available audio files and add them to the daemon's online database. This
# setting defaults to the XDG directory, otherwise the music directory will be
# be disabled and audio files will only be accepted over ipc socket (using
# file:// protocol) or streaming files over an accepted protocol.
#
music_directory "/mnt/dietpi_userdata/Music"
#
# This setting sets the MPD internal playlist directory. The purpose of this
# directory is storage for playlists created by MPD. The server will use
# playlist files not created by the server but only if they are in the MPD
# format. This setting defaults to playlist saving being disabled.
#
playlist_directory "/mnt/dietpi_userdata/Music"
#
# This setting sets the location of the MPD database. This file is used to
# load the database at server start up and store the database while the
# server is not up. This setting defaults to disabled which will allow
# MPD to accept files over ipc socket (using file:// protocol) or streaming
# files over an accepted protocol.
#
db_file "/mnt/dietpi_userdata/.mpd_cache/db_file"
#
# These settings are the locations for the daemon log files for the daemon.
# These logs are great for troubleshooting, depending on your log_level
# settings.
#
# The special value "syslog" makes MPD use the local syslog daemon. This
# setting defaults to logging to syslog, or to journal if mpd was started as
# a systemd service.
#
#log_file "/var/log/mpd/mpd.log"
#
# This setting sets the location of the file which stores the process ID
# for use of mpd --kill and some init scripts. This setting is disabled by
# default and the pid file will not be stored.
#
#pid_file "/run/mpd/pid"
#
# This setting sets the location of the file which contains information about
# most variables to get MPD back into the same general shape it was in before
# it was brought down. This setting is disabled by default and the server
# state will be reset on server start up.
#
state_file "/mnt/dietpi_userdata/.mpd_cache/state"
#
# The location of the sticker database.  This is a database which
# manages dynamic information attached to songs.
#
sticker_file "/mnt/dietpi_userdata/.mpd_cache/sticker.sql"
#
###############################################################################

# General music daemon options ################################################
#
# This setting specifies the user that MPD will run as. MPD should never run as
# root and you may use this setting to make MPD change its user ID after
# initialization. This setting is disabled by default and MPD is run as the
# current user.
#
#user "mpd"
[... 327 lines cut ...]
#
#replaygain_limit               "yes"
#
# This setting enables on-the-fly normalization volume adjustment. This will
# result in the volume of all playing audio to be adjusted so the output has
# equal "loudness". This setting is disabled by default.
#
#volume_normalization           "no"
#
###############################################################################

# Character Encoding ##########################################################
#
# If file or directory names do not display correctly for your locale then you
# may need to modify this setting.
#
filesystem_charset              "UTF-8"
#
###############################################################################
# Simple ALSA output
audio_output {
        type "alsa"
        name "DietPi ALSA"
}
root@DietPi:~#

KenMoreira, 2024-04-05:
the forcing to 192 did NOT work. shows 48k

MichaIng, 2024-04-06:
```console
root@DietPi:~# cat /etc/asound.conf
cat: /etc/asound.conf: No such file or directory
```
It seems there has been no (default) sound card selected. Please try to select it in `dietpi-config` > Audio Options > Sound card.

ALSA else uses internal defaults, as shown in `aplay -L` (`default` and `sysdefault`), but those might imply auto-conversion and stuff like that. Selecting a sound card via `dietpi-config` (and leaving "auto-conversion" "off") creates a `/etc/asound.conf` which assures plain input stream playback without any ALSA plugin manipulation. If would then throw an error if the audio format is not supported by the sound card/driver, instead of trying (or doing OOTB) any conversion.

KenMoreira, 2024-04-06:
Does it matter which one I choose, I'm on an allo usbridge signature.

MichaIng, 2024-04-06:
It matters in how far ALSA plugins are used, and e.g. it is possible that currently the RPi onboard audio is enabled and used as default ALSA PCM. The USBridge Sig audio device should show up at the bottom of the list in the "Auto detection" section, though I am not sure how exactly it is named.

KenMoreira, 2024-04-06:
Thanks, I'll check it out.

KenMoreira, 2024-04-06:
I selected the last one in the list and Enabled ASLA

KenMoreira, 2024-04-06:
[image]

KenMoreira, 2024-04-06:
Hw: selection sends me In a loop where it removes stuff then I get a pop up asking to install asla features, then I accept and I get back here.

MichaIng, 2024-04-06:
`hw:0,0` is it.

> Hw: (enhanc USB aAudio) selection sends me In a loop where it removes stuff then I get a pop up asking to install asla features, then I accept and I get back here.

Weird, can you copy&paste the full output of the log, after exiting `dietpi-config`?

KenMoreira, 2024-04-06:
I already closed it. What steps would you need me to perform

MichaIng, 2024-04-06:
1. Open `dietpi-config` audio options
2. Select `hw:0,0` as sound card
3. Regardless what happens, go back and exit `dietpi-config`
4. Select for copy&paste here all output done by `dietpi-config`

KenMoreira, 2024-04-07:
root@DietPi:~# dietpi-config
[ SUB1 ] DietPi-Set_hardware > soundcard (hw:0,0)
[ INFO ] DietPi-Set_hardware | Checking for required APT packages: alsa-utils
[ INFO ] DietPi-Set_hardware | Resetting all sound card settings...
Found hardware: "USB-Audio" "USB Mixer" "USB152a:87c0" "" ""
Hardware is initialized using a generic method
[  OK  ] DietPi-Set_hardware | sed --follow-symlinks -Ei -e /^[[:blank:]]*(dtove                                                                                                                                                             rlay=(allo-|applepi-dac|dionaudio-|googlevoicehat-soundcard|hifiberry-|i-sabre-q                                                                                                                                                             2m|iqaudio-|justboom-|rpi-dac)|dtparam=i2s)/d -e s/^[[:blank:]]*(hdmi_drive(:[01                                                                                                                                                             ])?=.*$)/#\1/ /boot/config.txt
[  OK  ] DietPi-Set_hardware | Desired setting in /boot/config.txt was already s                                                                                                                                                             et: dtparam=audio=off
[  OK  ] DietPi-Set_hardware | eval echo 'blacklist snd_bcm2835' > /etc/modprobe                                                                                                                                                             .d/dietpi-disable_rpi_audio.conf
[  OK  ] DietPi-Set_hardware | sed --follow-symlinks -Ei /root=/s/[[:blank:]]*sn                                                                                                                                                             d_bcm2835.enable_[^[:blank:]]*([[:blank:]]*$)?//g /boot/cmdline.txt
[ INFO ] DietPi-Set_hardware | Applying new sound card settings...
[  OK  ] DietPi-Set_hardware | Setting in /boot/dietpi.txt adjusted: CONFIG_SOUN                                                                                                                                                             DCARD=hw:0,0
[  OK  ] DietPi-Set_hardware | alsactl -g store
[  OK  ] soundcard hw:0,0 | Completed
[ SUB1 ] DietPi-Set_hardware > soundcard (none)
[ INFO ] DietPi-Set_hardware | APT purge alsa-utils libasound2-plugin-equal firm                                                                                                                                                             ware-intel-sound firmware-sof-signed, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
The following packages will be REMOVED:
  alsa-utils* libatopology2* libfftw3-single3*
0 upgraded, 0 newly installed, 3 to remove and 0 not upgraded.
                                                              After this operation, 4788 kB disk space will be freed.
(Reading database ... 35165 files and directories currently installed.)                                              (Reading database ...
Removing alsa-utils (1.2.8-1+rpt1) ...
Removing libatopology2:arm64 (1.2.8-1+rpt1) ...
Removing libfftw3-single3:arm64 (3.3.10-1) ...
Processing triggers for libc-bin (2.36-9+rpt2+deb12u4) ...
(Reading database ... 35067 files and directories currently installed.)
Purging configuration files for alsa-utils (1.2.8-1+rpt1) ...
[  OK  ] DietPi-Set_hardware | APT purge alsa-utils libasound2-plugin-equal firmware-intel-sound firmware-sof-signed
[ INFO ] DietPi-Set_hardware | Resetting all sound card settings...
[  OK  ] DietPi-Set_hardware | rm /etc/asound.conf
[  OK  ] DietPi-Set_hardware | sed --follow-symlinks -Ei -e /^[[:blank:]]*(dtoverlay=(allo-|applepi-dac|dionaudio-|googlevoicehat-soundcard|hifiberry-|i-sabre-q2m|iqaudio-|justboom-|rpi-dac)|dtparam=i2s)/d -e s/^[[:blank:]]*(hdmi_drive(:[01])?=.*$)/#\1/ /boot/config.txt
[  OK  ] DietPi-Set_hardware | Desired setting in /boot/config.txt was already set: dtparam=audio=off
[  OK  ] DietPi-Set_hardware | eval echo 'blacklist snd_bcm2835' > /etc/modprobe.d/dietpi-disable_rpi_audio.conf
[  OK  ] DietPi-Set_hardware | sed --follow-symlinks -Ei /root=/s/[[:blank:]]*snd_bcm2835.enable_[^[:blank:]]*([[:blank:]]*$)?//g /boot/cmdline.txt
[ INFO ] DietPi-Set_hardware | Applying new sound card settings...
[  OK  ] DietPi-Set_hardware | Setting in /boot/dietpi.txt adjusted: CONFIG_SOUNDCARD=none
[  OK  ] soundcard none | Completed
[  OK  ] DietPi-Config | Setting in /boot/dietpi/.installed adjusted: aSOFTWARE_INSTALL_STATE[5]=0
[ INFO ] DietPi-Config | Checking for required APT packages: alsa-utils
[ INFO ] DietPi-Config | APT update, please wait...
Hit:1 https://deb.debian.org/debian bookworm InRelease
Get:2 https://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Get:3 https://deb.debian.org/debian-security bookworm-security InRelease [48.0 kB]
Get:4 https://deb.debian.org/debian bookworm-backports InRelease [56.5 kB]
Hit:5 https://archive.raspberrypi.com/debian bookworm InRelease
Fetched 160 kB in 4s (44.9 kB/s)
Reading package lists...
[  OK  ] DietPi-Config | APT update
[ INFO ] DietPi-Config | APT install alsa-utils, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  libatopology2 libfftw3-single3
Suggested packages:
  dialog libfftw3-bin libfftw3-dev
[... cut ...]
[  OK  ] DietPi-Set_hardware | Desired setting in /boot/config.txt was already set: dtparam=audio=off
[  OK  ] DietPi-Set_hardware | eval echo 'blacklist snd_bcm2835' > /etc/modprobe.d/dietpi-disable_rpi_audio.conf
[  OK  ] DietPi-Set_hardware | sed --follow-symlinks -Ei /root=/s/[[:blank:]]*snd_bcm2835.enable_[^[:blank:]]*([[:blank:]]*$)?//g /boot/cmdline.txt
[ INFO ] DietPi-Set_hardware | Applying new sound card settings...
[  OK  ] DietPi-Set_hardware | Desired setting in /boot/dietpi.txt was already set: CONFIG_SOUNDCARD=none
[  OK  ] soundcard none | Completed

KenMoreira, 2024-04-07:
I eventually cycled through again. choose HW. and after it installed and asked to Me select a sound card. this time instead of saying None as usual. I still said HW so i chose exit. and then typed reboot.

KenMoreira, 2024-04-07:
(Deleted the log as it was no longer needed for you to go through)

MichaIng, 2024-04-07:
Hmm, there should not be a "none" option in the sound card menu. The only way to trigger this
```
[ SUB1 ] DietPi-Set_hardware > soundcard (none)
```
is to select "Disable" in the main audio menu. So I guess you selected "Ok", while "Disable" was highlighted, expecting this means to confirm/finish audio options, while instead the "Ok" button confirms the selection of the currently highlighted menu entry. "Back" or "Exit", i.e. the buttons on the right side, or hitting ESC key, would be the way to go.

A case where indeed the button text is misleading. In those whiptail menus, the button on the left side is equal to hitting enter/return. We really need to adjust the button texts to better represent what they actually mean. In case of those menus, where you can navigate through the menu entries and hit enter/return, while you can also select the left button instead, it would be actually best to just hide the button. Sadly only the cancel button can be hidden.

However, glad it was sorted. So this means you can now effectively play audio with all sample rates through the USBridge Sig?

KenMoreira, 2024-04-07:
Thanks for the explanation.
Yes it's working as expected with audirvana and with m connect streamer
software

MichaIng, 2024-04-07:
Okay great. I am happy that the driver is not broken 🙂. I'll mark the issue as closed then. Feel free to reopen if required.

closed 2024-04-07 by MichaIng