MichaIng/DietPi #6979: 7inch Capacitive Touch Screen LCD (C), 1024×600, HDMI, Does not work with DietPi v9.2.1

learncodingforweb, 2024-03-22:
I have waveshare 7inch Capacitive Touch Screen LCD (C), 1024×600, HDMI, IPS, Low Power TFT Lcd, which works fine on rasberry pi **Debian Bookworm OS 64 bit **after adding following lines into config.txt,

```
hdmi_group=2
hdmi_mode=87
hdmi_cvt 1024 600 60 6 0 0 0
hdmi_drive=1
```
But When i install DietPi v9.2 and enable desktop LXDE, Mate. on booting rasberry pi display is shown in small part. not on full screen. how to make dietpi takes full screen as per attached TFT Lcd 7 inch display

learncodingforweb, 2024-03-29:
resolved option. Step i followed
1) Install Dietpi
2) open sd card and edited config.txt by adding following
```
hdmi_group=2
hdmi_mode=87
hdmi_cvt 1024 600 60 6 0 0 0
hdmi_drive=1
```

closed 2024-03-29 by learncodingforweb

MichaIng, 2024-03-29:
Jep, DietPi an RPi OS are fully config compatible. Just note that these `hdmi_*` settings work only with the legacy graphics driver, not with KMS enabled. So I do actually wonder that this works OOTB on RPi OS Bookworm, where KMS is enabled OOTB. Did you remove the `dtoverlay=vc4-kms-v3d` line there?

With KMS enabled, display resolution and rotation needs to be applied a differently: https://www.raspberrypi.com/documentation/computers/configuration.html#command-line-options

learncodingforweb, 2024-03-30:
I have not changed anything other than config.txt by adding 
```
hdmi_group=2
hdmi_mode=87
hdmi_cvt 1024 600 60 6 0 0 0
hdmi_drive=1
```

MichaIng, 2024-04-13:
Interesting: On RPi OS Bookworm these options are all legacy and should not have any effect anymore, at least unless you manually revert to legacy framebuffer graphics by removing `dtoverlay=vc4-kms-v3d` from your `config.txt`. Is this line really still present?

Furthermore, as you want to set the resolution for your desktop (?): These have their own display and resolution settings, while the `hdmi_*` and `cmdline.txt` KMS settings are for console mode resolution only. Of course they are the initial resolution for X sessions as well, but the X server and/or desktop environment can and usually do change it, at least when you used their display settings once 🤔.

However, good that you found a solution that works for you. I'll need to play with all of this anyway, to migrate our `dietpi-config` resolution options for new RPi kernel/firmware stack.