MichaIng/DietPi #6980: Unable to set mixed locale

nplauritzen, 2024-03-22:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version:
- 
```
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=1
G_DIETPI_VERSION_RC=1
```

- Distro version: `bookworm 0`
- Kernel version: `Linux pi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- SBC model: `RPi 4 Model B (aarch64)`

#### Steps to reproduce
- Set language to `en_US.UTF-8`
- Generate additional locales
- Define LC variables from existing locales in `/etc/default/locale`

#### Expected behaviour
- `LC` environment variables are set and honored (e.g. `LC_TIME`for `date`).

#### Actual behaviour
- All formats are still set to `en_US.UTF-8`

#### Extra details

I am currently unable to define a mixed locale setting on dietpi. 
On my systems I prefer setting system language to `en_US. UTF-8` whereas formats remain in local format (`de_DE.UTF-8`). Locale in `dietpi-config` is set to `en_US. UTF-8`. I added `de_DE.UTF-8` as additional locales to `/etc/Locale. gen` and generated those. 
After that, I tried setting preferred locales in `/etc/default/locale`:
```
LANG=en_US.UTF-8
LC_CTYPE=en_US.UTF-8
LC_NUMERIC=de_DE.UTF-8
LC_TIME=de_DE.UTF-8
LC_COLLATE=en_US.UTF-8
LC_MONETARY=de_DE.UTF-8
LC_MESSAGES=en_US.UTF-8
LC_PAPER=de_DE.UTF-8
LC_NAME=en_US.UTF-8
LC_ADDRESS=de_DE.UTF-8
LC_TELEPHONE=de_DE.UTF-8
LC_MEASUREMENT=de_DE.UTF-8
LC_IDENTIFICATION=en_US.UTF-8
LC_ALL=
```

Nevertheless, all formats are still set to `en_US.UTF-8`. Showing all environment variables lists only `LANG` - all other defined `LC` variables are missing and are not applied. Is there something I am missing or why is not possible?

MichaIng, 2024-04-04:
Check whether the German locale really has been compiled:
```sh
locale -a
```
In case select and generate them via:
```sh
dpkg-reconfigure locales
```
While of course via `/etc/locale.gen` and `locale-gen` or `localedef` it works as well.

And assure to logout and login, `exec bash` or `reboot` to have the new `/etc/default/locale` loaded into the environment. It would also throw errors if the defined locale(s) were not available.

nplauritzen, 2024-05-04:
Hi, thanks for clarification! Unfortunately the additional locales are somehow still not respected. Maybe this is a quirk related to my setup or setting initially via the setup variables (`dietpi.txt`). I reconfigured to use all locales as `de_DE.UTF-8` and this is working fine.

closed 2024-05-04 by nplauritzen

MichaIng, 2024-05-04:
If `locale -a` shows them (hence they have been compiled) and `/etc/default/locale` sets the variables as you intend, it should work on any new shell session. Also, if locales defined in the latter are missing, the shell should throw errors.