MichaIng/DietPi #6982: Docker compose cannot be installed

boterocamilo, 2024-03-23:
root@DietPi:~# dietpi-software
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for conflicts and missing inputs

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for prerequisite software

[ INFO ] DietPi-Software | Docker will be reinstalled
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /var/www /opt /usr/local/bin
[  OK  ] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[  OK  ] DietPi-Software | chmod 0775 /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[ SUB1 ] DietPi-Services > stop 
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : docker
[  OK  ] DietPi-Services | stop : prowlarr
[  OK  ] DietPi-Services | stop : bazarr
[  OK  ] DietPi-Services | stop : radarr
[  OK  ] DietPi-Services | stop : sonarr
[  OK  ] DietPi-Services | stop : jellyfin
[ INFO ] DietPi-Software | APT update, please wait...
Hit:1 https://deb.debian.org/debian bookworm InRelease
Hit:2 https://deb.debian.org/debian bookworm-updates InRelease
Hit:3 https://apt.sonarr.tv/debian buster InRelease
Hit:4 https://deb.debian.org/debian-security bookworm-security InRelease
Hit:5 https://deb.debian.org/debian bookworm-backports InRelease
Hit:6 https://download.mono-project.com/repo/debian buster InRelease
Hit:7 https://archive.raspberrypi.com/debian bookworm InRelease
Ign:8 https://download.docker.com/linux/debian \ InRelease
Hit:9 https://download.docker.com/linux/debian bookworm InRelease
Hit:10 https://repo.jellyfin.org/debian bookworm InRelease
Err:11 https://download.docker.com/linux/debian \ Release
  404  Not Found [IP: 13.33.141.123 443]
Reading package lists...
E: The repository 'https://download.docker.com/linux/debian \ Release' does not have a Release file.
[FAILED] DietPi-Software | APT update
 - Command: apt-get -y -eany update
[ INFO ] DietPi-BugReport | Generating informative command outputs, please wait...
[ INFO ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT, please wait...
[  OK  ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT
[  OK  ] DietPi-BugReport | Packing upload archive
[  OK  ] DietPi-BugReport | Sending bug report
[  OK  ] DietPi-BugReport | Your bug report has been successfully uploaded.
- Reference code: 1568fffc-d7e8-4414-b734-9ede76bcaa07

Please file a related bug report at GitHub or our forum:
- https://github.com/MichaIng/DietPi/issues
- https://dietpi.com/forum/c/troubleshooting/10

The uploaded file will be removed automatically after 48 hours or when your issue has been solved. Additionally you can remove it by running:
- dietpi-bugreport -1

Press any key to continue...
[ INFO ] DietPi-BugReport | Generating informative command outputs, please wait...
[ INFO ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT, please wait...
[  OK  ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT
[  OK  ] DietPi-BugReport | Packing upload archive
[  OK  ] DietPi-BugReport | Sending bug report
[  OK  ] DietPi-BugReport | Your bug report has been successfully uploaded.
- Reference code: 1568fffc-d7e8-4414-b734-9ede76bcaa07

Please file a related bug report at GitHub or our forum:
- https://github.com/MichaIng/DietPi/issues
- https://dietpi.com/forum/c/troubleshooting/10

The uploaded file will be removed automatically after 48 hours or when your issue has been solved. Additionally you can remove it by running:
- dietpi-bugreport -1

Press any key to continue...
[ INFO ] DietPi-Software | APT update, please wait...
Hit:1 https://deb.debian.org/debian bookworm InRelease
Hit:2 https://deb.debian.org/debian bookworm-updates InRelease
Hit:3 https://deb.debian.org/debian-security bookworm-security InRelease
Hit:4 https://deb.debian.org/debian bookworm-backports InRelease
Hit:5 https://apt.sonarr.tv/debian buster InRelease
Hit:6 https://download.mono-project.com/repo/debian buster InRelease
Hit:7 https://archive.raspberrypi.com/debian bookworm InRelease
Ign:8 https://download.docker.com/linux/debian \ InRelease
Hit:9 https://download.docker.com/linux/debian bookworm InRelease
Err:10 https://download.docker.com/linux/debian \ Release
  404  Not Found [IP: 13.33.141.35 443]
Hit:11 https://repo.jellyfin.org/debian bookworm InRelease
Reading package lists...
E: The repository 'https://download.docker.com/linux/debian \ Release' does not have a Release file.
[FAILED] DietPi-Software | APT update
 - Command: apt-get -y -eany update

RPi 4 Model B (aarch64) | IP: 192.168.0.161 | Use up/down buttons to scroll text
│ Hit:1 https://deb.debian.org/debian bookworm InRelease
│ Hit:2 https://deb.debian.org/debian bookworm-updates InRelease
│ Hit:3 https://deb.debian.org/debian-security bookworm-security InRelease
│ Hit:4 https://deb.debian.org/debian bookworm-backports InRelease
│ Hit:5 https://apt.sonarr.tv/debian buster InRelease
│ Hit:6 https://download.mono-project.com/repo/debian buster InRelease
│ Hit:7 https://archive.raspberrypi.com/debian bookworm InRelease
│ Ign:8 https://download.docker.com/linux/debian \ InRelease
│ Hit:9 https://download.docker.com/linux/debian bookworm InRelease
│ Err:10 https://download.docker.com/linux/debian \ Release
│   404  Not Found [IP: 13.33.141.35 443]
│ Hit:11 https://repo.jellyfin.org/debian bookworm InRelease
│ Reading package lists...
│ E: The repository 'https://download.docker.com/linux/debian \ Release' does
│ not have a Release file.

boterocamilo, 2024-03-23:
Here the BLUE lines when encountered the error:
### Details:
- Date           | Sat Mar 23 15:35:49 GMT 2024
- DietPi version | v9.2.1 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 https://deb.debian.org/debian bookworm InRelease
Hit:2 https://deb.debian.org/debian bookworm-updates InRelease
Hit:3 https://deb.debian.org/debian-security bookworm-security InRelease
Hit:4 https://deb.debian.org/debian bookworm-backports InRelease
Hit:5 https://download.mono-project.com/repo/debian buster InRelease
Hit:6 https://apt.sonarr.tv/debian buster InRelease
Hit:7 https://archive.raspberrypi.com/debian bookworm InRelease
Ign:8 https://download.docker.com/linux/debian \ InRelease
Hit:9 https://repo.jellyfin.org/debian bookworm InRelease
Err:10 https://download.docker.com/linux/debian \ Release
  404  Not Found [IP: 13.33.141.123 443]
Reading package lists...
E: The repository 'https://download.docker.com/linux/debian \ Release' does not have a Release file.
```

Joulinar, 2024-03-23:
Maybe a temporary issue with the Docker own repository. Give it some time and try again later.

MichaIng, 2024-03-23:
The distro/suite seems to be missing in your Docker repo list file, or it is `\` instead of e.g. `bookwom`.

In your OP there is another functional `https://download.docker.com/linux/debian bookworm`, in the 2nd post this one is somehow missing 🤔. Check the files in `/etc/apt/sources.list.d`. There should be only one about Docker with
```
deb https://download.docker.com/linux/debian/ bookworm stable
```

boterocamilo, 2024-03-24:
> The distro/suite seems to be missing in your Docker repo list file, or it is `\` instead of e.g. `bookwom`.
>
> In your OP there is another functional `https://download.docker.com/linux/debian [...]

This is what I get:

root@DietPi:/etc/apt/sources.list.d# ls
archive_uri-https_download_docker_com_linux_debian-bookworm.list  raspi.list
root@DietPi:/etc/apt/sources.list.d# cat archive_uri-https_download_docker_com_linux_debian-bookworm.list
deb [arch=amd64] https://download.docker.com/linux/debian \ \ bookworm stable
# deb-src [arch=amd64] https://download.docker.com/linux/debian \ \ bookworm stable

boterocamilo, 2024-03-24:
Ok, what I did was proceed to delete the file running:
sudo rm archive_uri-https_download_docker_com_linux_debian-bookworm.list

Now, I did the installation again (using dietpi-software), and now it works.

Thanks for you help @MichaIng

closed 2024-03-24 by boterocamilo

Joulinar, 2024-03-24:
Did you add that list manually ?