MichaIng/DietPi #6983: Can I install a new dietpi on the old dietpi system?

xczh, 2024-03-23:
I am currently using the dietpi v6 system, which is too old to automatically update. I plan to install the v9 system, but I don't have an SD card writer. Currently, I'm able to access the existing dietpi v6 system through SSH. Can I use SSH to flash the v9 system image? For example, using `dd`. 

Thanks a lot :)

Joulinar, 2024-03-23:
Yes you can use `dd` to write the image

xczh, 2024-03-23:
> Yes you can use `dd` to write the image

Thank you for your reply 😄 

Can you give an example? Will using `dd` on a running dietpi system cause malfunctions?

```
root@DietPi:~# df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        59G  1.8G   55G   4% /
devtmpfs        455M     0  455M   0% /dev
tmpfs           488M     0  488M   0% /dev/shm
tmpfs           488M   13M  475M   3% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           488M     0  488M   0% /sys/fs/cgroup
tmpfs            50M     0   50M   0% /var/log
tmpfs           1.0G  4.0K  1.0G   1% /tmp
/dev/mmcblk0p1  253M   47M  206M  19% /boot
```

It seems that the SD card device is `/dev/mmcblk0`.

Joulinar, 2024-03-23:
You can't write to the very same SD card. It would require a 2nd one or a USB pen stick if your system supports booting from USB.

xczh, 2024-03-23:
Understood. I'm using Raspberry. There is no separate SD card reader left. It seems that we have come up with other ways.

closed 2024-03-23 by xczh

Joulinar, 2024-03-23:
Theoretically we could try to update current system. Depending on RPi model,  you could use USB stick

MichaIng, 2024-03-23:
You can also run the `dietpi-installer` on a running DietPi system, which will end up in a fresh system will all data purged. But there is a branch you can still use to update. I'll provide you with the information in some minutes.

MichaIng, 2024-03-23:
See here: https://github.com/MichaIng/DietPi/issues/6963#issuecomment-2000428747