MichaIng/DietPi #6985: Can’t connect a bluetooth headset

johackim, 2024-03-24:
### Creating a bug report/issue

[x] I have searched the existing open and closed issues

#### Required Information
- DietPi version:

```txt
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=2
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```

- Distro version: `bookworm 0`
- Kernel version: `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Architecture: `arm64`
- SBC model: `RPi 3 Model B (aarch64)`
- SD card used: `SanDisk ultra`

#### Steps to reproduce

1. `dietpi-config`
2. `Advanced Options`
3. `Bluetooth: [On]`
4. `bluetoothctl scan on`
5. `bluetoothctl pair DEVICE_ID`
6. `bluetoothctl trust DEVICE_ID`
7. `bluetoothctl connect DEVICE_ID`

#### Expected behaviour

- Successfully connecting my bluetooth headset (Bose QC45)

#### Actual behaviour

- I can't connect my bluetooth headset (Bose QC45), I receive this bluetoothctl error: `Failed to connect: org.bluez.Error.Failed br-connection-profile-unavailable`

MichaIng, 2024-04-04:
See here: #3456

Try this:
```sh
apt install bluez-alsa-utils
aplay -l
speaker-test -D bluealsa
```
I am not sure whether the `bluealsa` service does/tries binding to BT audio devices automatically or just allows manual binding, or is not related to this at all. Also probably a reboot is required, or unmuting the volume:
```sh
alsamixer -D bluealsa
```
When this BT audio PCM generally works, you can make it the default:
```sh
echo -e 'pcm.!default bluealsa\nctl.!default bluealsa' > /etc/asound.conf
```

johackim, 2024-04-30:
Thank you very much @MichaIng!

It works 😀!

However, I have to connect my Bluetooth headset before running Kodi, or exit Kodi to reconnect my Bluetooth headset if the connection is lost.

I can't autostart Kodi.

MichaIng, 2024-05-01:
Okay, so pairing still needs to be done manually. Somehow makes sense. I have not much experience with bluez, but wonder whether there is a native way to have it connecting to a particular device automatically (given that the device accepts it). Else, you could create a script for that and place it into `/var/lib/dietpi/postboot.d` to be executed automatically on boot.

Hoempi, 2024-08-19:
If I may, I'd like to piggyback on this Thread. I'm also trying to connect a bluetooth audio device, in my case an Alexa smart speaker. My end goal is to install plexamp-headless via docker and send the audio output via bluetooth to the Alexa.

But I always get an error when trying to connect using `bluetoothctl`: "Failed to connect: org.bluez.Error.Failed". But I also did not succeed in installing `bluez-alsa-utils` as apt does not find any package by that name.

I'm running DietPi v9.6.1 on a RPi 3 Model B (aarch64)

MichaIng, 2024-10-18:
@Hoempi sorry for the late reply. You are running a Bullseye system? Because that APT package is available since Bookworm only. And it is expectedly requires for Bluetooth audio to work.

Hoempi, 2024-10-21:
Hi @MichaIng ,

don't sweat it, you're probably providing assistance on your personal free time, so whenever you find time to do so, it's early enough. Similar to wizards always arriving precisely when they want to ;-)

Yes, you are right, my install is still a Bullseye installation. The installation is a couple of years old. Is there an easy way to switch through dietpi-update or something similar?

Joulinar, 2024-10-21:
Have a look to our blog post if you like to switch to Bookworm https://dietpi.com/blog/?p=3128

MichaIng, 2025-05-31:
Looping through old issues: Did it work after migrating to Bookworm?

MichaIng, 2025-06-06:
That might be expected if Kodi hogs the default audio device or scans for available sound devices at start only, failing if the previously chosen one is not available.

You can connect automatically at boot e.g. like this:
```sh
echo 'bluetoothctl connect <MAC>' > /var/lib/dietpi/postboot.d/bt-speakers.sh
```
I am actually surprised that `bluez` does not have a native config to connect to devices automatically at boot 🤔.