MichaIng/DietPi #6987: Neo3 not rebooting

murraythegoz, 2024-03-26:
### Creating a bug report/issue
- [ X] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
  G_DIETPI_VERSION_CORE=9
  G_DIETPI_VERSION_SUB=2
  G_DIETPI_VERSION_RC=1
  G_GITBRANCH='master'
  G_GITOWNER='MichaIng'

- Distro version | bookworm
- Kernel version | 
  Linux neo3-13.priv.bth 6.6.16-current-rockchip64 #1 SMP PREEMPT Fri Feb 23 08:25:28 UTC 2024 aarch64 GNU/Linux
- SBC model | NanoPi NEO3 (aarch64)
- Power supply used | USB-C PD
- SD card used | Samsung EVO plus 32GB

#### Additional Information (if applicable)
- Software title | none
- Was the software title installed freshly or updated/migrated? new installation
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Flash a new SDcard and insert into a NanoPi Neo3
2. finish the installation script and try to reboot

#### Expected behaviour

- device should get back online

#### Actual behaviour

- device do not reboot

#### Extra details

- using a stock dietpi image. there is a chance the generic device-tree for rockchip is not working, trying to rebuild myself but better have it upstream on the standard image.

murraythegoz, 2024-03-26:
root@neo3-13:/boot# cat /boot/dietpiEnv.txt 
rootdev=UUID=e0c99fd1-c0b7-4694-9dc7-c3cb3773deec
rootfstype=ext4
# The init system logs to the console defined last.
consoleargs=console=ttyS2,1500000
usbstoragequirks=
extraargs=net.ifnames=0
docker_optimizations=off
overlay_path=rockchip
overlay_prefix=rockchip
overlays=
user_overlays=
fdtfile=rockchip/rk3328-nanopi-neo3-rev02.dtb

MichaIng, 2024-03-27:
Hmm, I guess a kernel update has been applied on first boot? I can test later. Would be bad, of course, if the device tree got broken or is missing or such.

murraythegoz, 2024-03-27:
> Hmm, I guess a kernel update has been applied on first boot? I can test later. Would be bad, of course, if the device tree got broken or is missing or such.

on firstboot the kernel and tree  were updated.

Being an headless the firstboot must be done via network.
could try a reimage and a firstboot via console, but not sure I am able to.

MichaIng, 2024-04-06:
Hmm, in general it works well here, also the kernel did not receive an update since Feburary:
```console
root@NanoPiNEO3:~# uname -a
Linux NanoPiNEO3 6.6.16-current-rockchip64 #1 SMP PREEMPT Fri Feb 23 08:25:28 UTC 2024 aarch64 GNU/Linux
```
But indeed on first `reboot` it also did not come back online here, only on power cycle. But on next reboot it came back online, and on every next one. Not sure whether it was a coincidence or some systematic issue on first reboot, though no idea what would chance between first and consecutive reboots.

murraythegoz, 2024-04-08:
do you mind sharing the current device-tree which was installed?

thank you

murraythegoz, 2024-04-08:
Strange enough just retried and reboot is working, with ethernet.

Same image as yours.

Maybe the very latest kernel and device tree fixed it

Thank you

MichaIng, 2024-04-08:
As said, there was no kernel update since February, it is still `6.6.16-current-rockchip64`, using and was always using `/boot/dtb/rockchip/rk3328-nanopi-neo3-rev02.dtb`.

m7mdcc, 2024-04-09:
Confirmed, I'm facing the same issue.

MichaIng, 2024-04-28:
@murraythegoz @m7mdcc does this issue still exist, and does the new U-Boot build fix it?

murraythegoz, 2024-04-28:
right now it seems to reboot flawlessy.

uname -a
Linux neo3-15.priv.bth 6.6.16-current-rockchip64 #1 SMP PREEMPT Fri Feb 23 08:25:28 UTC 2024 aarch64 GNU/Linux

thank you

m.

MichaIng, 2024-04-28:
Okay great. But only after applying U-Boot update, right? I'll upload that version to our APT repository then.

closed 2024-05-12 by MichaIng