MichaIng/DietPi #6989: Rock 5B - Uarts

isarrider, 2024-03-27:
Hi,

on my Rock 5B I can only see 2 Uarts:
                                   │                                               ttyS6 console   : [On]                                                 │
                                   │                                               ttyFIQ0 console : [Off]                                                │
I guess thats is similar for everybody when not on a newer kernel
like here: https://forum.armbian.com/topic/27877-rs232-activation/

And it has to wait for a kernel update?

BR,
Alex

MichaIng, 2024-03-27:
Which kernel version do you currently have?
```sh
uname -a
```

isarrider, 2024-03-27:
Latest that come along:
Linux Rock5-4GB 5.10.160-legacy-rk35xx

MichaIng, 2024-03-27:
Ah right. I wanted to add builds for the new Linux 6.1 "vendor" branch Armbian generated for RK3588. But I need to add support for building those to our GitHub Actions workflow.

isarrider, 2024-03-27:
https://forum.armbian.com/topic/32108-how-to-switch-to-mainline-kernel-in-armbian-2311-on-orange-pi-5/
this guy already went to 6.7

isarrider, 2024-03-27:
https://gitlab.collabora.com/hardware-enablement/rockchip-3588/notes-for-rockchip-3588/-/blob/main/mainline-status.md

isarrider, 2024-03-27:
looking at this 6.1 might not be enough for UART...

isarrider, 2024-05-19:
This would also work here, right?
https://github.com/MichaIng/DietPi/issues/6797

isarrider, 2024-05-23:
Doing "apt install linux-{image,dtb}-vendor-rk35xx" on the Rock 5B works, but then u only have:
ttyFIQ0 console : [Off]
and nth else...
This link shows that 6.3 changes sth as there is then support...
Any plan to get the Rock 5B also to the latest kernel?

isarrider, 2024-05-23:
and it gives you btw this Kernel:
Linux Rock5-4GB 6.1.43-vendor-rk35xx #1 SMP Sat May 11 11:15:57 UTC 2024 aarch64 GNU/Linux

MichaIng, 2024-05-23:
Hmm, there are a bunch of overlays for RK3588 and UART, but not sure whether they apply in this particular case:
[image]
Can you try that: `rk3588` is the prefix, so use e.g. `uart0-m2` in the `overlays=` line (in case add it) in `/boot/dietpiEnv.txt` to enable UART0. Not sure which one is which on this SBC, and whether they really work on all RK3588 SBCs, but worth to try.

isarrider, 2024-05-23:
I saw that but it seemed so random and I was hpoing for a newer kernel where it should be in...
But let me try...

isarrider, 2024-05-23:
doenst change, still:
ttyFIQ0 console : [Off]
no matter if with or without prefix...

MichaIng, 2024-05-23:
It must be without prefix, else it cannot work. Also the file extension must be skipped. So really just "uart0-m2", respectively.
```sh
overlays=uart0-m2
```
And try that with all overlays, whether any of them works.

isarrider, 2024-05-23:
We get this (and again what learned ;)) with overlays=uart0-m2:
ttyS0 console   : [Off]
ttyFIQ0 console : [Off]

MichaIng, 2024-05-23:
So far so good. The question then is which pins that one uses, which UART you need and if it actually works 🙂.

isarrider, 2024-05-23:
actually I need more than one UART - so only a tiny step.
Like a car with one instead of zero wheel... ;)

MichaIng, 2024-05-23:
Just add more overlays, space-separated:
```sh
overlays=uart0-m2 uart1-m1
```
I read some time ago, but forgot what those `-mX` suffixes mean 🤔.

isarrider, 2024-05-23:
hmm - interesting - didnt know you can use more of them at once...
Could be a good workaround until a newer kernel is there...

MichaIng, 2024-07-10:
@isarrider 
Does it work to enable the UARTs you need? Since those overlays exist, it is expected that one must use them, a newer kernel won't change something about it (unless mainline defaults are different in this regards). But at some point I want to add a `dietpi-config` menu to toggle overlays.

isarrider, 2025-01-30:
As written in #7337 - > Uart4 is now available with overlays=uart4-m2 - I need to find a quick way to test if it really works...

isarrider, 2025-01-31:
If anybody has a suggestion how to test it quickly, I am all ears ;)
Maybe crosswire 2 uarts?

isarrider, 2025-02-17:
@MichaIng where did the testimages go?

MichaIng, 2025-02-18:
@isarrider 
I generated a new one: https://dietpi.com/downloads/images/testing/

isarrider, 2025-02-19:
nope, doenst boot for me...
Whats different from the other one?

isarrider, 2025-02-19:
OffTopic: with the 6.1. vendor Kernel, "cpu" reports 37°C instead of 42°C with (6.12), although both have no load on the cores...

MichaIng, 2025-02-19:
So it does boot? Possible that it runs at a higher (minimum) clock with mainline kernel: Maybe compare `cpu` output.

isarrider, 2025-02-19:
yap, it boots with 6.12... sry for not stating this clearly...

isarrider, 2025-03-28:
> If anybody has a suggestion how to test it quickly, I am all ears ;) Maybe crosswire 2 uarts?

https://www.elektronik-kompendium.de/sites/praxis/bauteil_usb-uart-adapter.htm
This seesm the way to go - will try...

isarrider, 2025-03-30:
or like said here:
https://docs.radxa.com/en/rock5/rock5a/getting-started/interface-usage/pin-40-test

isarrider, 2025-04-11:
For the time I will pause this as the Mainline Kernel only has a very basic DVFS...
The vendor kernel with a USB<>seems to be the easier and faster way for now...

isarrider, 2025-05-31:
I think this is the related commit:
https://github.com/torvalds/linux/commit/276856db91b46eaa7a4c19226c096a9dc899a3e9

MichaIng, 2025-05-31:
Hmm, CPU states could be added easily. The commit you linked has been merged with Linux 6.12, so those are available: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm64/boot/dts/rockchip/rk3588-opp.dtsi?h=linux-6.12.y

My NanoPi M6 running edge kernel):
```console
root@micha:~# uname -a
Linux micha.gnoedi.org 6.14.8-edge-rockchip64 #1 SMP PREEMPT Thu May 22 12:31:58 UTC 2025 aarch64 GNU/Linux
root@micha:~# cat /sys/devices/system/cpu/cpufreq/policy0/stats/time_in_state
408000 61079431
600000 57833
816000 31581
1008000 34239
1200000 81727
1416000 32163
1608000 12318
1800000 23824
root@micha:~# cat /sys/devices/system/cpu/cpufreq/policy4/stats/time_in_state
408000 61161553
600000 23461
816000 17231
1008000 12721
1200000 10406
1416000 7763
1608000 10869
1800000 5317
2016000 5194
2208000 4826
2400000 94860
root@micha:~# cat /sys/devices/system/cpu/cpufreq/policy6/stats/time_in_state
408000 61202219
600000 25184
816000 15256
1008000 9659
1200000 7447
1416000 5835
1608000 8491
1800000 4333
2016000 4292
2208000 4241
2400000 67680
```
Vendor kernel: https://github.com/armbian/linux-rockchip/blob/rk-6.1-rkr5.1/arch/arm64/boot/dts/rockchip/rk3588s.dtsi#L978
It has some smaller steps in the higher range, but with different `opp-supported-hw` property, hence I guess only of of them is available on each SBC. Does the table look different for you, with mainline kernel on ROCK 5B and/or with vendor kernel?

isarrider, 2025-06-03:
Ah, good pointer ;)
Yes, I see it the same way...
I usually "measure" this by looking at the temperature...
And in the case mainline vs vendor Kernel it was like 9°C...
Happy to test the edge branch though...
Just to see if it helps or other things in the vendor Kernel reduce the temperature by that much...

isarrider, 2025-06-03:
Which image shall I use?
Or will you generate a Mainline image with 6.15 and the patch applied?

isarrider, 2025-06-03:
Ok - I even have a passive cooler on the CPU, which make 9°C even more "more"...
I just tried the edge Kernel on my Rock5C, and it doesnt wanna come up anymore...

I have to test with the Rock 5B in the coming days...

MichaIng, 2025-06-03:
Oh, let me see if edge has ROCK 5C device tree, or whether it has been renamed. You used vendor before on that one?

isarrider, 2025-06-05:
Actually I found an old patch that I wanted to turn in someday to mainline where I added the missing P-States ;)

[rk3588s.dtsi.txt](https://github.com/user-attachments/files/20611393/rk3588s.dtsi.txt)

MichaIng, 2025-06-05:
But after upgrading the ROCK 5A image to edge kernel, with changed `fdtfile`, it does not boot anymore? Not sure at this point, the bootloader is the same, and edge mainline Linux has native ROCK 5C support, so it should work. Would need serial console output to know why it doesn't.

Your patch matches the one handed in by the Armbian contributor, which was however not accepted yet, due to the false assumption that lowering frequency without voltage would not reduce power consumption: https://lore.kernel.org/all/a5bc7892ec9f84793db560b81cdb21bc@superkali.me/T/
Hence instead of doubling that, it makes sense to contribute to this discussion, making clear that lowering the frequency without lowering the voltage has a significant impact on power consumption, and heat emission in turn. But first you should be sure that this is really the case, and compare power consumption or (idle) temperature with the same kernel, at best edge/mainline with the patch applied (already the case with our build), then limiting scheduling range as described. So you rule out that other factors/differences between the kernel builds are causing the temperature difference, while the frequency might not have any impact at all.

If edge kernel does not boot, then you can do the same with vendor kernel, just raising minimal scheduling frequency to 1 GHz and compare before/after.

isarrider, 2025-06-05:
> But after upgrading the ROCK 5A image to edge kernel, with changed `fdtfile`, it does not boot anymore?

Correct - does it overwrite th fdtfile somehow?

Not sure at this point, the bootloader is the same, and edge mainline Linux has native ROCK 5C support, so it should work. Would need serial console output to know why it doesn't.
> Your patch matches the one handed in by the Armbian contributor, which was however not accepted yet, due to the false assumption that lowering frequency without voltage would not reduce power [...]

Totally agreed... ;)

> 
> If edge kernel does not boot, then you can do the same with vendor kernel, just raising minimal scheduling frequency to 1 GHz and compare before/after.

Or test it on the Rock5B, but lets see when I find the time...

isarrider, 2025-06-11:
Interestingly this works:
https://dietpi.com/downloads/images/testing/DietPi_ROCK5A-ARMv8-Bookworm.img.xz

MichaIng, 2025-06-11:
The testing image has a newer bootloader build, though the old one did work for you earlier, so no idea what changed. Kernel is identical, aside of two more added device trees in the newer one.

I created this image for testing implementation of new boot script only. This also means, it is superior anyway: It does not use the legacy `uInitrd` anymore, but the boot script uses `initrd.img` directly. The boot script needed to be adjusted for this to work. So one less step when updating the initramfs, and one less file to deal with and cleanup.

I'll move that image to stable place then. Good to have these distributed ASAP.

MichaIng, 2025-06-11:
You can just install the edge kernel on any image as posted further above.

isarrider, 2025-06-11:
I always connect via SSH, and it looks like it doenst come really up - the Green Ethernet LED turns of quickly...
Looks like constant reboots...
Let me check the new build then...
Is it the same like on the 5A image?

MichaIng, 2025-06-12:
Only the Bookworm one is new. Check timestamps 😅.