MichaIng/DietPi #6990: MPD exception: Failed to read mixer for 'DietPi ALSA': no such mixer control: PCM

Sivamurugan72, 2024-03-27:
### Creating a bug report/issue
- [x ] I have searched the existing open and closed issues

#### Required Information
- DietPi version 
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=2
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version 
bullseye 1

- Kernel version 
Linux DietPi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux
 
- SBC model 
RPi Zero W (armv6l)

- Power supply used 
5V @1Amp
 
- SD card used 
Sandisk Ultra 32GB

#### Additional Information (if applicable)
- Software title 
MPD

- Was the software title installed freshly or updated/migrated?
Fresh Installation
 
- Can this issue be replicated on a fresh installation of DietPi?
Yes
 
#### Steps to reproduce
1. Install DietPi_RPi-ARMv6-Bullseye
2. Set audio card Allo Boss through dietpi-config
3. Install mpd & myMPD through dietpi-software
4. Play a sample file on myMPD. File plays but no audio. Volume slider is disabled in myMPD
5. systemctl status mpd returns this -> exception: Failed to open mixer for 'DietPi ALSA': no such mixer control: PCM

MichaIng, 2024-04-04:
Is the ALSA PCM detected?
```sh
aplay -l
```

Sivamurugan72, 2024-04-06:
root@DietPi:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: BossDAC [BossDAC], device 0: Boss DAC HiFi [Master] pcm512x-hifi-0 [Boss DAC HiFi [Master] pcm512x-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

Sivamurugan72, 2024-04-06:
root@DietPi:~# aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
hw:CARD=BossDAC,DEV=0
    BossDAC, Boss DAC HiFi [Master] pcm512x-hifi-0
    Direct hardware device without any conversions
plughw:CARD=BossDAC,DEV=0
    BossDAC, Boss DAC HiFi [Master] pcm512x-hifi-0
    Hardware device with all software conversions
sysdefault:CARD=BossDAC
    BossDAC, Boss DAC HiFi [Master] pcm512x-hifi-0
    Default Audio Device
dmix:CARD=BossDAC,DEV=0
    BossDAC, Boss DAC HiFi [Master] pcm512x-hifi-0
    Direct sample mixing device

Sivamurugan72, 2024-04-06:
root@DietPi:~# amixer
Simple mixer control 'DSP Program',0
  Capabilities: enum
  Items: 'FIR interpolation with de-emphasis' 'Low latency IIR with de-emphasis' 'High attenuation with de-emphasis' 'Fixed process flow' 'Ringing-less low latency FIR'
  Item0: 'FIR interpolation with de-emphasis'
Simple mixer control 'Analogue',0
  Capabilities: pvolume
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 1
  Mono:
  Front Left: Playback 1 [100%] [0.00dB]
  Front Right: Playback 1 [100%] [0.00dB]
Simple mixer control 'Analogue Playback Boost',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 1
  Front Left: 0 [0%] [0.00dB]
  Front Right: 0 [0%] [0.00dB]
Simple mixer control 'Auto Mute',0
  Capabilities: pswitch
  Playback channels: Front Left - Front Right
  Mono:
  Front Left: Playback [on]
  Front Right: Playback [on]
Simple mixer control 'Auto Mute Mono',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'Auto Mute Time Left',0
  Capabilities: enum
  Items: '21ms' '106ms' '213ms' '533ms' '1.07s' '2.13s' '5.33s' '10.66s'
  Item0: '21ms'
Simple mixer control 'Auto Mute Time Right',0
  Capabilities: enum
  Items: '21ms' '106ms' '213ms' '533ms' '1.07s' '2.13s' '5.33s' '10.66s'
  Item0: '21ms'
Simple mixer control 'Clock Missing Period',0
  Capabilities: enum
  Items: '1s' '2s' '3s' '4s' '5s' '6s' '7s' '8s'
  Item0: '1s'
Simple mixer control 'Deemphasis',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'Digital',0
  Capabilities: pvolume pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 207
  Mono:
  Front Left: Playback 207 [100%] [0.00dB] [on]
  Front Right: Playback 207 [100%] [0.00dB] [on]
Simple mixer control 'Max Overclock DAC',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 40
  Mono: 0 [0%]
Simple mixer control 'Max Overclock DSP',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 40
  Mono: 0 [0%]
Simple mixer control 'Max Overclock PLL',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 20
  Mono: 0 [0%]
Simple mixer control 'Volume Ramp Down Emergency Rate',0
  Capabilities: enum
  Items: '1 sample/update' '2 samples/update' '4 samples/update' 'Immediate'
  Item0: '1 sample/update'
Simple mixer control 'Volume Ramp Down Emergency Step',0
  Capabilities: enum
  Items: '4dB/step' '2dB/step' '1dB/step' '0.5dB/step'
  Item0: '4dB/step'
Simple mixer control 'Volume Ramp Down Rate',0
  Capabilities: enum
  Items: '1 sample/update' '2 samples/update' '4 samples/update' 'Immediate'
  Item0: '1 sample/update'
Simple mixer control 'Volume Ramp Down Step',0
  Capabilities: enum
  Items: '4dB/step' '2dB/step' '1dB/step' '0.5dB/step'
  Item0: '1dB/step'
Simple mixer control 'Volume Ramp Up Rate',0
  Capabilities: enum
  Items: '1 sample/update' '2 samples/update' '4 samples/update' 'Immediate'
  Item0: '1 sample/update'
Simple mixer control 'Volume Ramp Up Step',0
  Capabilities: enum
  Items: '4dB/step' '2dB/step' '1dB/step' '0.5dB/step'
  Item0: '1dB/step'

Sivamurugan72, 2024-04-06:
root@DietPi:~# amixer scontrols
Simple mixer control 'DSP Program',0
Simple mixer control 'Analogue',0
Simple mixer control 'Analogue Playback Boost',0
Simple mixer control 'Auto Mute',0
Simple mixer control 'Auto Mute Mono',0
Simple mixer control 'Auto Mute Time Left',0
Simple mixer control 'Auto Mute Time Right',0
Simple mixer control 'Clock Missing Period',0
Simple mixer control 'Deemphasis',0
Simple mixer control 'Digital',0
Simple mixer control 'Max Overclock DAC',0
Simple mixer control 'Max Overclock DSP',0
Simple mixer control 'Max Overclock PLL',0
Simple mixer control 'Volume Ramp Down Emergency Rate',0
Simple mixer control 'Volume Ramp Down Emergency Step',0
Simple mixer control 'Volume Ramp Down Rate',0
Simple mixer control 'Volume Ramp Down Step',0
Simple mixer control 'Volume Ramp Up Rate',0
Simple mixer control 'Volume Ramp Up Step',0

MichaIng, 2024-04-06:
Does it help to raise the `Analogue Playback Boost` values, e.g. via `alsamixer`?

As for the missing myMPD volume slider:
```
exception: Failed to open mixer for 'DietPi ALSA': no such mixer control: PCM
```
Indeed the default MPD ALSA control device name is "PCM", which does not exist in your case. It is strange that I never recognised issues with MPD like that, while "PCM" does not exist in most cases (? okay, on my VMs it exists), as if it usually is smart enough to take the first volume control it finds. Probably the issue here is that there is no master volume control,, but only those per-channel ones. ... Although, it is the same with the "PCM" control here:
```
Simple mixer control 'PCM',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 63
  Mono:
  Front Left: Playback 23 [37%] [0.00dB] [on]
  Front Right: Playback 23 [37%] [0.00dB] [on]`
```
In `alsamixer` it still appears as single slider. Another thing I wonder in your case is that "Analogue" and "Analogue Playback Boost" both have "Limits: 0 - 1", as if they are actually no sliders with multiple steps, but more switches to turn on/off output, respectively turn on/off additional volume boost. In case one of these controls works as sort of volume slider, you can apply it to MPD in the `audio_output` block of `/etc/mpd.conf` with an additional entry like `mixer_control NAME`.

Sivamurugan72, 2024-04-06:
Does it help to raise the Analogue Playback Boost values, e.g. via alsamixer? - No
I tried with mixer control "Digital" which is close to PCM and it works!. I can able to change the volume through slider from 0~100%. So what is the real issue? Is ALSA restructured as part of the evolution or Allo Boss dac driver issue?

MichaIng, 2024-04-06:
Ah great, I missed the "Digital" control completely. Probably it is correct here as volume control is done via DSP before the DAC.

> Is ALSA restructured as part of the evolution or Allo Boss dac driver issue?

Not that I am aware of. Probably it was never different, not sure.

Probably the latest RPi kernel has this changed: #6676

One problem with Allo in general is, that the company has shut down, and hence no one really maintains the drivers anymore. It is more luck that e.g. the Boss 2 driver/device tree overlay got fixed after it was broken, by community and the RPi kernel dev efforts.  Let's hope that there are sufficient active users of these sound devices, as they are/were pretty good, actually, from what I heared and read.

MichaIng, 2025-06-02:
Is this still an issue with latest kernel? We sadly cannot do much about it, as this would be a driver limitation.

closed 2025-06-22 by MichaIng