MichaIng/DietPi #6991: DietPi-Update failure

lrduncan, 2024-03-27:
#### Details:
- Date           | Wed Mar 27 13:18:10 EDT 2024
- Bug report     | deeea3da-2d80-4a4e-9f5d-a62d36e1f88c
- DietPi version | v9.1.1 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `cp -a DietPi-master/rootfs/. /`
- Exit code      | 1
- Software title | DietPi-Update
#### Steps to reproduce:

1. SSH into dietpi
2. run dietpi-update
#### Expected behaviour:

- Update should succeed
#### Actual behaviour:

- Update fails
#### Extra details:

- Failure occurs on this step:
- [FAILED] DietPi-Update | Installing new DietPi system files
 - Command: cp -a DietPi-master/rootfs/. /

#### Additional logs:
```
cp: cannot stat '/./var/lib/dietpi/postboot.d': Bad message
cp: cannot stat '/./var/lib/dietpi/services': Bad message
```

MichaIng, 2024-04-02:
Something was broken with this two directories, it seems, the ones extracted shortly before, which is a little strange, since the extraction itself implies that the download was successful and error-free. Are there any kernel errors?
```sh
dmesg -l 0,1,2,3
```
And please just retry the update, in case it was some random/one-time error.

lrduncan, 2024-04-02:
Seems to work now after retry today, thanks!

closed 2024-04-02 by lrduncan