MichaIng/DietPi #6992: DietPi-Software Logitech Media Server update fails

JRK0469, 2024-03-29:
#### Details:
- Date           | Fri Mar 29 11:31:12 CET 2024
- Bug report     | 39b72d2f-a406-4122-9baf-93c9a4e358d6
- DietPi version | v9.2.1 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux Pi-Homeserver 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `curl -sSfL https://downloads.slimdevices.com/nightly/logitechmediaserver_8.3.2~1702911674_arm.deb -o logitechmediaserver_8.3.2~1702911674_arm.deb`
- Exit code      | 22
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
curl: (22) The requested URL returned error: 404
```
#### Extra details

- new repository location: https://github.com/LMS-Community/lms-server-repository/blob/master/stable.xml

MichaIng, 2024-03-29:
Many thanks. We'll switch to the XML on GitHub, though the old link still works when allowing redirects (missing `-L` flag in our `curl` call), being redirected and should result in the very same LMS v8.5.1 download now: https://lyrion.org/lms-server-repository/stable.xml

MichaIng, 2024-03-29:
Fixed with: https://github.com/MichaIng/DietPi/commit/b97891f

JRK0469, 2024-03-29:
> Fixed with: [b97891f](https://github.com/MichaIng/DietPi/commit/b97891f)

I have adapted “dietpi-software” on my Raspi. Everything works as usual again. Thanks.

MichaIng, 2024-03-29:
Great, thanks for testing.

Sabb0, 2024-04-03:
How can one install LMS before 9.3 is released? I tried to change the command but I have limited knowledge!

Joulinar, 2024-04-03:
You would need to switch to dev branch.

MichaIng, 2024-04-03:
Try this:
```sh
sed -i '/fallback_url.*slimdevices/c\local fallback_url="https://downloads.lms-community.org/nightly/logitechmediaserver_8.5.1~1711624004_$arch.deb"' /boot/dietpi/dietpi-software
```

Sabb0, 2024-04-03:
It worked! Thank you very much :)

Try this:

Joulinar, 2024-04-08:
@MichaIng 
Probably we need to recheck on the fallback URL. Seems not reliable.

```
 DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing Logitech Media Server: fka. SlimServer, SqueezeboxServer, SliMP3

[FAILED] DietPi-Software | Automatic latest Logitech Media Server download URL detection failed.
[FAILED] DietPi-Software | "https://downloads.lms-community.org/nightly/logitechmediaserver_8.5.1~1711624004_arm.deb" will be used as fallback, but a newer version might be available.
[FAILED] DietPi-Software | Please report this at: https://github.com/MichaIng/DietPi/issues
[  OK  ] DietPi-Software | cd /tmp/DietPi-Software
[FAILED] DietPi-Software | curl -sSfL https://downloads.lms-community.org/nightly/logitechmediaserver_8.5.1~1711624004_arm.deb -o logitechmediaserver_8.5.1~1711624004_arm.deb
```

Name of nightly build changed.

```
https://downloads.lms-community.org/nightly/logitechmediaserver_8.5.1~1712473975_arm.deb
https://downloads.lms-community.org/nightly/logitechmediaserver_8.5.1~1711624004_arm.deb
```

MichaIng, 2024-04-08:
They seem to remove older builds immediately, then fallback URLs have no point. But why is the auto detection not working for you? It works fine here.

Joulinar, 2024-04-08:
It was not tested on Dev branch. I just checked the fallback

MichaIng, 2024-04-08:
Ah no I understand the purpose, since I shared a simple workaround using the fallback URL.

closed 2024-04-16 by MichaIng