MichaIng/DietPi #6993: Regression: GPIO | Latest kernel deprecates sysfs API #5385

maartenlangeveld, 2024-03-29:
Hi @MichaIng,

As of lately my NanoHatOLED has stopped wroking.
It seems GPIO devices are missing again in latest kernel.

Would you be so kind to re-address this issue at Armbian devs, you have some authority moreover you arte knowledgeable whereas I am not? So that they can include GPIO in the kernel.

Thanks,
Maarten

Joulinar, 2024-03-29:
Well our relationship to Armbian devs is not the best. Usually they don't like DietPi and issues addressed from or side.

maartenlangeveld, 2024-03-30:
> Well our relationship to Armbian devs is not the best. Usually they don't like DietPi and issues addressed from or side.

Meanwhile I have addressed this issue at Armbian user forum, we leet you know if and when this issue will be resolved.

Joulinar, 2024-03-30:
Maybe you like to share the related form link?

MichaIng, 2024-03-30:
A build is running with this re-enabled: https://github.com/MichaIng/DietPi/actions/runs/8491886407/job/23264312017

You will find the kernel image and dtb packages here once done: https://dietpi.com/downloads/binaries/testing/
Would be great if you could give them a try.

maartenlangeveld, 2024-03-30:
Hi @MichaIng,

Eager to help, but do not know how to install the kernel. If you provide guidance, I will give it a try.

MichaIng, 2024-03-30:
```sh
cd /tmp
wget https://dietpi.com/downloads/binaries/testing/linux-{image,dtb}-current-sunxi64.deb
dpkg -i linux-{image,dtb}-current-sunxi64.deb
reboot
```

maartenlangeveld, 2024-03-30:
It boots normal however, NanoHatOLED still dark....

MichaIng, 2024-03-30:
Hmm,  the `gpio` command is WiringPi, which can be installed only on RPi and some Odroids. `/sys/class/gpio` however should exist. Which SBC are you using?

Can you verify it's using this Linux 6.6.23 build:
```sh
uname -a
```

I'll test on Orange Pi Zero 3 in an hour.

maartenlangeveld, 2024-03-30:
NanoPi NEO2

MichaIng, 2024-03-30:
Weird. Indeed I do not see the `libgpio-sysfs` driver being compiled in the logs, and the final kernel config does not contain `CONFIG_GPIO_SYSFS`, as if some dependency was missing. But all dependencies/selects are explicitly enabled: https://www.kernelconfig.io/config_gpio_sysfs?q=&kernelversion=6.6.23&arch=arm64
```
CONFIG_GPIOLIB=y
CONFIG_GPIO_SYSFS=y
CONFIG_GPIO_CDEV=y
CONFIG_SYSFS=y
```
Also the build log says:
```
[🐳|🌿] No misconfigurations or missing kernel option dependencies detected
```
I need to check other families configs.

MichaIng, 2024-03-30:
Ah, found the old PR which re-enabled this for 32-bit Allwinner chips: https://github.com/armbian/build/pull/3635
There is another flag missing indeed, respectively a patch to match the sysfs API independent of this other flag.

MichaIng, 2024-03-30:
Next try, now with the patch added: https://github.com/MichaIng/DietPi/actions/runs/8491886407/job/23266906946
If I am not mistaken, while `CONFIG_GPIO_SYSFS=y` was set for e.g. 32-bit Allwinner and Rockchip boards already, it is not effective anymore since Linux 6.6, as the patch to make it independent of `CONFIG_EXPERT=y` was not ported. And it is still needed: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/gpio/Kconfig#n61

maartenlangeveld, 2024-03-30:
Hi @MichaIng,

Next try did not help either unfortunately

```
2024-03-30 22:44:08 root@M-DNS:~# uname -a
Linux M-DNS 6.6.23-current-sunxi64 #1 SMP Tue Mar 26 22:22:53 UTC 2024 aarch64 GNU/Linux
2024-03-30 22:43:42 root@M-DNS:~# ls -l /sys/class/gpio
ls: cannot access '/sys/class/gpio': No such file or directory
```

MichaIng, 2024-03-30:
I need to learn how the patch system works. Looks like they need to be added in two config files: https://github.com/MichaIng/build/commit/52f6b19
Comparing with another added patch, now it should be applied: https://github.com/armbian/build/commit/cf392c1

Next attempt: https://github.com/MichaIng/DietPi/actions/runs/8491886407/job/23268009107

MichaIng, 2024-03-30:
Now it works:
```console
root@OrangePiZero3:~# ls -l /sys/class/gpio
total 0
--w------- 1 root root 4096 Mar 31 00:18 export
lrwxrwxrwx 1 root root    0 Mar 31 00:17 gpiochip0 -> ../../devices/platform/soc/300b000.pinctrl/gpio/gpiochip0
lrwxrwxrwx 1 root root    0 Mar 31 00:17 gpiochip352 -> ../../devices/platform/soc/7022000.pinctrl/gpio/gpiochip352
--w------- 1 root root 4096 Mar 31 00:18 unexport
```

maartenlangeveld, 2024-03-30:
Yes, I can confirm. THANKS !!!

MichaIng, 2024-03-30:
For some reason I cannot open a PR at Armbian. I would not wonder if Igor added me to some blacklist. Will try to check back tomorrow at Discord.

maartenlangeveld, 2024-03-30:
Thanks again, @MichaIng. It's very kind of you to provide this tremendous support !!

maartenlangeveld, 2024-04-02:
> Maybe you like to share the related form link?

https://forum.armbian.com/topic/36901-armbian-6616-sunxi64-kernel-breaks-gpio-on-nanopi-neo2/#comment-186330

MichaIng, 2024-05-13:
Our own kernel build with GPIO sysfs re-implemented, should have been installed with DietPi v9.4. In the meantime, also Armbian re-enabled it for all their builds, but by adding `CONFIG_EXPERT=y` for all of them, instead of decoupling `CONFIG_GPIO_SYSFS=y` from it: https://github.com/armbian/build/commit/fb17a2a
The upside is, that I can drop our patch then, keeping the diff with upstream as small as possible.

closed 2024-05-13 by MichaIng