MichaIng/DietPi #6997: Donation: Paypal "Donations to this recipient aren't supported in this country"

yuukiAme, 2024-04-02:
I have been using Dietpi for over 5 years and I want to give back to the developers and contributors to the project like I did to Pi-Hole project.

But Paypal doesn't allow me due to my country of origin.

This hurts.

[image]

What can I do to send you some money @MichaIng ?

Like I use Debit/Credit card but Paypal is blocking this attempt transaction.

yuukiAme, 2024-04-02:
Oh hell. since 2015

https://dietpi.com/forum/t/support-the-dietpi-project-donations-contribution/53/4

Paypal blocked Vietnam donation.

MichaIng, 2024-04-02:
Nasty, IIRC even from Japan it is not possible currently, and some other east/south-east asian countries. The way I understand things, it is not so much PayPal's decision to block this or that country, but the local regulations make it difficult to accept certain payment methods in certain counties. For the same reason, we cannot get BAT (Basic Attention Tokens, from Brave browser advertising system) donations anymore, since the only supported custodial wallet provider cannot offer accounts for German citizens because of local regulations ...

So there is nothing we can do about it. I guess it works when having a PayPal account. Otherwise you may consider donation via Patreon: https://www.patreon.com/DietPi

There are ideas to accept donations through GitHub Sponsors and/or Open Collective, but it is not yet there.

yuukiAme, 2024-04-02:
I can't do it on Patreon either. Same issues. But for some reasons, Pi-hole setups their donation system somehow to allow Visa Debit/Credit card. 

[image]

Maybe if you could setup like their donation setup, other users can donate as long as they have VISA card.