MichaIng/DietPi #6999: dietpi-update | Error with php apt signing key

LeoMirots, 2024-04-04:
#### Details:
- Date           | Thu Apr  4 19:26:00 CEST 2024
- Bug report     | 7612e170-1a77-4847-83d0-c694f4bdb89b
- DietPi version | v8.25.2 (MichaIng/8)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | `Linux DietPi 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | `apt-get -y update`
- Exit code      | 255
- Software title | DietPi-Update
- 
#### Steps to reproduce:
1. run `dietpi-update` command
#### Expected behaviour:
- clean update of the system
#### Actual behaviour:
- error with php apt repository signing key
#### Extra details:
- **SOLVED** importing new keyring with command:  `wget -O /etc/apt/trusted.gpg.d/php.gpg https://packages.sury.org/php/apt.gpg`
- check out this comment: [https://github.com/oerdnj/deb.sury.org/issues/1575#issuecomment-801251077](https://github.com/oerdnj/deb.sury.org/issues/1575#issuecomment-801251077)
- Maybe this was a problem related only to my machine, I share a bug with the solution also for other people that could have the same issue
#### Additional logs:
```
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Get:2 https://packages.sury.org/php buster InRelease [7559 B]
Hit:3 https://archive.raspberrypi.org/debian buster InRelease
Err:2 https://packages.sury.org/php buster InRelease
  The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
Fetched 7559 B in 1s (8327 B/s)
Reading package lists...
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://packages.sury.org/php buster InRelease: The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
W: Failed to fetch https://packages.sury.org/php/dists/buster/InRelease  The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
W: Some index files failed to download. They have been ignored, or old ones used instead.
```

MichaIng, 2024-04-04:
Since Debian Buster, we actually do not use this repository for PHP anymore. In case you did not add it yourself by purpose, you could remove it. Debian Buster provides a sufficiently high PHP version for the `dietpi-software` options.

Joulinar, 2024-04-04:
And Debian Buster is quite an outdated Debian version we have dropped support beginning of this year. You should consider moving to Debian Bullseye or even current Debian version Bookworm.

LeoMirots, 2024-04-04:
> Since Debian Buster, we actually do not use this repository for PHP anymore. In case you did not add it yourself by purpose, you could remove it. Debian Buster provides a sufficiently high PHP version for the `dietpi-software` options.

Many Thanks for your quick answer!
Yes, I addedd by myself (I think when dietpi-software was not compleatly developed yet)

LeoMirots, 2024-04-04:
> And Debian Buster is quite an outdated Debian version we have dropped support beginning of this year. You should consider moving to Debian Bullseye or even current Debian version Bookworm.

This is a good advice, I will try to update Debian version.
Do you have some good links to perform this upgrade?
Thanks in advance

Joulinar, 2024-04-04:
Upgrade to Bullseye https://dietpi.com/blog/?p=811
Upgrade to Bookworm https://dietpi.com/blog/?p=3128

LeoMirots, 2024-04-04:
> Upgrade to Bullseye https://dietpi.com/blog/?p=811 Upgrade to Bookworm https://dietpi.com/blog/?p=3128
> 
> Whatever you do, create a backup beforehand. Upgrade from such an old system could lead to some challenges. Especially around PHP 😜

Many thanks!

closed 2024-04-04 by LeoMirots