MichaIng/DietPi #7004: available image for Pi-5

DrCWO, 2024-04-07:
Hello,
I provide a ready to run image with my software for Pi 3/4 based on DietPi v8.10.2.

Can I use this image for Pi-5?
If yes what didn't work with Pi5 using this image?

Thanks'
DrCWO

MichaIng, 2024-04-07:
DietPi v8.10 won't run on this image, but the version is ships with.

What does not work yet via DietPi scripts is:
- Display/resolution settings
- Camera settings
- All software which currently still relies on the legacy framebuffer graphics or camera stack
- Overclocking

I made some ToDo here: https://github.com/MichaIng/DietPi/issues/6676#issuecomment-1902641095

DrCWO, 2024-04-09:
Right now I got a Pi5 and inserted my DietPi image  v8.10.2, uname:
Linux rooExtend 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux

```
The installed operating system (OS) does not indicate support for Raspüberry Pi 5
Update the OS or set os_check=0
```

I added os_check=0 in /boot/config.txt 
This time I see nothing at HDMI if I boot and also my IP scanner was not able to find the device.
For me this means: My custom image did not boot any more on Pi5.

**Any suggestions how I can make it work?**

MichaIng, 2024-04-09:
As I said, this cannot work, as the kernel simply does not support the RPi 5. You need to use our RPi 5 image.

DrCWO, 2024-04-09:
Oh sorry seems I got you wrong reading:
"But as long as you do not rely on affected camera software and can in case configure display and overclocking following the RPi docs, it should work pretty well."

closed 2025-07-17 by MichaIng