MichaIng/DietPi #7005: Faile install Logitech Media server

kallekoths, 2024-04-07:
#### Details:
- Date           | Sun Apr  7 18:50:30 CEST 2024
- Bug report     | c3402199-06df-45db-9e37-d0728117dadf
- DietPi version | v9.2.1 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi Zero 2 W (armv7l) (ID=3)
- Kernel version | `Linux DietPi-LMS 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `curl -sSfL https://downloads.slimdevices.com/nightly/logitechmediaserver_8.3.2~1702911674_arm.deb -o logitechmediaserver_8.3.2~1702911674_arm.deb`
- Exit code      | 22
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
curl: (22) The requested URL returned error: 404

MichaIng, 2024-04-07:
See here: https://github.com/MichaIng/DietPi/issues/6992#issuecomment-2035196171

closed 2024-04-07 by MichaIng

kallekoths, 2024-04-07:
it works - thanks a lot