MichaIng/DietPi #7006: logname: no login name

rlesac, 2024-04-07:
### Creating a bug report/issue
- [X ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=2
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version |bookworm 1

- Kernel version | Linux DietPi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux
- SBC model | RPi B (armv6l)
- Power supply used | 2A noname
- SD card used | 2gb noname

#### Additional Information (if applicable)

- Can this issue be replicated on a fresh installation of DietPi?
Yes
- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. install dietpi
2. type logname

#### Expected behaviour

- logname should return the current username

#### Actual behaviour

- logname: no login name

#### Extra details

- I used default settings, except using static IP, and choosing to remove wireless support. I also chose to install vim tiny 
-

Joulinar, 2024-04-07:
If I'm not mistaken, `logname` expecting a tty session which ssh normally does not provide.

rlesac, 2024-04-08:
I could not install various software I needed to run on the Raspberry pi because of it.
I got a hint that it is SSH related and switched from Dropbear to OpenSSH, and it's now working.

MichaIng, 2024-04-08:
`logname` returns the name of the user logged in based in `/var/run/utmp`, ignoring `sudo`/`runuser` and possible subshells. Dropbear seems to not create that entry, similar to how it does not use all PAM features etc. One can check the `w` command for registered login sessions.

To be true, I never heard of this command before, but `$USER`, `$LOGNAME` variables or `whoami` command. As you say "various software", which software exactly requires the `logname` command to be installed?

rlesac, 2024-04-09:
> `logname` returns the name of the user logged in based in `/var/run/utmp`, ignoring `sudo`/`runuser` and possible subshells. Dropbear seems to not create that entry, similar to how it does not use [...]

It's https://github.com/Stefal/rtkbase

MichaIng, 2024-04-09:
I see. It uses `bash` for the installer, and suggests to run it with `sudo`, so a failsafe replacement for `logname` would be `$SUDO_USER`, else `$USER` variables, which is assured to be set in `bash`, hence also makes this "empty value" check obsolete.

closed 2025-07-18 by MichaIng