MichaIng/DietPi #7009: Dietpi-Software Portainer HTTPS?

yuukiAme, 2024-04-08:
Hi,
I use Portainer with Docker and Docker-compose in DietPi.
I noticed that Portainer script install only use HTTP on 9002.
Which is why the browser threw an error when I uploaded a self-signed certificate and a private key.
Portainer became inaccesible after that.
So I had to reinstall with `dietpi-software reinstall 185` but this will result in the same HTTP. 
I already knew that so I tried the following commands with the help of [Portainer Docs](https://docs.portainer.io/start/upgrade/docker).

```

# list running containers
docker ps
# stop the running portainer container
docker stop portainer
# remove the running portainer container
docker rm portainer --force
# recreate the portainer container with ssl enabled
docker run -d -p 8000:8000 -p 9000:9000 -p 9443:9443 --name=portainer --restart=always -v /var/run/docker.sock:/var/run/docker.sock -v ~/docker/portainer:/data portainer/portainer-ce

```

After that recreation command, I will have to setup Portainer fresh on HTTPS port 9443 again with new account and password.

My question is:

# Why didn't DietPi easy script open HTTPS port like the Portainer docs?

Because if I run another `dietpi-software reinstall 185` now, Portainer will go back to HTTP on port 9002 only. No HTTPS.

yuukiAme, 2024-04-08:
I checked the closed issue and found [this comment](https://github.com/MichaIng/DietPi/issues/5586#issuecomment-1198902017). Not sure whether to discuss it in that thread or my thread.

MichaIng, 2024-04-18:
When we enable HTTPS OOTB, we should also add a (self-signed) certificate OOTB. Can this be easily imported into the Portainer container/volume via command prompt/script?

Joulinar, 2024-04-18:
https://docs.portainer.io/advanced/ssl#using-your-own-ssl-certificate-on-docker-standalone