MichaIng/DietPi #7013: After Bullseye upgrade, `apt-get -y -eany update` fails with `option 'e' [from -eany] is not understood`

tanderson1992, 2024-04-12:
I know I'm way behind but finally followed the [steps ](https://dietpi.com/blog/?p=811)to upgrade to bullseye (in preparation for upgrade to bookworm).

After the reboot, I noticed pihole was running but I could not access the web interface. I assumed it was because I needed to reinstall Lighttpd. Upon selecting the software to install, the error `E: Command line option 'e' [from -eany] is not understood in combination with the other options.` stopped the process. This same error occurs when I try to run `dietpi-update` or the script to upgrade to bookworm, whose output is shown below. Any ideas what I need to fix?

### Creating a bug report/issue
- [x ] I have searched the existing open and closed issues

#### Details:
- Date           | Fri Apr 12 10:21:23 EDT 2024
- DietPi version | v8.25.2 (MichaIng/8)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | `Linux DietPi 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | dietpi-bookworm-upgrade
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Command line option 'e' [from -eany] is not understood in combination with the other options.
```
output of apt-update:
```
root@DietPi:/home/dietpi# apt update
Hit:1 https://download.docker.com/linux/raspbian bullseye InRelease
Hit:2 https://download.mono-project.com/repo/debian raspbianbuster InRelease
Hit:3 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:4 https://apt.sonarr.tv develop InRelease
Hit:5 https://archive.raspberrypi.org/debian bullseye InRelease
Hit:6 https://repo.mosquitto.org/debian bullseye InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
51 packages can be upgraded. Run 'apt list --upgradable' to see them.
```
FWIW, I briefly toyed with running Sonarr on the RPi4 but moved it to another location. I'm fairly certain I uninstalled the software after giving up on it, but I see there are remnants in the apt list. I also note that I can't seem to find pihole in the software list. I'm looking now to see if that is also expected behavior. I felt like I read up enough before the upgrade but apparently missed some things.

MichaIng, 2024-04-12:
Hmm, it says "Bullseye", but the kernel version and this APT error indicate that not all packages have been fully upgraded to Bullseye yet. Can you show this output, please:
```sh
dpkg -l raspberrypi-kernel apt
```

tanderson1992, 2024-04-12:
D'oh! I missed two steps when following the manual upgrade process because I was scrolling ahead to see what came next. I'm re-starting the process from the `apt-upgrade` step and will report back after that's complete.

tanderson1992, 2024-04-12:
FWIW, here are the two steps I scrolled past when following the guide:
```
apt full-upgrade
```
and
```
/boot/dietpi/func/dietpi-obtain_hw_model
. /boot/dietpi/.hw_model
```
Missing one or both of those apparently led to not all packages upgrading, which then made the following step (which I did run) fail to reinstall pihole, lighttpd, etc because the packages hadn't been upgraded.

I started over with `apt-upgrade` then ran the two steps I'd skipped. At this point the system had marked pihole and others as not installed because I had messed up the previous try, I manually installed lighttpd, pihole, docker-compose and anything else I saw in the list that I had been using. After finishing that and the rest of the steps, I have a mostly working system again. The pihole and Home Assistant web interfaces work again. I'll close this since it seems I was the failure point here.

closed 2024-04-12 by tanderson1992

MichaIng, 2024-04-12:
Yes, the `apt full-upgrade` was the essential missing part then 🙂.