MichaIng/DietPi #7015: CPU Governor on NanoPi NEO3 seems faulty

maartenlangeveld, 2024-04-14:
### Creating a bug report/issue
- [**X**] I have searched the existing open and closed issues

#### Required Information
- DietPi version |
`G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=2
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`

- Distro version | `bookworm`
- Kernel version | `Linux M-CONTROL 6.6.16-current-rockchip64 #1 SMP PREEMPT Fri Feb 23 08:25:28 UTC 2024 aarch64 GNU/Linux`
- SBC model | `NanoPi NEO3 (aarch64)`
- Power supply used | `5V 1A`
- SD card used | `SanDisk Ultra`

#### Steps to reproduce

1. Run `dietpi-config`, choose `3  : Performance Options`
2. Choose `CPU Governor`
3. Choose `schedutil` or `ondemand`

#### Expected behaviour

`CPU frequency to vary depending on load`

#### Actual behaviour

`CPU always maxes out at the set max. frequency`

#### Extra details
`'The CPU Governor apparently works well on NanoPi NEO2 but not on NanoPi NEO3`

MichaIng, 2024-04-14:
How did you check the CPU frequency? Can you check the following to see whether the CPU was really on highest frequency all the time?
```sh
cat /sys/devices/system/cpu/cpufreq/policy0/stats/time_in_state
```

maartenlangeveld, 2024-04-15:
Did check it with cpu command. Maybe running the cpu command on my NanoPi Neo3 is enough to have all four cores hit the max (as opposed to my NanoPi Neo2) ? Will check /sys/devices/system/cpu/cpufreq/policy0/stats/time_in_state and let you know?

MichaIng, 2024-04-15:
> Maybe running the cpu command on my NanoPi Neo3 is enough to have all four cores hit the max (as opposed to my NanoPi Neo2) ?

This is what I was thinking about. The CPUFreq stats will tell you exactly how much time the CPU were in which state.

maartenlangeveld, 2024-04-15:
The stats look more confident:

```
2024-04-15 23:52:39 root@M-CONTROL:~# cat /sys/devices/system/cpu/cpufreq/policy0/stats/time_in_state
408000 8010673
600000 72077
816000 99663
1008000 33839
1200000 16463
1296000 63856
```

Seems that running cpu is not a good indicator.

MichaIng, 2024-04-15:
Indeed, on no high-end SBCs, the script itself raises the frequency already. `schedutil` governor generally reacts very fast and `ondemand` applies max frequency at 50% usage threshold. And I am not sure whether this is 50% from 400 MHz (respectively whichever frequency is applied currently), which would be a very low usage.

Probably we should add the info from these stats to the `cpu` output to avoid confusion, and since it is interesting.

maartenlangeveld, 2024-04-15:
Thanks, @MichaIng 
Would be good idea to provide the stats or an average for the last x minutes.

closed 2024-04-18 by MichaIng