MichaIng/DietPi #7022: Update failed from v9.2.1 to 9.3.0

vanGeek, 2024-04-15:
### Creating a bug report/issue
- [ x ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=2
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master' `
- Distro version | `bullseye 0`
- Kernel version | `Linux homePI 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux `
- SBC model | `RPI Model 4` or (EG: RPi3)
- Power supply used | 5V 1A RAVpower)
- SD card used |SanDisk ultra 128gb

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Dietpi-update 
2. ...

#### Expected behaviour

- update dietpi 

#### Actual behaviour

- fail to handshake : error in pull function 

#### Extra details

- homebridge / adguard home

MichaIng, 2024-04-15:
At which command did this happen exactly? At best, could you copy&paste the full update log/output here?

In case it happens during the APT update when pulling files from our APT repo: I get this by times when accessing resource on `dietpi.com` behind Cloudflare, but since no one else reported it yet, I was hoping it is an issue with my personal network/ISP/location. The same happens with `curl` (OpenSSL) and `wget`/`apt` (GnuTLS). Simply retrying the same thing solves it, so it is happening randomly, also independent of the actual Cloudflare edge involved:
```console
root@micha:/tmp/apt/lists# curl -I https://dietpi.com/motd
curl: (35) Recv failure: Connection reset by peer
root@micha:/tmp/apt/lists# wget --spider https://dietpi.com/motd
Spider mode enabled. Check if remote file exists.
--2024-04-15 23:44:25--  https://dietpi.com/motd
Resolving dietpi.com (dietpi.com)... 2606:4700:20::ac43:4565, 2606:4700:20::681a:4f3, 2606:4700:20::681a:5f3, ...
Connecting to dietpi.com (dietpi.com)|2606:4700:20::ac43:4565|:443... connected.
GnuTLS: Error in the pull function.
Unable to establish SSL connection.
```
Since APT uses GnuTLS as well, it could happen when updating/upgrading from our new APT repo. If so, I'll open a ticket to get this investigated.

Disabling IPv4 solves it permanently here, but this is really not something I want to accept. We even temporarily switched to the Cloudflare Pro plan to test some routing optimisations, but while it helped for another issue, it did not help in my case.

Joulinar, 2024-04-16:
as workaround, you can try to disable IPv6

MichaIng, 2024-04-16:
Yes, or just "Retry" from the error handler. Okay I'll open a ticket with Cloudflare. It is really an issue with their edge server(s), maybe in combination with some settings, as it does not happen when accessing our server directly.

vanGeek, 2024-04-16:
Retry did not work, i tried it several times but always ran into the same issue. 
I tried it today and it worked fine.

Power-onoff, 2024-04-17:
I'm having the same problem

- bug report reference code: 6691dfaa-3bee-47d7-b6b5-b34978f14cb7

`#### Details:
- Date           | Wed Apr 17 21:29:04 KST 2024
- DietPi version | v9.2.1 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | Native PC (x86_64) (ID=21)
- Kernel version | `Linux nas-linebox 6.1.0-20-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.85-1 (2024-04-11) x86_64 GNU/Linux`
- Distro         | bookworm (ID=7)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 https://deb.debian.org/debian bookworm InRelease
Hit:2 https://deb.debian.org/debian bookworm-updates InRelease
Hit:3 https://deb.debian.org/debian-security bookworm-security InRelease
Get:4 https://download.docker.com/linux/debian bookworm InRelease [43.3 kB]
Get:5 https://dietpi.com/apt bookworm InRelease [3522 B]
Hit:6 https://deb.debian.org/debian bookworm-backports InRelease
Err:5 https://dietpi.com/apt bookworm InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 1DF45FDB9C6CE851
Reading package lists...
W: https://deb.debian.org/debian/dists/bookworm/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/dietpi.asc are ignored as the file is not readable by user '_apt' executing apt-key.
W: https://deb.debian.org/debian/dists/bookworm-updates/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/dietpi.asc are ignored as the file is not readable by user '_apt' executing apt-key.
W: https://deb.debian.org/debian-security/dists/bookworm-security/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/dietpi.asc are ignored as the file is not readable by user '_apt' executing apt-key.
W: https://download.docker.com/linux/debian/dists/bookworm/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/dietpi.asc are ignored as the file is not readable by user '_apt' executing apt-key.
W: https://dietpi.com/apt/dists/bookworm/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/dietpi.asc are ignored as the file is not readable by user '_apt' executing apt-key.
W: GPG error: https://dietpi.com/apt bookworm InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 1DF45FDB9C6CE851
E: The repository 'https://dietpi.com/apt bookworm InRelease' is not signed.
W: https://deb.debian.org/debian/dists/bookworm-backports/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/dietpi.asc are ignored as the file is not readable by user '_apt' executing apt-key.
```
`

Joulinar, 2024-04-17:
> I'm having the same problem

It's different problem

> Err:5 https://dietpi.com/apt bookworm InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 1DF45FDB9C6CE851

Somehow the public key is missing. 🤔 Usually this should be applied during update

Power-onoff, 2024-04-17:
> > 나는 같은 문제가 있습니다
> 
> 그것은 다른 문제입니다
> 
> > 오류:5 https://dietpi.com/apt 책벌레 InRelease 공개 키를 사용할 수 없으므로 다음 서명을 확인할 수 없습니다. NO_PUBKEY 1DF45FDB9C6CE851
> 
> 어쨌든 공개 키가 없습니다. 🤔 일반적으로 업데이트 중에 적용해야 합니다.

As you told me, I think the permission for the public key file is wrong

```
ls -l /etc/apt /etc/apt/trusted.gpg.d/
/etc/apt:
total 28
drwxr-xr-x 2 root root 4096 Apr 15 04:32 apt.conf.d
drwxr-xr-x 2 root root 4096 May 25  2023 auth.conf.d
drwxr-xr-x 2 root root 4096 May 25  2023 keyrings
drwxr-xr-x 2 root root 4096 Aug  1  2023 preferences.d
-rw-r--r-- 1 root root  372 Aug  8  2023 sources.list
drwxr-xr-x 2 root root 4096 Apr 17 21:16 sources.list.d
drwxr-xr-x 2 root root 4096 Apr 17 21:16 trusted.gpg.d

/etc/apt/trusted.gpg.d/:
total 96
-rw-r--r-- 1 root root 11861 Mar 29  2023 debian-archive-bookworm-automatic.asc
-rw-r--r-- 1 root root 11873 Mar 29  2023 debian-archive-bookworm-security-automatic.asc
-rw-r--r-- 1 root root   461 Mar 29  2023 debian-archive-bookworm-stable.asc
-rw-r--r-- 1 root root 11861 Mar 29  2023 debian-archive-bullseye-automatic.asc
-rw-r--r-- 1 root root 11873 Mar 29  2023 debian-archive-bullseye-security-automatic.asc
-rw-r--r-- 1 root root  3403 Mar 29  2023 debian-archive-bullseye-stable.asc
-rw-r--r-- 1 root root 11093 Mar 29  2023 debian-archive-buster-automatic.asc
-rw-r--r-- 1 root root 11105 Mar 29  2023 debian-archive-buster-security-automatic.asc
-rw-r--r-- 1 root root  1704 Mar 29  2023 debian-archive-buster-stable.asc
-rw-r--r-- 1 root root  2760 Aug  8  2023 dietpi-docker.gpg
-rw-r----- 1 root root  4694 Apr 17 21:16 dietpi.asc
```

```
chmod 644 /etc/apt/trusted.gpg.d/dietpi.asc
ls -l /etc/apt/trusted.gpg.d/
-rw-r--r-- 1 root root 11861 Mar 29  2023 debian-archive-bookworm-automatic.asc
-rw-r--r-- 1 root root 11873 Mar 29  2023 debian-archive-bookworm-security-automatic.asc
-rw-r--r-- 1 root root   461 Mar 29  2023 debian-archive-bookworm-stable.asc
-rw-r--r-- 1 root root 11861 Mar 29  2023 debian-archive-bullseye-automatic.asc
-rw-r--r-- 1 root root 11873 Mar 29  2023 debian-archive-bullseye-security-automatic.asc
-rw-r--r-- 1 root root  3403 Mar 29  2023 debian-archive-bullseye-stable.asc
-rw-r--r-- 1 root root 11093 Mar 29  2023 debian-archive-buster-automatic.asc
-rw-r--r-- 1 root root 11105 Mar 29  2023 debian-archive-buster-security-automatic.asc
-rw-r--r-- 1 root root  1704 Mar 29  2023 debian-archive-buster-stable.asc
-rw-r--r-- 1 root root  2760 Aug  8  2023 dietpi-docker.gpg
-rw-r--r-- 1 root root  4694 Apr 17 21:16 dietpi.asc
```

I think "Other" read permission was lost due to umask setting while downloading the public key file

MichaIng, 2024-04-17:
Ah right, it was our key only, but since it is in `trusted.gpg.d`, it is checked for every repo. Did you change your default umask? I then wonder why the `/etc/apt/sources.list.d/dietpi.list` is not affected.

Probably we should set it to `0022` in all our scripts. It counter-acts security ideas when it is intentionally set to `0027`, but otherwise a lot of things our scripts to rely on it being `0022` 🤔.

Power-onoff, 2024-04-17:
> Ah right, it was our key only, but since it is in `trusted.gpg.d`, it is checked for every repo. Did you change your default umask? I then wonder why the `/etc/apt/sources.list.d/dietpi.list` is not [...]

The default umask value is 0022.

However, before running the detpi-update, I set the umask value to 0027

The current "/etc/apt/trusted.gpg.d/dietpi.asc" file has been successfully read and the dietpi-update has been successfully run

Later, I checked the permissions of the "/etc/apt/sources.list.d/dietpi.list" file and found that you still don't have read permissions

```
root@localhost:/etc/apt/sources.list.d# ls -l
total 8
-rw-r----- 1 root root 41 Apr 17 23:46 dietpi.list
-rw-r--r-- 1 root root 62 Aug  8  2023 docker.list
```

MichaIng, 2024-04-18:
Okay, then the lists are not read by this `_apt` user. A check via `htop` while `apt update` runs, reveals that indeed only the cryptography stuff and the HTTPS connection is done by this user, after the list has been read already.

I think it is indeed best when we actively set `0022` in our scripts. It applies only for the script/subshell, hence does not affect any other/parent shell. In case checking and changing modes of every generated file would be much too much additional code and work, an we often rely on group R/W or all/other R/O modes, to allow access to downloads/media files by downloaders and media players/servers, or leave certain (config) files owned by `root:root` when the consuming service is not supposed to edit them directly.

I aligns with the enforced `PATH` and `LC_ALL`/`LANG`: https://github.com/MichaIng/DietPi/commit/c9cc8df

MichaIng, 2025-05-25:
Is anyone still facing these issues? I cannot replicate them anymore, but I also moved to anther ISP in the meantime.

MichaIng, 2025-05-31:
Closing this for now. Feel free to reopen if the issue persists.

closed 2025-05-31 by MichaIng