MichaIng/DietPi #7023: DietPi has encountered an error during update to v9.3.0

gioxx, 2024-04-16:
#### Details:
- Date           | Tue Apr 16 11:05:48 CEST 2024
- DietPi version | v9.2.1 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | Virtual Machine (x86_64) (ID=20)
- Kernel version | `Linux Pyxmox 5.10.0-28-amd64 #1 SMP Debian 5.10.209-2 (2024-01-31) x86_64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. dietpi-update
#### Expected behaviour:

- DietPi should update
#### Actual behaviour:

- Can't update DietPi
#### Additional logs:
```
Hit:1 https://cli.github.com/packages stable InRelease
Hit:2 https://deb.debian.org/debian bullseye InRelease
Hit:3 https://deb.debian.org/debian bullseye-updates InRelease
Hit:4 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:5 https://deb.debian.org/debian bullseye-backports InRelease
Ign:6 https://deb.debian.org/debian buster-backports InRelease
Err:7 https://deb.debian.org/debian buster-backports Release
  404  Not Found [IP: 146.75.54.132 443]
Get:8 https://pkgs.tailscale.com/stable/debian bullseye InRelease
Reading package lists...
E: The repository 'https://deb.debian.org/debian buster-backports Release' no longer has a Release file.
```

Joulinar, 2024-04-16:
You have a very old Debian Buster repository configured that would be needed to remove.

gioxx, 2024-04-16:
You're right. I have removed https://deb.debian.org/debian/ buster-backports main from sources.list.
Thanks!

closed 2024-04-16 by gioxx

MichaIng, 2024-04-18:
Solved: https://github.com/MichaIng/DietPi/commit/83c19ca