MichaIng/DietPi #7027: USB0 Network setting

mrfoxie, 2024-04-19:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- Currently im working on raspberry pi 0 2w to convert it into usb/ethernet gadget its works completely fine with new beta update v9.3.0

#### Describe the solution you'd like:

- can we have network configuration setting in the dietpi-config > Network Options: Adapters menu aside with ethernet and Wi-Fi setting.
- also we can add a new command such as dietpi-usb0 or something like that so that it will be easy to control.

if this is not useful you may close this request, and if this is worth it can be developed by the team
thank you in advance😊

MichaIng, 2024-04-19:
There is a feature request for this here, which contains info on how to configure it manually, using `usb0` as interface name instead of `eth0`: #3253

I'll close this one as duplicate.

closed 2024-04-19 by MichaIng

mrfoxie, 2024-04-25:
thank you can you add wiki of this in the docs?