MichaIng/DietPi #7028: Pine A64 - update from 9.2.1 to 9.3.0 fails while upgrading APT packages

aabarlaam, 2024-04-20:
### Creating a bug report/issue
- [ x] I have searched the existing open and closed issues

#### Required Information
- Date           | Sat Apr 20 18:55:17 CEST 2024
- DietPi version | v9.2.1 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | PINE A64 (aarch64) (ID=40)
- Kernel version | `Linux DietPi-A64 6.6.16-current-sunxi64 #2 SMP Fri Feb 23 08:25:28 UTC 2024 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7)
- Command        | `apt-get -y --with-new-pkgs upgrade`
- Exit code      | 100

#### Additional Information (if applicable)
- Software title | DietPi-Update

#### Steps to reproduce

1. Run dietpi-update from console or from dietpi-launcher

#### Expected behaviour

- Dietpi is updated from 9.2.1 to 9.3.0

#### Actual behaviour

- APT terminates with error code 100 after applying the pre-patches

#### Extra details

- note: this issue seems specific to 9.3.0 , or at least it didn't happen in the previous upgrades

- Sample run:

root@DietPi-XXXX:~# dietpi-launcher

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[  OK  ] DietPi-Update | Checking IPv4 network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version
[  OK  ] DietPi-Update | Got valid latest version: 9.3.0
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v9.2.1
[ INFO ] DietPi-Update | Latest version  : v9.3.0

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for update pre-requirements

[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=13495 MiB | required=100 MiB
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : nfs-kernel-server
[  OK  ] DietPi-Services | stop : avahi-daemon

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | Downloading pre-patches
[  OK  ] DietPi-Update | Applying execute permission
[  OK  ] DietPi-Update | Successfully applied pre-patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 https://deb.debian.org/debian bookworm InRelease
Hit:2 https://deb.debian.org/debian bookworm-updates InRelease
Get:3 https://deb.debian.org/debian-security bookworm-security InRelease [48.0 kB]
Hit:4 https://deb.debian.org/debian bookworm-backports InRelease
Hit:5 https://mirrors.xtom.de/armbian bookworm InRelease
Fetched 48.0 kB in 4s (12.5 kB/s)
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.
[FAILED] DietPi-Update | APT upgrade
 - Command: apt-get -y --with-new-pkgs upgrade

Joulinar, 2024-04-20:
> E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.

Did you already tried `'dpkg --configure -a`

aabarlaam, 2024-04-20:
Tried right now:  after running  **dpkg --configure -a**  the update worked as expected .

MichaIng, 2024-04-20:
Great. As the error message says, this happens when APT installs/upgrades were interrupted before, e.g. by hitting CTRL+C or when an error occurred which aborts the configuration step. `dpkg --configure -a` runs all non-finished configuration steps, i.e. finishes all aborted APT installs/upgrades, when possible.

closed 2024-04-20 by MichaIng