MichaIng/DietPi #7031: Elecrow 10.7" 1920x1080 IPS HDMI - Current console font on boot looks unreadable.

oxoocoffee, 2024-04-22:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- Default font looks small/tiny on boot and in run level 3.

#### Describe the solution you'd like:

- I have no solution, But I could try to look into it if I can point to the correct source/file/etc
- If I select ```Elecrow 1024x600 7"``` listed, it gets a bit better after reboot.
- I know I can use ```setupcon``` but it would be great to be part of the installer.

#### Describe alternatives you've considered:

- None

#### Additional context

- If there is any information from my hardware I can provide , please let me know

MichaIng, 2024-04-22:
Is this what you re looking for?
```sh
dpkg-reconfigure console-setup
```

oxoocoffee, 2024-04-22:
- That did the trick vs manual. The fonts got bigger. But for my taste and screen type, I would like to see either one more level font size in the list for `fixed` or `Terminus`. But it is much better than initially
- Did I miss an initial step somewhere to pick the console font type and size when no desktop is selected for installation?

Cheers

oxoocoffee, 2024-12-09:
Are there any options to force different font size before initial setup?
I always have to do it on large monitor first and switch console font size before I switch to 10" screen. Anything in config.txt I can modify before attempting first boot?

MichaIng, 2024-12-12:
Hmm, no direct option for that. But you could do with a custom script. Put a `Automation_Custom_PreScript.sh` into the setup FAT partition, right after flashing the image, or into the `/boot` directory if the `ext4` partition, if the FAT partition does not exist. It is automatically copied in place and executed early on first boot.

closed 2025-07-18 by MichaIng