MichaIng/DietPi #7035: Issue with /dev/dri/ with no desktop running

oxoocoffee, 2024-04-22:
### Creating a bug report/issue
- [x ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | 
```bash
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=3
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `bookworm 0`
- Kernel version | `Linux DietPi 6.6.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux`
- SBC model | `RPi 5 Model B (aarch64)` or (EG: RPi3)
- Firmware/EEPROM | 
```bash
BOOTLOADER: up to date
   CURRENT: Sat Apr 20 10:53:30 UTC 2024 (1713610410)
    LATEST: Sat Apr 20 10:53:30 UTC 2024 (1713610410)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader-2712/default)
            Use raspi-config to change the release.
```
- Power supply used | `200W multi-port USB charger`
- SD card used | `SanDisk 64G`

#### Additional Information (if applicable)
- Software title | `Discovered DRI when researching ticket` #7029
- Was the software title installed freshly or updated/migrated? `My test program which fais`
- Can this issue be replicated on a fresh installation of DietPi? `Yes`

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Boot from a fresh image on rPi 5 and complete installation with our desktop selected.
2. Install build tools `sudo apt install build-essential git cmake pkg-config libdrm-dev drm_info`
3. `ls /dev/dri/`
```bash
drwxr-xr-x 2 root root        100 Apr 22 16:33 by-path
crw-rw---- 1 root video  226,   0 Apr 22 16:33 card0
crw-rw---- 1 root video  226,   1 Apr 22 16:33 card1
crw-rw---- 1 root render 226, 128 Apr 22 16:33 renderD128
```
4. Uncompress 
[dri-test.tgz](https://github.com/MichaIng/DietPi/files/15069331/dri-test.tgz)

5. `cd dri-test/build && cmake .. && make`

#### Expected behaviour

- See a full screen with a test chess checker. I do not know if it fully work since DRI initialization fails.

#### Actual behaviour

- Since my program terminates on a second call after `open` which is `device.resources = drmModeGetResources(device.fd)` with and error `Failed to get DRM resources. Operation not supported`

#### Extra details

- `groups` produced `tty dialout sudo audio video users input render netdev gpio i2c spi`

MichaIng, 2024-04-24:
KMS is enabled, right?
```sh
grep dtoverlay=vc4-kms-v3d
```
Did you start the test program from within a desktop session? I don't know what it is doing, where "device" and "fd" (file descriptor?) are coming from, but maybe it requires a context or input to know which of the 2 DRI devices it should use?

Did you test this with RPi OS Bookworm?

oxoocoffee, 2024-04-25:
@MichaIng 

Yes. All tests were performed with
` dtoverlay=vc4-kms-v3d`

- I did not stop from within a desktop. I have no desktop installed/running
- I did not test rPI OS. I can try to download it and see. What specific version do you suggest?

oxoocoffee, 2024-05-19:
While looking into this issue some more today I found this [#5049](https://github.com/raspberrypi/linux/issues/5049)

`Change the config.txt file to dtoverlay=vc4-kms-v3d-pi4`

oxoocoffee, 2024-12-09:
What are the differences between vc4-kms-v3d and vc4-kms-v3d-pi4
I have rPi4 and rPI5 that I am messing around with.

MichaIng, 2024-12-10:
On RPi 4, `vc4-kms-v3d` is internally rewritten to `vc4-kms-v3d-pi4`. As of the different RPi 4 GPU, it requires a different overlay, but the firmware handles it internally, to make it easier.

So `vc4-kms-v3d` is all you ever need to enable KMS on any RPi model.

closed 2025-06-22 by MichaIng