MichaIng/DietPi #7039: Armbian to Dietpi conversion not working (Armbian_community_24.5.0-trunk.433_Aml-s9xx-box_bookworm)

bigboo3000, 2024-04-24:
#### Required Information
- DietPi version | 9.3 (conversion script)
- Distro version | Armbian_community_24.5.0-trunk.433_Aml-s9xx-box_bookworm_current_6.6.28_minimal.img
- Kernel version | 6.6.28
- SBC model | TV box H96 max X3 Amlogic S905x3 4Gb RAM/64Gb Gigabit ethernet 
- Power supply used | original 5v 
- SD card used | 32Gb sdcard

#### Steps to reproduce

1. install Armbian_community_24.5.0-trunk.433_Aml-s9xx-box_bookworm_current_6.6.28_minimal.img to my tv box, reboot, everything is working
2. run the conversion script, reboot
3. sdcard is not bootable anymore, boot to internal emmc android

#### Expected behaviour

- boot to dietpi

#### Actual behaviour

- sdcard boot broken, nothing happens or displayed, box directly boot to internal emmc android

#### Extra details

I see these error messages in the script:

```
ln: failed to create hard link '/boot/initrd.img-6.6.28-current-meson64.dpkg-bak' => '/boot/initrd.img-6.6.28-current-meson64': Operation not permitted
update-initramfs: Generating /boot/initrd.img-6.6.28-current-meson64
update-initramfs: Converting to U-Boot format
Image Name:   uInitrd
Created:      Wed Apr 24 14:41:00 2024
Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
Data Size:    15290812 Bytes = 14932.43 KiB = 14.58 MiB
Load Address: 00000000
Entry Point:  00000000
ln: failed to create symbolic link '/boot/uInitrd': Operation not permitted
renamed '/boot/uInitrd-6.6.28-current-meson64' -> '/boot/uInitrd'
```

Full output of the conversion script:
[armbian to dietpi script.txt](https://github.com/MichaIng/DietPi/files/15094833/armbian.to.dietpi.script.txt)

Is it related to changes to boot process in recent armbian builds?
https://github.com/armbian/build/pull/4938
https://github.com/armbian/build/pull/6214
https://github.com/armbian/build/pull/6215
https://github.com/armbian/build/pull/6287

What can I do to fix this?

MichaIng, 2024-04-24:
The error you pasted is unrelated. Obviously, `/boot` is a FAT partition mount in this case, so that creating a symlink fails. Instead, the initramfs is renamed (next line), hence all fine with this.

These Amlogic S905 TV boxes seem to use `/boot/extlinux/extlinux.conf` instead of `/boot/boot.scr`? Does it contain versioned dtb/kernel/initramfs filenames or the version-agnostic copies?
```sh
cat /boot/extlinux/extlinux.conf
```

... ah, or probably this is the issue?
```
ln: failed to create hard link '/boot/initrd.img-6.6.28-current-meson64.dpkg-bak' => '/boot/initrd.img-6.6.28-current-meson64': Operation not permitted
```
Checking `/usr/sbin/update-initramfs`:
```sh
        ln -f "${initramfs}" "${initramfs_bak}" \
                || cp -a "${initramfs}" "${initramfs_bak}"
```
So similar to our script, when generating the link fails, the file is copied instead.

So both errors are not related. Can you also show the content of `/boot`:
```sh
ls -l /boot
```
And is there a way to attach a UART adapter to check serial console output?

bigboo3000, 2024-04-24:
After writing the armbian img to the sdcard, you need to customize extlinux.conf manually with the correct dtb for your box, and manually rename u-boot.ext with the correct one for your CPU, before the first boot

I changed this file based on template (I juste added FDT /dtb/amlogic/meson-sm1-h96-max.dtb line):

```
root@aml-s9xx-box:~# cat /boot/extlinux/extlinux.conf
label Armbian_community
  kernel /Image
  initrd /uInitrd
  fdtdir /dtb/
  FDT /dtb/amlogic/meson-sm1-h96-max.dtb

  append root=UUID=f0e88e45-dc65-441c-b1db-fed5dad80831 rootflags=data=writeback console=ttyAML0,115200n8 console=tty0 rw no_console_suspend consoleblank=0 fsck.fix=yes fsck.repair=yes net.ifnames=0 splash plymouth.ignore-serial-consoles
```
```
root@aml-s9xx-box:~# cat /boot/extlinux/extlinux.conf.template
label Armbian-unofficial
  kernel /Image
  initrd /uInitrd
  fdtdir /dtb/

#Insert the correct FDT line for your box's dtb in the extlinux.conf file
#The dtbs for Amlogic s9xx boxes are located in /boot/dtb/amlogic
#For example for a TX3 mini box the FDT line to insert would look like this:
  FDT /dtb/amlogic/meson-gxl-s905w-tx3-mini.dtb

  append root=UUID=<root partition uuid> rootflags=data=writeback  console=ttyAML0,115200n8 console=tty0 rw no_console_suspend consoleblank=0 fsck.fix=yes fsck.repair=yes net.ifnames=0 splash plymouth.ignore-serial-consoles
```

```
root@aml-s9xx-box:~# ls -l /boot
total 64900
-rwxr-xr-x 1 root root      800 Apr 18 00:58  aml_autoscript
-rwxr-xr-x 1 root root    38518 Apr 18 09:43  boot.bmp
drwxr-xr-x 2 root root     4096 Apr 18 09:42  build-u-boot
-rwxr-xr-x 1 root root   253371 Apr 17 16:37  config-6.6.28-current-meson64
drwxr-xr-x 3 root root     4096 Apr 18 09:42  dtb
-rwxr-xr-x 1 root root      174 Apr 18 00:58  emmc_autoscript
drwxr-xr-x 2 root root     4096 Apr 18 09:42  extlinux
-rwxr-xr-x 1 root root 27425280 Apr 17 16:37  Image
-rwxr-xr-x 1 root root 15289522 Apr 18 09:46  initrd.img-6.6.28-current-meson64
-rwxr-xr-x 1 root root      537 Apr 18 00:58  s905_autoscript
-rwxr-xr-x 1 root root  4417431 Apr 17 16:37  System.map-6.6.28-current-meson64
drwxr-xr-x 2 root root     4096 Apr 24 16:04 'System Volume Information'
-rwxr-xr-x 1 root root   798672 Apr 18 00:58  u-boot.ext
-rwxr-xr-x 1 root root   609247 Apr 18 00:58  u-boot-s905
-rwxr-xr-x 1 root root   851352 Apr 18 00:58  u-boot-s905x2-s922
-rwxr-xr-x 1 root root   798672 Apr 18 00:58  u-boot-s905x3
-rwxr-xr-x 1 root root   646455 Apr 18 00:58  u-boot-s905x-s912
-rwxr-xr-x 1 root root 15289586 Apr 18 09:46  uInitrd
```
```
root@aml-s9xx-box:~# ls -l /boot/dtb/amlogic
total 5088
-rwxr-xr-x 1 root root  3108 Apr 17 16:37 amlogic-c3-c302x-aw409.dtb
-rwxr-xr-x 1 root root  3373 Apr 17 16:37 amlogic-t7-a311d2-an400.dtb
-rwxr-xr-x 1 root root  3689 Apr 17 16:37 amlogic-t7-a311d2-khadas-vim4.dtb
-rwxr-xr-x 1 root root  4065 Apr 17 16:37 meson-a1-ad401.dtb
-rwxr-xr-x 1 root root 49780 Apr 17 16:37 meson-axg-jethome-jethub-j100.dtb
-rwxr-xr-x 1 root root 49997 Apr 17 16:37 meson-axg-jethome-jethub-j110-rev-2.dtb
-rwxr-xr-x 1 root root 49822 Apr 17 16:37 meson-axg-jethome-jethub-j110-rev-3.dtb
-rwxr-xr-x 1 root root 52750 Apr 17 16:37 meson-axg-s400.dtb
-rwxr-xr-x 1 root root 74833 Apr 17 16:37 meson-g12a-radxa-zero.dtb
-rwxr-xr-x 1 root root 75041 Apr 17 16:37 meson-g12a-radxa-zero-spidev.dtb
-rwxr-xr-x 1 root root 75571 Apr 17 16:37 meson-g12a-sei510.dtb
-rwxr-xr-x 1 root root 72626 Apr 17 16:37 meson-g12a-u200.dtb
-rwxr-xr-x 1 root root 74784 Apr 17 16:37 meson-g12a-x96-max.dtb
-rwxr-xr-x 1 root root 78632 Apr 17 16:37 meson-g12b-a311d-bananapi-m2s.dtb
-rwxr-xr-x 1 root root 79311 Apr 17 16:37 meson-g12b-a311d-khadas-vim3.dtb
-rwxr-xr-x 1 root root 79519 Apr 17 16:37 meson-g12b-a311d-khadas-vim3-spidev.dtb
[... 57 lines cut ...]
-rwxr-xr-x 1 root root 43817 Apr 17 16:37 meson-gxm-t95z-plus.dtb
-rwxr-xr-x 1 root root 41804 Apr 17 16:37 meson-gxm-vega-s96.dtb
-rwxr-xr-x 1 root root 42366 Apr 17 16:37 meson-gxm-wetek-core2.dtb
-rwxr-xr-x 1 root root  3518 Apr 17 16:37 meson-s4-s805x2-aq222.dtb
-rwxr-xr-x 1 root root 73850 Apr 17 16:37 meson-sm1-a95xf3-air.dtb
-rwxr-xr-x 1 root root 74222 Apr 17 16:37 meson-sm1-a95xf3-air-gbit.dtb
-rwxr-xr-x 1 root root 75425 Apr 17 16:37 meson-sm1-bananapi-m2-pro.dtb
-rwxr-xr-x 1 root root 77557 Apr 17 16:37 meson-sm1-bananapi-m5.dtb
-rwxr-xr-x 1 root root 74429 Apr 17 16:37 meson-sm1-h96-max.dtb
-rwxr-xr-x 1 root root 76901 Apr 17 16:37 meson-sm1-khadas-vim3l.dtb
-rwxr-xr-x 1 root root 77105 Apr 17 16:37 meson-sm1-khadas-vim3l-spidev.dtb
-rwxr-xr-x 1 root root 76913 Apr 17 16:37 meson-sm1-khadas-vim3l-spinor.dtb
-rwxr-xr-x 1 root root 75717 Apr 17 16:37 meson-sm1-odroid-c4.dtb
-rwxr-xr-x 1 root root 75885 Apr 17 16:37 meson-sm1-odroid-c4-spidev.dtb
-rwxr-xr-x 1 root root 76617 Apr 17 16:37 meson-sm1-odroid-hc4.dtb
-rwxr-xr-x 1 root root 76695 Apr 17 16:37 meson-sm1-sei610.dtb
-rwxr-xr-x 1 root root 73900 Apr 17 16:37 meson-sm1-x96-air.dtb
-rwxr-xr-x 1 root root 74268 Apr 17 16:37 meson-sm1-x96-air-gbit.dtb
drwxr-xr-x 2 root root  8192 Apr 18 09:42 overlay
```

I spent hours to find uart pins on the board, unfortunately after checking 4pda russian forum for this box it seems recent board versions have no uart pins connected anymore :(

I took a photo of armbian early boot process (sorry for the bad quality :p ):
[image]

MichaIng, 2024-04-24:
Okay, everything looks fine, all files are present/have been renamed as they should, extlinux uses them and U-Boot loads all of them. Only problem is that the kernel for some reason hangs on startup. Since there was a kernel upgrade, could you test `apt update && apt upgrade` on the Armbian image, prior to conversion, and see whether it still reboots fine?

bigboo3000, 2024-04-24:
Sorry for the misunderstanding, all the things in my previous message was from the original armbian image.

I found the issue:

this is the boot folder just after running the dietpi conversion script (and before reboot):

```
root@aml-s9xx-box:~# ls -l /boot
total 62072
-rwxr-xr-x 1 root root 27425280 Apr 17 16:37  Image
drwxr-xr-x 2 root root     4096 Apr 24 17:45 'System Volume Information'
-rwxr-xr-x 1 root root  4417431 Apr 17 16:37  System.map-6.6.28-current-meson64
-rwxr-xr-x 1 root root   253371 Apr 17 16:37  config-6.6.28-current-meson64
drwxr-xr-x 4 root root     4096 Apr 24 18:49  dietpi
-rwxr-xr-x 1 root root    18092 Apr 14 19:32  dietpi-LICENSE.txt
-rwxr-xr-x 1 root root    16059 Apr 14 19:32  dietpi-README.md
-rwxr-xr-x 1 root root    17823 Apr 24 18:49  dietpi.txt
drwxr-xr-x 3 root root     4096 Apr 24 17:11  dtb
drwxr-xr-x 2 root root     4096 Apr 24 18:42  extlinux
-rwxr-xr-x 1 root root 15290960 Apr 24 18:45  initrd.img-6.6.28-current-meson64
-rwxr-xr-x 1 root root   798672 Apr 18 00:58  u-boot.ext
-rwxr-xr-x 1 root root 15291024 Apr 24 18:45  uInitrd
root@aml-s9xx-box:~#
```

you can see the script removed 3 files related to amlogic boot:
aml_autoscript
emmc_autoscript
s905_autoscript

That's why it can't boot anymore.

I copied them back from the original armbian image and then diepi boot correctly!

I don't understand why the script removed these files.

MichaIng, 2024-04-24:
Hmm, our script does not actively remove those files, so I guess they are part of a DEB package, which contains them. It might the the Armbian BSP package. ... indeed, it is `armbian-bsp-cli-aml-s9xx-box-current`. Those packages are vastly incompatible with DietPi, so the best I can currently think of is to host an own package which contains only those U-Boot blobs and scripts, and is installed as replacement, when this particular BSP package is found. But with all this manual renaming, the blobs in general, the plan to build those from source as part of the Armbian build system (reasonably, see readme in `/boot/build-u-boot`) looks all very much WIP. I perfectly understand why Armbian never officially supported those TV boxes.

bigboo3000, 2024-04-25:
Tv boxes are a mess :( I also have random kernel panic when booting, also with pure armbian image, so it's not really usable for now.

MichaIng, 2024-05-01:
Indeed they are. Vendors care even less about (mainline/recent) Linux support than in case of SBCs, and the amount of technically-wised users per box (to care about and e.g. port device trees and drivers) is much lower.

At least we know now which files are missing after the conversion, and why.