MichaIng/DietPi #7043: Dietpi-VPN IPVanish installation - Possible bugs with IpVanish's configs.zip and also "keysize" in .ovpn file

DreamPhreak, 2024-04-25:
#### Required Information
- DietPi version | `9.3.0`
- Distro version | `bookworm 1`
- Kernel version | `Linux DietPi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux`
- SBC model | `RPi Zero W (armv6l)`
- Power supply used | `5v 1a usb n/a`
- SD card used | `sandisk ultra 64gb`

#### Additional Information (if applicable)
- Software title | `DietPi-VPN, OpenVPN 2.6.3`
- Was the software title installed freshly or updated/migrated? `yes, fresh install`
- Can this issue be replicated on a fresh installation of DietPi? `yes, this is my 2nd fresh install of DietPi (for a different reason) and I came across the same repeated issue`

## Configs.zip issue
#### Steps to reproduce

1. DietPi-Launcher -> DietPi-VPN option on the main menu. -> IPVanish VPN
2. After filling in all the relevant auth info and trying to start the VPN, it throws this error: 
`Archive: configs.zip  -  End-of-central-directory signature not found.  Either this file is not a zipfile, or it constitutes one disk of a multi-part archive. In the latter case the central directory and zipfile comment will be found on the last disk(s) of this archive.  -  
unzip: cannot find zipfile directory in one of configs.zip or configs.zip.zip, and cannot find configs.zip.ZIP, period.`
3. The command that downloads IPVanish's configs.zip file (`curl -sSfLO 'https://www.ipvanish.com/software/configs/configs.zip'`) seems to point to the wrong page. While yes, this URL does exist, it redirects to `https://configs.ipvanish.com/configs/` - notice now it no longer has configs.zip at the end but is now a normal web page. This must confuse curl, because even with the -L flag, its not able to correctly redirect and then download the configs.zip file from the new URL.
4. I tried to run the old curl command manually in a different session and it just freezes without completing or downloading anything. 
5. Went into the bash shell during installation and ran: `rm -rf configs.zip` to delete the erroneous file that curl made, then ran `wget https://configs.ipvanish.com/configs/configs.zip` (the CORRECT configs.zip file)
6.  The installation continues as normal.
7. My theory is that `https://www.ipvanish.com/software/configs/configs.zip` must have been an older version of the page and they created a new subdomain for it at `https://configs.ipvanish.com/configs/configs.zip`. Editing this line in the `/boot/dietpi/dietpi-vpn` would fix this issue. No reason to continue to point to the old url.

## IPVanish .ovpn issue
#### Steps to reproduce
1. During the installation of the IPVanish VPN, there is another error: `Options error: Unrecognized option or missing or extra parameter(s) in /etc/openvpn/client.ovpn:20: keysize (2.6.3)`
2. `keysize` was depreciated before OpenVPN 2.6.0, and then removed entirely with the release of 2.6.0. Since the latest version is 2.6.3, I don't think IPVanish updated their .ovpn files to reflect this change. (https://community.openvpn.net/openvpn/wiki/DeprecatedOptions#Option:--keysizeStatus:RemovedinOpenVPNv2.6)
3. I fixed this by editing the files `/etc/openvpn/ipvanish/ipvanish-US-Dallas-dal-a01.ovpn` and `/etc/openvpn/client.ovpn`, and deleting the `keysize 256` line entirely. 
4. Saved. Now installation continues normally, works, connected to vpn.
5. Perhaps there's a way to have the dietpi-vpn installation script automatically delete that line?

Michael-Robson, 2024-05-01:
I came to report the same issue too.

MichaIng, 2024-05-01:
Many thanks for reporting. This should fix it: https://github.com/MichaIng/DietPi/commit/9f55bae

Please test:
```sh
curl -sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/dietpi/dietpi-vpn' > /boot/dietpi/dietpi-vpn
```

MichaIng, 2024-05-01:
Indeed, `keysize` is invalid now and still present in their configs. Will be removed on Bookworm and above: https://github.com/MichaIng/DietPi/commit/7b6a556

Will leave it in place on Bullseye. While it has been deprecated there already, it still seems to have an effect. AES-256-CBC basically implies this size, but I want to avoid any chance for a regression. It would still fail if someone installs OpenVPN from Debian Bullseye backports, or of course a recent custom build, but I think that is a rare case.

Joulinar, 2024-05-01:
Probably a IPVanish VPN user can report it to the provider to fix their configuration.

Michael-Robson, 2024-05-01:
> Many thanks for reporting. This should fix it: [9f55bae](https://github.com/MichaIng/DietPi/commit/9f55bae)
> 
> Please test:
> 
> ```shell
> curl -sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/dietpi/dietpi-vpn' > /boot/dietpi/dietpi-vpn
> ```
> 
> If on Bookworm, OpenVPN fails to start due to the `keysize` entry, the indeed we need to remove it as well.

That fixed it for me, thanks!

closed 2024-05-09 by DreamPhreak

MichaIng, 2024-05-09:
Many thanks for testing 👍.