MichaIng/DietPi #7044: Dietpi connecting to Chinese sites?

dr-raypc, 2024-04-26:
I installed Dietpi on a VM and noticed it was connecting to Chinese websites. I figured that VM may have been compromised so I wiped it and created a new one using DietPi's latest release, however Dietpi is still connecting to Chinese websites. Why is that?

MichaIng, 2024-04-26:
Which Chinese website in particular do you mean, and at which step?

- Internet connection checks are done at Quad9 DNS service by default (can be changed).
- DNS resolution is done via whatever your DHCP server serves or you configure manually.
- APT updates are done via `deb.debian.org` by default (can be changed).
- Some downloads from GitHub and `dietpi.com` are done at DietPi updates.
- `dietpi.com` is also connected to for the MOTD (can be disabled) and `dietpi-survey` (after installs and updates).

Apart from this, I do not have any other connections in mind, as long as you do not install software which needs to be downloaded from elsewhere, of course.

MichaIng, 2024-10-26:
I'll mark this as closed due to outstanding reply. Feel free to reopen if required.

closed 2024-10-26 by MichaIng