MichaIng/DietPi #7045: Camera Led not working

mrfoxie, 2024-04-27:
#### Required Information
- DietPi version
```text
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=3
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Kernel version
```text
Linux Server 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux
```
- SBC model `Raspberry Pi 0 2W`
- Power supply used `Raspberry Pi 12.5W Micro USB Power Supply`
- SD card used `Sandisk 128GB Class A`

I have attached Raspberry Pi Camera Module 2 but in that LED is not been turning on although i have turned it on from `dietpi-config` and the camera module, power supply, sdcard and pi is newly purchased

[image]

MichaIng, 2024-05-01:
Did you reboot after changing the setting? If so, can you show the output of the following command:
```sh
vcgencmd get_config disable_camera_led
```
And just in case: You so see the camera actually recording with some software? The LED is intended to turn on only while recording.

mrfoxie, 2024-05-01:
> Did you reboot after changing the setting? If so, can you show the output of the following command:
> 
> ```shell
> vcgencmd get_config disable_camera_led
> ```
> 
> And just in case: You so see the camera actually recording with some software? The LED is intended to turn on only while recording.

yes i rebooted my pi

```
[Server:~]                                         
[root]# vcgencmd get_config disable_camera_led
disable_camera_led=0
```
this is the result of output of the command and im using motioneye to record a video but nope not working

MichaIng, 2024-05-01:
Then I don't know either. Does it work with Bullseye (legacy) RPi OS? We are using the exact same kernel/firmware/bootloader, so whatever works in RPi OS _should_ work in DietPi as well, and the other way round.

MichaIng, 2024-05-13:
Not Bullseye vs Bookworm is relevant, but whether the older `raspberrypi-kernel` kernel is used (which our images provide) or the newer `raspi-firmware` package stack, which still require some development our end to be nicely supported: #6676

In case of RPi OS, the "legacy" Bullseye images come with the old package stack, and RPi OS Bookworm with the new one. The Debian version itself is however generally independent of the underlying kernel/firmware.

mrfoxie, 2024-05-14:
okay no problem, than i will be buying raspberry pi 0 w then🥹😊

MichaIng, 2024-05-14:
Referencing your issue at the RPi firmware repo: https://github.com/raspberrypi/firmware/issues/1890

Weird that there is an LED, if it is not controlled by any modern RPi, without this being documented anywhere. You could configure an own GPIO to control it, but quite a hassle for something that should implicitly work as a basic privacy feature of any camera, without the ability to manipulate it with custom GPIO, IMO.