MichaIng/DietPi #7049: Pi-hole setup fails after applying workaround from #6592 / bond0 interface

arnemoor, 2024-04-28:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | 9.3.0
- Distro version | bookworm
- Kernel version | Linux xxxxx 5.10.160-legacy-rk35xx #1 SMP Fri Feb 2 07:51:33 UTC 2024 aarch64 GNU/Linux

- SBC model | Orange Pi 5 Plus (aarch64)
- Power supply used | -
- SD card used | NVMe

#### Additional Information (if applicable)
- Software title | Pi-hole
- Was the software title installed freshly or updated/migrated? --> Fresh install
- Can this issue be replicated on a fresh installation of DietPi? Most likely not
- Bug report ID | -

#### Steps to reproduce

1. Take a Orange Pi 5 Plus 
2. Apply the networking workaround from https://github.com/MichaIng/DietPi/issues/6592#issuecomment-1997454332 
3. Comment out the "# Ethernet" block in /etc/network/interfaces make the above working
4. Feel free to reboot
5. Log in and start dietpi-software
6. Select "93  Pi-hole: block adverts for any device on your network" for installation
7. Confirm to use unbound
8. Reach the stage "A static IP address is essential..." - this has "Ok" and "Cancel" as options.

#### Expected behaviour
One of the options allows me to continue with setup

#### Actual behaviour
- "If you have already setup your static IP, please ignore this message." -> As bond0 was defined in the above step 2. I would expect "Cancel" to skip that step. But it cancels the installation
- Selecting "OK" brings one to a dialog that explains the Static Setup - "Once completed, select "Apply Save Changes", then exit DietPi-Config to resume setup." - Doing this in whatever way configuring a static ip will exit the setup too and brings one back to "DietPi-Software"

#### Extra details
-

MichaIng, 2024-04-28:
> I would expect "Cancel" to skip that step. But it cancels the installation

It does not cancel the installation, as at this stage the installation has not yet started. Selecting "Cancel" as well as selecting "Ok" and from DietPi-Config "Exit" at any time, brings you back to the DietPi-Software menu, as expected. Then navigate down and select "Install".

When you use the "Search Software" or "Browse Software" dialogs, you afterwards always need to select "Install", to start the installation of the selected software titles.

arnemoor, 2024-05-05:
Thanks @MichaIng - I got confused and didn't realize that this is the normal behaviour I used so often before.  Closing the ticket and thanking for your answer.

closed 2024-05-05 by arnemoor