MichaIng/DietPi #7052: Unable to install Wazuh agent on Pi 5

privacyadmin, 2024-05-01:
### Creating a bug report/issue
- [ y] I have searched the existing open and closed issues

#### Required Information
DietPi version
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=3
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

Distro version 
bookworm 0
Kernel version
Linux DietPi 6.6.28+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.28-1+rpt1 (2024-04-22) aarch64 GNU/Linux

Attempts to install wazuh-agent on Dietpi results in this error.
root@DietPi:~# wget https://packages.wazuh.com/4.x/apt/pool/main/w/wazuh-agent/wazuh-agent_4.7.3-1_arm64.deb && sudo WAZUH_MANAGER='wazuh..com' WAZUH_AGENT_GROUP='default' WAZUH_AGENT_NAME='Dietpi' dpkg -i ./wazuh-agent_4.7.3-1_arm64.deb
--2024-05-01 17:24:23--  https://packages.wazuh.com/4.x/apt/pool/main/w/wazuh-agent/wazuh-agent_4.7.3-1_arm64.deb
Resolving packages.wazuh.com (packages.wazuh.com)... 108.156.133.105, 108.156.133.90, 108.156.133.21, ...
Connecting to packages.wazuh.com (packages.wazuh.com)|108.156.133.105|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 5181326 (4.9M) [binary/octet-stream]
Saving to: ‘wazuh-agent_4.7.3-1_arm64.deb.1’

wazuh-agent_4.7.3-1_arm64.deb 100%[=================================================>]   4.94M  17.3MB/s    in 0.3s

2024-05-01 17:24:24 (17.3 MB/s) - ‘wazuh-agent_4.7.3-1_arm64.deb.1’ saved [5181326/5181326]

(Reading database ... 23937 files and directories currently installed.)
Preparing to unpack .../wazuh-agent_4.7.3-1_arm64.deb ...
Unpacking wazuh-agent (4.7.3-1) over (4.7.3-1) ...
dpkg: dependency problems prevent configuration of wazuh-agent:
 **wazuh-agent depends on lsb-release; however:
  Package lsb-release is not installed.**

dpkg: error processing package wazuh-agent (--install):
 dependency problems - leaving unconfigured
Processing triggers for libc-bin (2.36-9+rpt2+deb12u6) ...
Errors were encountered while processing:
 wazuh-agent

Based on the error, can I install this "lsb-release"? If yes, how do I go about it?

Thank you in advance

Joulinar, 2024-05-01:
> Based on the error, can I install this "lsb-release"? If yes, how do I go about it?

like every other Debian based system.

```
apt install lsb-release
```

MichaIng, 2024-05-01:
Generally, when you install manually downloaded DEB packages, do it via APT:
```sh
apt install ./wazuh-agent_4.7.3-1_arm64.deb
```
APT handles dependencies automatically, while plain `dpkg` does not.

privacyadmin, 2024-05-01:
Noted with thanks on the solutions given. I confirm that the agent has been installed with no issue once lsb-release is installed.

Appreciate the help, @MichaIng @Joulinar

closed 2024-05-01 by privacyadmin